TheImagingSource/tiscamera #183: 'Override Scanning Mode' / binning does not function on 'DFK 33UX264' cameras

jraygauthier, 2018-05-25:
We're using the color version of the following camera on *Ubuntu 16.04* through the
`tiscamera` repo at tag `v-tiscamera-0.9.1`:

```bash
$ tcam-ctrl -l
Available devices:
Model		Type	Serial

DFK 33UX264	V4L2	33710150
```

Note that we get the same results using the most recent version of the `master` branch
(at `50d4ea65bb3513f999bcc97cbd72f5eec936fa58`). 

Attempting to set `"Override Scanning Mode"` through `tcam-ctrl` always gave us
the following response:

```bash
$ tcam-ctrl 33710150 -s "Override Scanning Mode=2"
No property with name 'Override Scanning Mode'
```

Looking at `data/uvc-extensions/usb3.xml`, I can see that this control has
id `0x0199e257`:

```
    <constant type="integer">
      <id>V4L2_OVERRIDE_SCANNING_MODE</id>
      <value>0x0199e257</value>
    </constant>
```

However, and most strangely, I get the following output from `uvcdynctrl`, 'Override Scanning Mode'/`0x0199e257`
being the only parameter from `usb3.xml` which is left unmentionned by the output of `uvcdynctrl`:

```bash
$ sudo uvcdynctrl -c
[libwebcam] Unknown V4L2 user control ID encountered: 0x0098090E (V4L2_CID_USER_BASE + 14)
[libwebcam] Unknown V4L2 user control ID encountered: 0x0098090F (V4L2_CID_USER_BASE + 15)
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E201
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E201, name = 'Exposure Time (us)'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E202
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E202, name = 'Auto Shutter'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E203
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E203, name = 'Auto Exposure Reference'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E205
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E205, name = 'Gain Auto'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E208
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E208, name = 'Trigger Mode'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E209
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E209, name = 'Software Trigger'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E210
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E210, name = 'Trigger Delay'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E211
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E211, name = 'Strobe Enable'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E212
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E212, name = 'Strobe Polarity'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E213
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E213, name = 'Strobe Exposure'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E214
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E214, name = 'Strobe Duration'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E215
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E215, name = 'Strobe Delay'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E216
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E216, name = 'GPOUT'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E217
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E217, name = 'GPIN'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E218
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E218, name = 'ROI Offset X'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E219
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E219, name = 'ROI Offset Y'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E220
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E220, name = 'ROI Auto Center'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E234
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E234, name = 'Trigger Polarity'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E236
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E236, name = 'Trigger Exposure Mode'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E237
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E237, name = 'Trigger Burst Count'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E238
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E238, name = 'Trigger Debounce Time (us)'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E239
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E239, name = 'Trigger Mask Time (us)'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E240
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E240, name = 'Trigger Noise Supression Time'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E241
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E241, name = 'Auto Functions ROI Control'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E242
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E242, name = 'Auto Functions ROI Left'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E243
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E243, name = 'Auto Functions ROI Top'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E244
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E244, name = 'Auto Functions ROI Width'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E245
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E245, name = 'Auto Functions ROI Height'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E246
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E246, name = 'Whitebalance Mode'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E247
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E247, name = 'White Balance Auto Preset'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E248
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E248, name = 'White Balance Green Component'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E249
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E249, name = 'Whitebalance Temperature Preset'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E250
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E250, name = 'White Balance Temperature'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E251
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E251, name = 'Flip Horizontal'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E252
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E252, name = 'Flip Vertical'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E253
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E253, name = 'Highlight Reduction'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E254
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E254, name = 'Exposure Auto Upper Limit Auto'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E255
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E255, name = 'Exposure Auto Lower Limit'
[libwebcam] Unknown V4L2 control ID encountered: 0x0199E256
[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E256, name = 'Exposure Auto Upper Limit'
Listing available controls for device video0:
  Brightness
  Saturation
  Hue
  White Balance Component, Auto
  White Balance Red Component
  White Balance Blue Component
  Gamma
  Gain
```

TIS-Stefan, 2018-05-28:
Hello

The values 5, 6 and 7 enable skipping. After setting the value, the stream must be restarted. In tcam_view you can simply set another frame rate to restart the stream.

Stefan

jraygauthier, 2018-05-28:
@TIS-Stefan : I'm not sure I understand your comment nor why the issue has been closed. I see no new changesets added to the master branch. Also, these values do not change anything:

~~~{.bash}
$ tcam-ctrl 33710150 -s "Override Scanning Mode=5"
No property with name ' Override Scanning Mode'

$ tcam-ctrl 33710150 -s "Override Scanning Mode=6"
No property with name ' Override Scanning Mode'

$ tcam-ctrl 33710150 -s "Override Scanning Mode=7"
No property with name ' Override Scanning Mode'
~~~

Also tested with python, and the size of received images is still unchanged.

TIS-Stefan, 2018-05-28:
Hello

Found it. your camera uses firmware 459. This firmware does not support skipping in Linux (but in Windows.)  Please see the firmware update with Windows Update program below:
[fw781.zip](https://github.com/TheImagingSource/tiscamera/files/2045169/fw781.zip)
The Linux 33U Update does not work currently. After the update please do 
`tcam-ctrl -p 33710150`
The output should contain 
`Override Scanning Mode                  (int) min=1 max=16 step=1 default=1 value=0 category=Unknown
Now you can set these modes.
Sorry for the delay of fixing that issue. 

Stefan

jraygauthier, 2018-05-31:
@TIS-Stefan : Hi again,

I successfully installed the fw update using a windows 10 VM. I can now indeed see:

```bash
$ tcam-ctrl -p 33710150
# ..
Override Scanning Mode (int) min=1 max=16 step=1 default=1 value=0 category=Unknown
```

however, value `2` is not accepted:

```bash
$ tcam-ctrl 33710150 -s "Override Scanning Mode=2"
Found property!
$ tcam-ctrl -p 33710150
# ..
$ Override Scanning Mode  (int) min=1 max=16 step=1 default=1 value=0 category=Unknown
```

values 1, 5,6,7 are accepted:

```bash
# ..
Override Scanning Mode  (int) min=1 max=16 step=1 default=1 value=7 category=Unknown
```

however, the pipeline hangs for both 5 and 6 and has no effect on the frame size for 1 and 7.

Anything else I should be aware of?

jraygauthier, 2018-05-31:
It does not seem to be possible wither to use the "2x skipping" through IC Capture on windows with this fw.

[image]

[image]

TIS-Stefan, 2018-06-01:
Hello
I have the impression, that you do not get any images at all on your computer, even without binning. Is that correct? Maybe your VM does not forward USB 3.0 correctly?

Also: You must set a small video format in Linux (and in Windows ) before you can use binning and skipping. The Override Scanning Mode (Skipping/Binning) does not change the video format's width and height.

Stefan

jraygauthier, 2018-06-01:
Indeed, IC Capture does not seem to work properly on a VM with . It start in live in YUY2 but as soon as I change anything, live mode won't start again and the application even crashes on some settings. However, that is not my target (target being a non VM linux), I just wanted to see what I should expect to see when changing the override scanning mode. 

Can you please give me a complete example of such small video format on linux (BGR8 pixels) and also provide me with a enumeration of the meaning of each integer values in "Override Scanning Mode"?

I'm not sure I understand what you mean with skipping that won't affect received frame's width and height. From the documentation:

> In skipping modes, the image sensor regularly skips a number of rows and/or columns
during readout. The resulting output image is smaller but contains the same field of
view than the image produced by default readout mode.

it seems to be clear that I will get an resulting frame with less row/column thus smaller height / width.

TIS-Stefan, 2018-06-01:
Hello

I already ordered a camera, because my was removed by a colleague. As soon as I have the camera, I will create a pipeline.

jraygauthier, 2018-06-01:
@TIS-Stefan : Just saw you last comment. We will be waiting for your input then.

TIS-Stefan, 2018-06-01:
Did you try that with the tcam-view example, as written above?
The values 5, 6 and 7 enable skipping. After setting the value, the stream must be restarted. In tcam_view you can simply set another frame rate to restart the stream

tcam_view is in examples/cpp. Build them and you can try that.

jraygauthier, 2018-06-01:
No not yet but will do.

jraygauthier, 2018-06-01:
Running tcam_view is a failure on both Ubuntu 16.04 and my dev machine. Output is:

```bash
$ ./tcam_view/tcam_view 
terminate called after throwing an instance of 'Glib::FileError'
Aborted (core dumped)
```

any idea?

jraygauthier, 2018-06-01:
Ah, found why, the application attempt to load its glade file from the current working directory:

```bash
TcamView::TcamView()
{
    builder_ = Gtk::Builder::create_from_file("tcam_view.glade");
//...
```

so:

```bash
$ cd ./tcam_view
$ ./tcam_view 
```

launches the ui.

It might be a good thing if you report a user friendly error to the user instead of this crash.

TIS-Stefan, 2018-06-04:
In the readme at examples/cpp (https://github.com/TheImagingSource/tiscamera/tree/master/examples/cpp)  is written
`In addition to the dependencies of the tiscamera software, the following dependencies must be installed (development versions):

gtkmm >= 3.22`

That is the glade stuff.

Stefan
P.S. Still waiting for the DFK 33UX264 camera here....

TIS-Stefan, 2018-06-04:
Hello
I did on the command line:
` gst-launch-1.0 tcambin serial=48610630 ! video/x-raw,format=BGRA,width=640,height=480,framerate=30/1 ! videoconvert ! ximagesink`
This gives a 640x480 image with a small field of view. I stopped this pipeline. Then I entered:

`
tcam-ctrl 48610630 -s "Override Scanning Mode=5"
 gst-launch-1.0 tcambin serial=48610630 ! video/x-raw,format=BGRA,width=640,height=480,framerate=30/1 ! videoconvert ! ximagesink`
With the same resolution of 640x480 pixels I get a double sized field of view.

Override Scanning Modes can be set for resolutions, that are half (2x) or quarter (4x) of the sensors resolutions only.

No the bad new comes: Higher resolutions than 640x480 can not be used in the DFK 33UX264 for Override Scanning Mode, because the next offered resolution is 1920x1080. This resolution is bigger than the half of 2448x2048, which would be 1224x1024, therefore, the Override Scanning Mode=5 and others are invalid for this resolution. The list of available video formats is hardcoded in the USB descriptor of the camera and V4L2 will handle the formats in this list only.  The only solution for this is to add the half and if supported with binning 4x the quarter video formats. But the memory size for the USB descriptor is limited.
I will speak with the firmware engineers team to add that for the next major firmware release.

Stefan

jraygauthier, 2018-06-04:
Ok, now I understand how "Override Scanning Mode" work. 

The field of view provided by `640x480` + `OSM=5` won't be enough for our application. It would definitely help if we could get the half and/or quarter res  available so that we can use binning with whole FoV.

jraygauthier, 2018-06-04:
Can you provide us with a schedule / release date for this firmware release as soon as it become available?

TIS-Stefan, 2018-06-05:
The beta release date is now. Please download the file below. The update can be done in Windows only, the  Linux firmware update program will work in next release.
[fw249_33ux264.zip](https://github.com/TheImagingSource/tiscamera/files/2071939/fw249_33ux264.zip)

The pipeline is:
`gst-launch-1.0 tcambin serial=33710150 ! video/x-raw, format=BGRx,width=1216,height=1024,framerate=60/1 ! videoconvert ! ximagesink
`
I tested with Override Scanning Mode =1 and 5.

jraygauthier, 2018-06-06:
Impressive! Thanks. Will test on my end.

jraygauthier, 2018-06-06:
I currently have a camera without proper lens but I can nonetheless observe a noticeable x offset (more than the expected 8 pixel due to width not being exactly half the max res) to the left when comparing `1216x1024` w `OSM=5` to what's produced by straight `2448x2048`. Here some vertical slices of screen captures (I placed a black bolt vertically and centered it horizontally in the FOV):

20180606_1216x1024_strange_x_offset:

[image]

20180606_2448x2048:

[image]

Note that:

```bash
$ tcam-ctrl 43710279 | grep OffsetOffset X                                (int) min=0 max=2192 step=2 default=0 value=0 category=Partial Scan
Offset Y                                (int) min=0 max=2044 step=2 default=0 value=0 category=Partial Scan
Offset Auto Center                      (bool) default=true value=false category=Partial Scan
```

I will recheck this with the lens once I can get my hand on it.

Another observation is the pipeline hanging when transitioning from `1216x1024` w `OSM=5` to straight `2448x2048`:

```bash
$ tcam-ctrl 43710279 -s "Override Scanning Mode=5"; gst-launch-1.0 tcambin serial=43710279 ! video/x-raw, format=BGR,width=1216,height=1024,framerate=60/1 ! videoconvert ! ximagesink
# Ok output + ok stream here.
$ tcam-ctrl 43710279 -s "Override Scanning Mode=1"; gst-launch-1.0 tcambin serial=43710279 ! video/x-raw, format=BGR,width=2448,height=2048,framerate=15/1 ! videoconvert ! ximagesink
Found property!
Setting pipeline to PAUSED ...
# Hang here, no stream window.
# Ctrl+c
$ tcam-ctrl 43710279 -s "Override Scanning Mode=1"; gst-launch-1.0 tcambin serial=43710279 ! video/x-raw, format=BGR,width=2448,height=2048,framerate=15/1 ! videoconvert ! ximagesink
# Ok output + ok stream here.
```
So, it finally shows on the second attempt.

TIS-Stefan, 2018-06-06:
Hi

The hanging occurs, because before changing back to the high resolution you must set the Override Scanning Mode back to 1. This is is not done automatically.

I do not understand the x offset part. But I tried that with a lens and it worked as expected.

jraygauthier, 2018-06-06:
> The hanging occurs, because before changing back to the high resolution you must set the Override Scanning Mode back to 1. This is is not done automatically.

But, this is what is done if you look closely at the bash invocations, `tcam-ctrl` is called before launching the pipeline: `tcam-ctrl 43710279 -s "Override Scanning Mode=1"`. Doing it separately still result in a hang.

 > But I tried that with a lens and it worked as expected.

So it might be the lack of lens then.

TIS-Stefan, 2018-06-06:
I found a stupid workaround:

```
#!/bin/sh
clear

tcam-ctrl 48610630 -s "Override Scanning Mode=5"
gst-launch-1.0 tcambin serial=48610630 ! video/x-raw, format=BGR,width=1216,height=1024,framerate=60/1 ! videoconvert ! videoscale ! ximagesink

tcam-ctrl 48610630 -s "Override Scanning Mode=1"; 
gst-launch-1.0 tcamsrc serial=48610630 num-buffers=1 ! video/x-bayer, format=rggb,width=1216,height=1024,framerate=60/1 ! fakesink
gst-launch-1.0 tcambin serial=48610630 ! video/x-raw, format=BGR,width=2448,height=2048,framerate=5/1 ! videoconvert ! videoscale !  ximagesink
```
The trick is changing the Override Scanning Mode back to 1, but run the camera with the small video format. I use tcamsrc and num-buffers=1, so the workaround pipeline terminates on its own.

Stefan

jraygauthier, 2018-06-07:
Just tested with a proper lens and the FoV in OSM=5 in indeed perfectly fitted to max res. 

Do you think the hanging issue will be fixed by the time of the official FW release?

TIS-Stefan, 2018-06-08:
> Do you think the hanging issue will be fixed by the time of the official FW release?
I hope so.

Stefan
--

jraygauthier, 2018-06-08:
@TIS-Stefan : Thank you for the impressive support by the way! We're really impressed about how quickly you were able to respond.

TIS-Stefan, 2018-06-08:
Thank you for your compliments. My firmware programmer Tim made a new firmware, that fixes this issue. I tested that in Linux and Windows and confirm, this problem is gone. The firmware will go in release test, which can last a few weeks, since nearly all 33U and 33G cameras are affected. Some camera tests lasts more than one day.
[fw942.zip](https://github.com/TheImagingSource/tiscamera/files/2084566/fw942.zip)

jraygauthier, 2018-06-08:
Ok, will install and test this.

jraygauthier, 2018-06-11:
Works fine for us. Hanging issue indeed fixed. Thanks.

closed 2018-06-11 by TIS-Stefan