TheImagingSource/tiscamera #191: DFM 72AUC02 in Ubuntu 16.04

Eloring, 2018-06-14:
I want to install the camera DFM 72AUC02 in Ubuntu 16.04. I have finished all the work following your instructions, it seems all right, however,

When I tried to detect the camera through tcam-ctrl -l , I got the message:
No device found.

Does DFM 72AUC02 have the Linux support?
How I can use this camera under Linux?
Thank you so much!

TIS-Stefan, 2018-06-14:
Hi

I suppose, you did not change the firmware in your camera.
Please refer to https://github.com/TheImagingSource/tiscamera/wiki/Getting-Started-with-USB-2.0-CMOS-Cameras

Stefan

Eloring, 2018-06-14:
Thanks for your reply.
When I issue the command: sudo ./firmware-update -ud 7214101 -f ../../../data/firmware/usb2/dfk72uc02_3012.euvc 
the firmware file matches my camera, but it was rejected:
Firmware Size: 19968 EEPROM Size: 16384
Firmware does not fit in EEPROM. Aborting!

Eloring, 2018-06-15:
Thank you so much!!
So it can just used as a monochrome camera under Ubuntu?  But it works well under Windows.

TIS-Stefan, 2018-06-18:
Hello

I tried the beta-release branch and with the new tcam-capture I get a color image using this firmware. Same with a standard pipeline like
'gst-launch-1.0 tcambin serial=15810834 ! video/x-raw,format=BGRx,width=640,height=480,framerate=30/1 ! videoconvert ! videoscale ! ximagesink'

Stefan

closed 2018-06-18 by TIS-Stefan

TIS-Stefan, 2018-07-09:
Hello

You changed the firmware in the camera for Linux. For Windows, you must change it back to dfk72uc02_129.euvc

Stefan

Eloring, 2018-07-09:
Wow, thank you so much!