TheImagingSource/tiscamera #193: How can I find all of the possible resolutions?

yoelk, 2018-06-18:
Hello,
I'm using `tcam-ctrl -f <SERIAL>` to get the supported video formats for my camera.
I get:
```
Available format settings:
Format: Mono8 - Fourcc(808466521)
	Resolutionrange: 480x4 - 1920x1200
		  0.9245 fps
		  2.0000 fps
		  3.0000 fps
		  4.0000 fps
		  5.0000 fps
		  6.0000 fps
		  7.0000 fps
		  8.0000 fps
		  9.0000 fps
		 10.0000 fps
		 11.0000 fps
		 12.0000 fps
		 13.0000 fps
		 14.0000 fps
		 15.0000 fps
		 16.0000 fps
		 17.0000 fps
		 18.0000 fps
		 19.0000 fps
		 20.0000 fps
		 30.0000 fps
		 40.0000 fps
		 41.4000 fps

Format: Mono16 - Fourcc(540422489)
	Resolutionrange: 480x4 - 1920x1200
		  0.9245 fps
		  2.0000 fps
		  3.0000 fps
		  4.0000 fps
		  5.0000 fps
		  6.0000 fps
		  7.0000 fps
		  8.0000 fps
		  9.0000 fps
		 10.0000 fps
		 11.0000 fps
		 12.0000 fps
		 13.0000 fps
		 14.0000 fps
		 15.0000 fps
		 16.0000 fps
		 17.0000 fps
		 18.0000 fps
		 19.0000 fps
		 20.0000 fps
		 20.7000 fps
```

The resolution range is written, but how can I know which possible resolutions are there?
Are all integer resolutions supported in this range?

Thanks,
Joel

yoelk, 2018-06-18:
And adding to this question.
Is it possible to get this information from the python intrerface?
I see the following interface functions:
```
iface->get_property_names = gst_tcam_src_get_property_names;
    iface->get_property_type = gst_tcam_src_get_property_type;
    iface->get_property = gst_tcam_src_get_tcam_property;
    iface->get_menu_entries = gst_tcam_src_get_menu_entries;
    iface->set_property = gst_tcam_src_set_tcam_property;
    iface->get_device_serials = gst_tcam_src_get_device_serials;
    iface->get_device_info = gst_tcam_src_get_device_info;
```

I've tried using them but didn't see which one gives me information on the current video format (pixel format, resolution, fps) and which are the possible values.

Thanks a gain! Much appreciated!

TIS-Stefan, 2018-06-18:
Hello 

First of all:
tcam-ctrl -c <serial number> lists video formats and framerates. You can get these information in Python too, as shown in the samples at https://github.com/TheImagingSource/tiscamera/tree/master/examples/python

You know the current video format the camera uses, because usually you define it by yourself.  Otherwise, if you need to know resolution and bytes per pixel, then you use the caps of the GstSample of the appsink module. The following is an excerpt from a script I wrote:

``` Pyhton
        p = 'tcambin serial="%s" name=source ! video/x-raw,format=%s,width=%d,height=%d,framerate=%d/1' % (serial,format,width,height,framerate,)
        p += ' ! videoconvert ! appsink name=sink'

        print(p)
        try:
            self.pipeline = Gst.parse_launch(p)
        except GLib.Error as error:
            print("Error creating pipeline: {0}".format(err))
            raise

        self.pipeline.set_state(Gst.State.READY)
        self.pipeline.get_state(Gst.CLOCK_TIME_NONE)
        # Query a pointer to our source, so we can set properties.
        self.source = self.pipeline.get_by_name("source")

        # Query a pointer to the appsink, so we can assign the callback function.
        self.appsink = self.pipeline.get_by_name("sink")
        self.appsink.set_property("max-buffers",5)
        self.appsink.set_property("drop",1)
        self.appsink.set_property("emit-signals",1)
        self.appsink.connect('new-sample', self.on_new_buffer)

    def on_new_buffer(self, appsink):
        self.newsample = True
        if (self.samplelocked == False):
            try:
                self.sample = appsink.get_property('last-sample')
            except GLib.Error as error:
                print("Error on_new_buffer pipeline: {0}".format(err))
                raise
        return False
```

```    
def Get_image(self):
        # Sample code from https://gist.github.com/cbenhagen/76b24573fa63e7492fb6#file-gst-appsink-opencv-py-L34
        if( self.sample != None and self.newsample == True):
            self.samplelocked = True
            buf = self.sample.get_buffer()
            caps = self.sample.get_caps()
            bpp = 4;
            if( caps.get_structure(0).get_value('format') == "BGRx" ):
                bpp = 4;

            if(caps.get_structure(0).get_value('format') == "GRAY8" ):
                bpp = 1;

            self.img_mat = numpy.ndarray(
                (caps.get_structure(0).get_value('height'),
                 caps.get_structure(0).get_value('width'),
                 bpp),
                buffer=buf.extract_dup(0, buf.get_size()),
                dtype=numpy.uint8)
            self.newsample = False
            self.samplelocked = False
            return self.img_mat
        return None

```

Stefan
--

yoelk, 2018-06-19:
Thanks Stefan, that was very helpful. I found both ways.

May also ask, if it's possible to set video-formats on the fly, for an existing pipeline?
e.g. setting the fps or the size.

Or is it necessary to start a new pipeline?

Thanks again,
Joel

yoelk, 2018-06-21:
Hi Stefan,
After reading yesterday evening [here](https://gstreamer.freedesktop.org/documentation/application-development/advanced/pipeline-manipulation.html) on GStreamer (see the `Changing format in a PLAYING pipeline` part), I found out that it's possible to change the video format on the fly without restarting the pipeline, by using a capsfilter in the pipeline to begin with.
The caps can then be used to set the video format while playing.
FYI

TIS-Stefan, 2018-06-21:
That is, want I meant by stopped and be started again.
Stefan