TheImagingSource/tiscamera #194: How to properly close a tcamsrc interface and pipelines

yoelk, 2018-06-20:
Hello,
I tried modifying the two-cameras.py example (taken from https://github.com/TheImagingSource/tiscamera/issues/117).
The result is [here](https://gist.github.com/yoelk/9d57fac416fea8017e0c233dffaf67a5)

I saw that if I add two lines before the initialization of the TIS camera object, I can't get captured frames (see in the code):
```
tcam1 = Gst.ElementFactory.make('tcamsrc')
tcam1.set_property("serial", "20810358")
```

As far as I understood until now, these lines open a tcam interface which I can use to read information from cameras (e.g. available video formats, available FPS, camera properties, etc.)

In my code, I need to read this information before I start a capture. Is this possible?
Or, is it possible to close the opened tcam interface, so that I can continue with the rest of the code?

Another side question:
Using this example (without the two problematic lines), I can't close the program properly.
When I press `Esc`, the frames stop coming but the window stays open and I have to force-quit the window.
Debugging showed that in the line `LeftCam.Stop_pipeline()`, the following method call doesn't return, which makes the whole program stuck: `self.pipeline.set_state(Gst.State.READY)`

So how should one close an existing pipeline?

Thanks in advance. It's really helpful to have someone to address such questions to.
Joel

TIS-Stefan, 2018-06-21:
Hello

In the samples at https://github.com/TheImagingSource/tiscamera/tree/master/examples/python you will find sample code that shows, how to list video formats frames etc.

The live stop issue should be fixed in the beta-release branch of the repository.

Stefan

yoelk, 2018-06-22:
Thanks, looking forward for the beta-release.
Is there an expected release date?

TIS-Stefan, 2018-06-22:
> Is there an expected release date?
In case you do not find errors, I guess by end of next week.

Hopefully we also do not find errors.

Stefan

yoelk, 2018-06-25:
Thanks Stefan.
I've noticed that stopping the pipeline isn't a problem in all of your examples. Only in the `two_cameras` and `opencv` examples does this problem occur.

In the `live_video.py` example, instead of using `pipeline_sink.connect('new-sample', callback)` you use `bus.connect("message::eos", self.on_eos)`.

Can you tell me a bit about the difference between the two? The second one seems to close properly.
Can I use this option with `appsink`? Or is this only working for `gtksink`?

Thanks!
Joel

closed 2019-11-29 by TIS-Stefan