TheImagingSource/tiscamera #195: list_formats() gives wrong results that are not in tcam-ctrl -c

yoelk, 2018-06-20:
Hello Stefan,
I followed your answer from https://github.com/TheImagingSource/tiscamera/issues/193.

Comparing the results I get from `list_formats()` (from the examples folder) and `tcam-ctrl -c` I see that they are not the same.
`list_formats()` gives a few extra results, which are not in `tcam-ctrl -c`.

For my camera (DMK Z12GX236), `tcam-ctrl -c` gives:
```
video/x-raw, format=(string)GRAY8, width=544, height=480, fps={...}
video/x-raw, format=(string)GRAY8, width=640, height=480, fps={...}
video/x-raw, format=(string)GRAY8, width=576, height=480, fps={...}
video/x-raw, format=(string)GRAY8, width=720, height=480, fps={...}
video/x-raw, format=(string)GRAY8, width=960, height=720, fps={...}
video/x-raw, format=(string)GRAY8, width=1280, height=720, fps={...}
video/x-raw, format=(string)GRAY16_LE, width=544, height=480, fps={...}
video/x-raw, format=(string)GRAY16_LE, width=640, height=480, fps={...}
video/x-raw, format=(string)GRAY16_LE, width=576, height=480, fps={...}
video/x-raw, format=(string)GRAY16_LE, width=720, height=480, fps={...}
video/x-raw, format=(string)GRAY16_LE, width=960, height=720, fps={...}
video/x-raw, format=(string)GRAY16_LE, width=1280, height=720, fps={...}
```

The results from `list_formats()` have two new sizes:
```
1: GRAY8 (480x4) <------- NEW
2: GRAY8 (544x480)
3: GRAY8 (640x480)
4: GRAY8 (576x480)
5: GRAY8 (720x480)
6: GRAY8 (960x720)
7: GRAY8 (1280x720)
8: GRAY8 (1560x900) <------- NEW
9: GRAY16_LE (480x4) <------- NEW
10: GRAY16_LE (544x480)
11: GRAY16_LE (640x480)
12: GRAY16_LE (576x480)
13: GRAY16_LE (720x480)
14: GRAY16_LE (960x720)
15: GRAY16_LE (1280x720)
16: GRAY16_LE (1560x900) <------- NEW
```

When I try to set these new sizes to a pipeline I get an error, so I guess `list_formats()` gives wrong results. Is there a way to filter such wrong results? Any idea why they are there?

Thanks again,
Joel

yoelk, 2018-06-20:
Well, I solved it by comparing the two, but if there is a cleaner way to get only the right video formats from `list_formats()`, I'd be happy to know about it.

yoelk, 2018-06-21:
And another small remark:
I have to use `list_formats()` as only there I can get the numerator (N) and denominator (D) of the FPS values.
`tcam-ctrl -c` only gives the quotient (FPS = N/D), but this is not enough.
When I try to set the FPS (for example, using the capsfilter), I need to specify N and D: 
`"video/x-raw,format=GRAY8,width=480,height=4,framerate=33/1"`. 
Simply writing `"framerate=33"` or `"framerate=33.0"` doesn't work.

So to conclude, I need `list_formats()` to get N and D, but it also gives additional wrong sizes.
So I read the right sizes from `tcam-ctrl -c` and filter out the wrong sizes from the `list_formats()` results.

Does that make sense?

TIS-Stefan, 2018-06-21:
You are right, the N/D stuff is known. What are "wrong sizes"?

yoelk, 2018-06-21:
By "wrong sizes" I refer to the reported sizes listed in my first comment.
For example, it is written that `GRAY8 (480x4)` is one of the possible video formats but it isn't.
Trying to use this in a pipeline results in an error.
You can also see that this format is not listed in the `tcam-ctrl -c` output for the same camera

TIS-Stefan, 2018-06-21:
The video formats are correct. Did you also set a frame rate of the list?

The camera has variable video formats, beginning at 480x4 ending at 1920x1200. There is a stepping in width and height, but the list of possible video formats has a few 100.000 entries. Therefore, the programs show only a few formats. I have to admit, I do not know, how these formats are handled by GStreamer, but using Aravis directly, you can set all of them.

Stefan

yoelk, 2018-06-22:
Well, I've tried running `tiscamera/examples/python/live_video.py`, with my camera's serial (type: `DMK Z12GX236`).
This worked:
```
pipeline_str = 'tcambin name=src ' \
                       '! queue max_size_buffers=2 ' \
                       '! videoconvert ' \
                       '! capsfilter caps="video/x-raw,format=GRAY8,width=544,height=480,framerate=20/1" ' \
                       '! videoconvert ' \
                       '! gtksink name=sink'
```

but this didn't:
```
pipeline_str = 'tcambin name=src ' \
                       '! queue max_size_buffers=2 ' \
                       '! videoconvert ' \
                       '! capsfilter caps="video/x-raw,format=GRAY8,width=480,height=4,framerate=20/1" ' \
                       '! videoconvert ' \
                       '! gtksink name=sink'
```

As for "using Aravis directly", what do you mean by that?

TIS-Stefan, 2018-06-22:
Hello

You are right, 480x4 does not work with Aravis. In Windows it works fine with our driver, that transfers GigEVision to DirectShow. But: NI MAX using GigEVision offers 480x4 but that does not work, while starting at 480x8 images are provided. 

That is not intended.

Stefan

TIS-Stefan, 2018-06-22:
Some additional information:
arv-viewer does not show 480x4, until you set the packet size:

`arv-tool-0.6 control GevSCPSPacketSize=2984 -n TIS-<camera serial number>`

However, that wont help with the tcambin, because it lets Aravis auto-negotiate the packet size again which results in a wrong packet size and therefore you wont get images. Unfortunately the error is in Aravis. I will see, what can be done about that.

Stefa

TIS-Stefan, 2018-06-22:
And another addition regarding wrong list of formats.

There is no list of formats, there is a range only. You can (could) start at 480x4 and end at 1920x1200. The line stepping is 4, I guess, the column stepping is also 4 or 8. That means, you can set also a video format like 480x32 or 800x64.

The tools show only a small excerpt of possible video formats, otherwise you will have a list with 107.640 formats. ( (1920-480) * (1200-4) * 4). 

I admit, I did not knew that too, I thought, even for GigE cameras this list is limited, like for V4L2 devices. 

Stefan

closed 2019-11-29 by TIS-Stefan