TheImagingSource/tiscamera #197: Not all properties are shown in `tcam-ctrl -p`

yoelk, 2018-06-21:
For my camera (DMK Z12GX236), `tcam-ctrl -p` gives:
```
Found 32 propert(y/ies)
WDR                                     (bool) default=false value=false category=Unknown
Exposure                                (int) min=20 max=30000000 step=1 default=66666 value=66666 category=Exposure
Trigger Selector                        (enum)  default=FrameStart category=Special
						value=FrameStart
							FrameStart
Trigger Mode                            (enum)  default=Off    category=Special
						value=Off
							Off
							On
Trigger Source                          (enum)  default=Any    category=Special
						value=Any
							Any
							Line1
							Software
Trigger Activation                      (enum)  default=RisingEdge category=Special
						value=RisingEdge
							FallingEdge
							RisingEdge
Software Trigger                        (button) category=Special
Trigger Delay (us)                      (double) min=0 max=1e+06 step=0.001 default=4 value=4 category=Special group=429785097
TimestampReset                          (button) category=Unknown
TimestampLatch                          (button) category=Unknown
TimestampLatchValue                     (int) min=0 max=281474976710655000 step=1 default=0 value=0 category=Unknown
Offset X                                (int) min=0 max=1440 step=1 default=688 value=688 category=Partial Scan
Offset Y                                (int) min=0 max=1196 step=1 default=360 value=360 category=Partial Scan
Offset Auto Center                      (enum)  default=On     category=Partial Scan
						value=On
							Off
							On
Reverse X                               (bool) default=false value=false category=Unknown
Reverse Y                               (bool) default=false value=false category=Unknown
Binning Vertical                        (int) min=1 max=2 step=1 default=1 value=1 category=Unknown
Binning Horizontal                      (int) min=1 max=2 step=1 default=1 value=1 category=Unknown
Gain                                    (double) min=0 max=48 step=0.001 default=0 value=0 category=Exposure group=429785092
Black Level                             (double) min=0 max=511 step=0.001 default=60 value=60 category=Unknown group=0
GPIO                                    (int) min=0 max=3 step=1 default=2 value=2 category=Special
GPIn                                    (int) min=0 max=1 step=1 default=1 value=1 category=Special
GPOut                                   (int) min=0 max=1 step=1 default=0 value=0 category=Special
Strobe Enable                           (enum)  default=Off    category=Special
						value=Off
							Off
							On
Strobe Polarity                         (enum)  default=ActiveLow category=Special
						value=ActiveLow
							ActiveHigh
							ActiveLow
Strobe Operation                        (enum)  default=Exposure category=Special
						value=Exposure
							Exposure
							FixedDuration
Strobe Duration                         (int) min=0 max=32767 step=1 default=100 value=100 category=Special
Strobe Delay                            (int) min=0 max=32767 step=1 default=0 value=0 category=Special
Zoom                                    (int) min=0 max=100 step=1 default=0 value=0 category=Lens
Focus                                   (int) min=0 max=1000 step=1 default=500 value=500 category=Lens
Iris                                    (int) min=0 max=1023 step=1 default=1023 value=1023 category=Lens
IRCutFilterEnableElement                (bool) default=false value=false category=Unknown
```

In the [camera's manual](https://s1-dl.theimagingsource.com/api/2.5/packages/documentation/manuals-trm/trmdmk33g445/4d895679-5642-5ebd-8e95-239d62a512a4/trmdmk33g445.en_US.pdf) and also when I read the properties from python code, I see there are other options, such as `Exposure Auto`.

Is there a bug that prevents this (and other) options from being shown in `tcam-ctrl -p`?

Thanks
Joel

TIS-Stefan, 2018-06-21:
Hello 

Some camera models have no automatics on board, they are done by software. Usually with out GStreamer modules for auto expouse, gain, white balance and focus. tcam-ctrl does not load the GStreamer modules, therefore the properties are not shown.
In Windows our DirectShow driver implementes these automatics.

Stefan

yoelk, 2018-06-22:
I see, you mean your `tcambin` GStreamer object, right?
Using it, indeed I get the automatics properties.
I did notice though, that I get them in the properties list only AFTER I've started the video stream (`self.pipeline.set_state(Gst.State.PLAYING)`).

Is this right? Can I get them also before starting to play?

TIS-Stefan, 2018-06-22:
> I see, you mean your tcambin GStreamer object, right?
Correct.

> did notice though, that I get them in the properties list only AFTER I've started the video stream 
> (self.pipeline.set_state(Gst.State.PLAYING)).

Unfortunately this is correct too. Does state.READY do not work?

Stefan

yoelk, 2018-06-22:
```
 Does state.READY do not work?
```
No, unfortunately not.

TIS-Stefan, 2018-06-25:
Hello

You are right, even in the beta-release the software generated properties of the tcambin are not available after state.Ready. Only after state.PLAYING they are available. Therefore, as workaround, you should set the PLAYING state once.

Stefan

closed 2018-06-25 by TIS-Stefan