TheImagingSource/tiscamera #202: Can't keep `Focus Auto` on

yoelk, 2018-08-08:
- OS: Ubuntu 16.04
- python: 3.6
- TIS version: 0.9.1
- TIS version source: https://github.com/TheImagingSource/tiscamera/releases/download/v-release-candidate-0_9_1-1/tiscamera_0.9.1_amd64.deb
- Camera type: DMK Z12GX236
- test code: https://gist.github.com/yoelk/e9c8130311c67971f5797c9c0d58f4df
- The code is based on Two-cams.zip which I found here: https://github.com/TheImagingSource/tiscamera/issues/117

In the test code:
- pressing `1` sets `Gain Auto`, `Exposure Auto` and `Focus Auto` to True
- pressing `2` sets `Gain Auto`, `Exposure Auto` and `Focus Auto` to False
- pressing `3` prints the current state of `Gain Auto`, `Exposure Auto` and `Focus Auto`

The problem is, that if I set them to True and print them a few times, after a a few seconds `Focus Auto` returns to `False`, while the other two stay `True`.

Am I doing something wrong? Is there an additional step for turning `Focus Auto` to True?

Thanks a lot!
Joel

TIS-Edgar, 2018-08-08:
Hi Joel,

The behavior you see is correct. Setting Focus Auto to true causes the algorithm to run once, determining the sharpest focus point and ending. After this Focus Auto is returned to False.
If you look at the property type in the introspection interface it is a button not a boolean.
Currently there is no permanent auto focus just a "auto focus run once".

yoelk, 2018-08-08:
Thanks for the answer!

closed 2018-08-10 by yoelk