TheImagingSource/tiscamera #204: Problem with the tcam_demo.cpp example

yoelk, 2018-08-10:
- OS: Ubuntu 16.04
- TIS version: 0.9.1
- TIS version source: https://github.com/TheImagingSource/tiscamera/releases/download/v-release-candidate-0_9_1-1/tiscamera_0.9.1_amd64.deb
- Camera type: DMK Z12GX236
- test code: https://github.com/TheImagingSource/tiscamera/blob/master/examples/cpp/examples/tcam_demo.cpp

I built and ran the code. I could see that the camera is recognized and that the video formats and properties are printed (although without the automatics properties).

The problem is that I the  GTK window doesn't show (I see its existence on the sidebar but not on the screen) and the `new_frame_cb` callback isn't called at all.

Can someone please help me see what I'm missing here?

Thanks a lot!
Joel

TIS-Arne, 2018-08-10:
The problem is that the example does not set a video format on the camera. Add a line similar to this somewhere before ```cam.start()``` and it should work:

```
    cam.set_capture_format("GRAY8", {640,480}, {30,1});
```

yoelk, 2018-08-14:
Thanks, that solved it

closed 2018-08-14 by yoelk