TheImagingSource/tiscamera #205: GigE camera setup for beginners: connection; tcam-gigetool, setup...

calocedrus, 2018-08-16:
Hello,

I have a DFK Z30GP031 camera that I'm trying to connect to with ubuntu 16.04. In a nutshell, I am lost with what else I should do to operate my camera, and hopefully this post and the answers it may receive could help other Imaging source gige camera novices.
Here is what I've done.
I've installed the required software according to the readme, with aravis build (this is well documented and went smoothly):

```
git clone --recursive https://github.com/TheImagingSource/tiscamera.git
cd tiscamera
mkdir build
cd build
cmake -DBUILD_ARAVIS=ON -DBUILD_GST_1_0=ON -DBUILD_TOOLS=ON -DBUILD_V4L2=ON -DCMAKE_INSTALL_PREFIX=/usr ..
make
sudo make install
```
All dependencies were installed, the above ended with no error.
I've started the gige-daemon:
```
sudo systemctl daemon-reload                 # make systemd aware of gige-daemon
sudo systemctl enable gige-daemon.service    # start on every boot
sudo systemctl start gige-daemon.service     # start the actual daemon
sudo systemctl status gige-daemon.service    # check if statemd say everything is ok

```
The later command returned:
```
gige-daemon.service - GigE indexing daemon
   Loaded: loaded (/lib/systemd/system/gige-daemon.service; enabled; vendor pres
   Active: active (running) since Thu 2018-08-16 08:42:48 CST; 7h ago
 Main PID: 1021 (gige-daemon)
   CGroup: /system.slice/gige-daemon.service
           └─1021 /usr/bin/gige-daemon start --no-fork

Aug 16 08:42:48 croster systemd[1]: Started GigE indexing daemon.
Aug 16 14:09:32 croster systemd[1]: Started GigE indexing daemon.
```

I power the camera with the included power supply and plug it directly into the ethernet port of my computer.
In addition to the readme, I am also following the guide "GigE Camera Start Up" found at https://s1-dl.theimagingsource.com/api/2.5/packages/publications/whitepapers-cameras/wpgigecamerastartup/4df6c9cb-9fb5-50d0-8b42-82259fd2124b/wpgigecamerastartup_1.0.en_US.pdf

My problems and questions: 
1) how should I connect the camera? Directly in my ethernet port? Via a switch? Via a router? All of these are possible?
2) addendum to question 1: depending on how I've connected my camera, what do I have to do to set it up: do I have to create a new wired connection in ubuntu? (in the "GigE Camera Start Up" , referring to tcam-gigetools, it is written "This tool can be used to set the IP addresses of the cameras" but no further instructions are given on how to do so.
3) `tcam-gigetool` doesn't exist (command not found). Neither the readme nor the "GigE Camera Start Up" explain how to get this tool. Digging into the closed issues, I've found this way and report it so others can hopefully locate it more easily:```

cd path_to_tiscamera_build # the build directory of tiscamera
cd tools/tcam-gigetool
make
sudo python3 setup.py install
```
(it would be helpful for others to include these steps either in the readme or the above guide)

4) now that I have installed tcam-gigetools and my camera plugged in the ethernet port of my computer, tcam-gigetool list returns nothing. 
ifconfig returns:
enp5s0    Link encap:Ethernet  HWaddr 34:97:f6:xx:3x:x7  
(no inet6 address, no IP address)

5) In the GigE camera Start up guide, I read: "If a camera is in an unsuitable network and is not listed, the rpfilter must be turned off": if I do so will it affect my normal internet connection (using the same ethernet port on my computer)?

6) following the troubleshooting section of the "Gige camera startup" guide: both LED on the camera are on, the yellow one flashes intermitently.
I have tried the tcam-gigetools rescue with my camera's serial number (but which IP address should I provide there?):
tcam-gigetool rescue --ip 192.168.2.129 --netmask 255.255.255.0 --gateway 0.0.0.0 19410440
But:
Could not find camera '19410440'. Aborting.

7) from the github readme,
` gst-launch-1.0 tcambin ! videoconvert ! ximagesink `
returns:
```
in ! videoconvert ! ximagesink
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
Setting pipeline to NULL ...
Freeing pipeline ...
```
8) the camera has been tested on a Windows machine with the GigECam IP Config program and it was recognized within few seconds after direct connection into the computer's ethernet port without any particular setup.

Could someone guide me toward what other steps I could take?

TIS-Arne, 2018-08-16:
> how should I connect the camera? Directly in my ethernet port? Via a switch? Via a router? All of these are possible?

The camera needs power so either you need to connect an external power supply or you need to connect the camera to a powered port of a gigabit ethernet switch. 

> ifconfig returns:
> enp5s0 Link encap:Ethernet HWaddr 34:97:f6:xx:3x:x7

This means that your network is unconfigured/down and thus you could not find any camera. If you do not connect to an existing network on this network interface, you need to configure this interface. You could set a static IP address on your network interface like this: https://www.howtoforge.com/linux-basics-set-a-static-ip-on-ubuntu

After the network interface is configured, tcam-gigetool should show your camera and you could set a proper IP adress for the device. Afterwards it should come up in tcam-capture just fine.

calocedrus, 2018-08-17:
Arne, thank you.
> The camera needs power so either you need to connect an external power supply

It was, as I wrote in my original post (granted, it was a lot to read).

> you need to configure this interface. You could set a static IP address on your network interface like this: https://www.howtoforge.com/linux-basics-set-a-static-ip-on-ubuntu

Thank you, this was a good pointer. There was an important additional command to run which wasn't indicated in the above link.
I summarize here what I did (linux ubuntu 16.04LTS):
Backup the original /etc/network/interfaces file.
Edit /etc/network/interfaces:
`sudo vi /etc/network/interfaces`
and modify its content according to the following:
```
source /etc/network/interfaces.d/*

# The loopback network interface
auto lo
iface lo inet loopback

# The primary network interface
# get its name with ifconfig (it usually starts with enp...)
auto enp5s0 
iface enp5s0 inet static
# any static ip address
address 192.168.1.100 
netmask 255.255.255.0
gateway 192.168.1.1
dns-nameservers 8.8.4.4 8.8.8.8
Then the following is required:
sudo ip addr flush enp5s0
sudo systemctl restart networking.service
```
(reference: https://askubuntu.com/a/849641/645810)

After this indeed I could use `tcam-gigetools list` and run some examples.
Thanks again.

closed 2018-08-17 by calocedrus