TheImagingSource/tiscamera #208: DFK 37BUX290 Highlight Reduction

kylemcdonald, 2018-08-22:
I have an image with a large dynamic range and I would like to apply highlight reduction. Section 5.3.4 of the technical document for DFK 37BUX290 the describes this feature https://s1-dl.theimagingsource.com/api/2.5/packages/documentation/manuals-trm/trmdfk37bux290/857db88f-86ac-5331-af87-722934cb8ff0/trmdfk37bux290.en_US.pdf

But when I check the available parameters of the camera I do not see this option. Here is the example output of `tcam-ctrl`:

```
$ tcam-ctrl -p 10810671
Found 35 propert(y/ies)
Brightness                              (int) min=0 max=511 step=1 default=240 value=0 category=Exposure
Whitebalance Auto                       (bool) default=true value=true category=Color
Whitebalance Red                        (int) min=0 max=255 step=1 default=64 value=64 category=Color
Whitebalance Blue                       (int) min=0 max=255 step=1 default=64 value=64 category=Color
Gain                                    (int) min=0 max=240 step=1 default=0 value=0 category=Exposure
Exposure                                (int) min=20 max=30000000 step=1 default=333 value=2000 category=Exposure
Exposure Auto                           (bool) default=true value=true category=Exposure
Exposure Auto Reference                 (int) min=0 max=255 step=1 default=128 value=46 category=Exposure
Gain Auto                               (bool) default=true value=false category=Exposure
Trigger Mode                            (bool) default=false value=false category=Special
Software Trigger                        (button) category=Special
Trigger Delay (us)                      (int) min=0 max=100000 step=1 default=0 value=31 category=Special
Strobe Enable                           (bool) default=false value=false category=Special
Strobe Polarity                         (bool) default=false value=false category=Special
Strobe Exposure                         (bool) default=true value=true category=Special
Strobe Duration                         (int) min=0 max=32767 step=1 default=100 value=100 category=Special
Strobe Delay                            (int) min=0 max=32767 step=1 default=0 value=0 category=Special
GPOut                                   (int) min=0 max=1 step=1 default=0 value=0 category=Special
GPIO                                    (int) min=0 max=1 step=1 default=0 value=0 category=Special
Offset X                                (int) min=0 max=1552 step=2 default=0 value=0 category=Partial Scan
Offset Y                                (int) min=0 max=824 step=2 default=0 value=0 category=Partial Scan
Offset Auto Center                      (bool) default=true value=true category=Partial Scan
Trigger Polarity                        (enum)  default=RisingEdge category=Special
						value=RisingEdge
							FallingEdge
							RisingEdge
Auto Functions ROI Control              (bool) default=true value=true category=Special
Whitebalance                            (bool) default=false value=false category=Color
Whitebalance Auto Preset                (enum)  default=Any    category=Color
						value=Warm White
							Any
							Cool White
							Daylight
							Deep Shade
							Warm White
Whitebalance Green                      (int) min=0 max=255 step=1 default=64 value=64 category=Color
White Balance Temperature Preset        (enum)  default=       category=Color
						value=Cool White LED
							Blue Sky
							Cool White Fluorescent Light
							Cool White LED
							Cool White/Daylight (CFL)
							Daylight
							Halogen Light
							Neutral White
							Sodium-Vapur Lamp
							Warm Light
Whitebalance Temperature                (int) min=2500 max=10000 step=100 default=6500 value=7500 category=Color
Reverse X                               (bool) default=false value=false category=Unknown
Reverse Y                               (bool) default=false value=false category=Unknown
Exposure Auto Upper Limit Auto          (bool) default=true value=true category=Exposure
Exposure Auto Upper Limit               (int) min=20 max=1000000 step=1 default=333 value=400000 category=Exposure
Override Scanning Mode                  (int) min=1 max=16 step=1 default=1 value=0 category=Unknown
Auto Functions ROI Preset               (enum)  default=Center 50% category=Special
						value=Center 50%
							Bottom Half
							Center 25%
							Center 50%
							Custom Rectangle
							Full Sensor
							Top Half
```

There also appears to be some typos in the manual, for example: `tcam-ctrl` lists "Gain Auto" and "Exposure Auto", but the manual lists "Auto Gain" and "Auto Exposure". Or `tcam-ctrl` lists "Trigger Delay (us)" but the manual lists "Trigger Delay".

TIS-Stefan, 2018-08-23:
Hello Kyle

the properties are available in Linux with firmware version 910. Your camera has version 830. 
[33UFirmware.910.zip](https://github.com/TheImagingSource/tiscamera/files/2314506/33UFirmware.910.zip)

I suppose, the file will be saved into your "~/Downloads" directory. Unzip the file, you will get a file named _33UFirmware.910.fwpack_, which contains the firmware and is read by the update program.

Open a terminal.

I would like you to update the repository with 
```
git pull
cd build 
make -j
sudo make install
```
The Linux firmware update tool is built with the _tiscamera_ repository, which you already built. Please change into your _tiscamera_ directory.

Then enter
```
cd build/toos/firmware-update
sudo ./firmware-update -u -f ~/Downloads/33UFirmware.910.fwpack -d 10810671
```
Now you will be warned, the camera can be damaged due to firmware update. Then continue. Some percent number will be shown. The update process can last up to two minutes.
When finished, the program advises to disconnect your camera. Reconnect the camera and check the properties again. Highlight Reduction and some more new properties will be available. 
Unfortunately I do not have a firmware change log, so I can not provide information, what else did change.

Stefan

kylemcdonald, 2018-08-23:
Great! Updating the firmware made the option available. Unfortunately I seem to have read the description too quickly, and now I realized that "Highlight Reduction" only works in collaboration with Auto Exposure to set the exposure. I was hoping it actually modified the image.

This is a separate question, but is there any chance that this model camera will get an option to apply a gamma curve before the 8-bit digitization step? For me, some kind of gamma preprocessing to avoid posterization in darker regions after color correction would make this camera incredibly useful.

TIS-Stefan, 2018-08-24:
"Enabling Highlight Reduction lets the auto exposure and auto gain functions reduce
overexposed areas in the output image." That means, gain and exposure are reduced, until the over exposed areas reduced. The image becomes darker then. So you can use lower exposure times to remove the overexposed areas.

Gamma is applied after digitization. 

If you used Windows, I advise to use a 16 bit image format and enable tone mapping. This is for converting high dynamic range images to 8 bit. But you could do the same in Linux. I must admit, I do not know, how to do that with our tcambin, but the base module tcamsrc can receive the caps. A suitable pipeline could be 

`tcamsrc  ! video/x-bayer, format=rggb16, width=640, height=480, framerate=15/1 ! appsink`

This results in a bayer raw image with 16 bit per pixel. From this only the upper 10 bit contain valid data. Combined with lower exposure times, you may get the image information you need. You can try to create an OpenCV image from this and let openCV debayer (cvConvert()) that.

If you have questions about this, please let me know your programming language, when asking. 

Stefan

kylemcdonald, 2018-08-24:
Unfortunately the DFK 37BUX290 doesn't output real 16-bit data: "The sensor of the DFK 37BUX290 camera is not able to provide 16-bit data output. The pixel data is transmitted in the most significant bits."

Thanks for your help!

closed 2018-08-24 by kylemcdonald

TIS-Stefan, 2018-08-27:
Hello 

You are right, max bits is 10 as written on the website of the camera at https://www.theimagingsource.com/products/industrial-cameras/usb-3.1-color/dfk37bux273/
`Dynamic range | 8/10 bit`

Stefan