TheImagingSource/tiscamera #211: Line incorrect?

blazini36, 2018-08-29:
https://github.com/TheImagingSource/tiscamera/blob/2fdabb21bc037243bdb0bbd8feb233d87dcf6b50/data/uvc-extensions/usb3.xml#L2010

This line is replicated from line 2024 and causes issues with it's value being set out of range

TIS-Stefan, 2018-08-29:
Hi

Did you write that by email to us too? Regarding default value being 6500 instead of something between 0 and 8?
I forwarded this to the programming team.

Stefan

blazini36, 2018-08-29:
I left a message on the site for support, then once I figured out the issue I noted it on GitHub. Lines 2010 and 2024 have the same control ref causing the values to swap.

I can’t say for certain that’s all that needs to be fixed but I’m sure it’s most of the issue.

TIS-Stefan, 2018-08-30:
Hello

The problem is, that you have two properties writing on the same register. Therefore, it looks a little bit strange regarding the values.
However, that is my knowledge about that. I forwarded this also to the programmers.

closed 2018-08-30 by TIS-Stefan