TheImagingSource/tiscamera #212: How to see which MTU is negotiated with the camera

yoelk, 2018-08-29:
In [this issue](https://github.com/TheImagingSource/tiscamera/issues/203) it was suggested to set an environment variable (`TCAM_LOG=DEBUG`) to see what MTU is negotiated between the camera and the PC.

I've tried it on my PC (Ubuntu 16.04), which is directly connected to the camera (DMK Z12GX236) via a dedicated ethernet port and these are the logs I'm getting:
```
2745215    <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
2745463    <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
3063898    <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
3064029    <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
42756      <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:147: The current value of Strobe Operation is NULL. Ignoring
42782      <ERROR> /src/tiscamera/src/aravis/AravisDevice.cpp:943: Property 'StrobeOperation' is null
3555342    <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
3555465    <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
58107      <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:147: The current value of Strobe Operation is NULL. Ignoring
58127      <ERROR> /src/tiscamera/src/aravis/AravisDevice.cpp:943: Property 'StrobeOperation' is null
3806784    <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
3806905    <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
4059822    <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
4059957    <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
4309900    <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
4310009    <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
4561764    <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
4561883    <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
4832107    <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
4832231    <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
5081794    <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
5081944    <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
5314348    <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
5314458    <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
```

So the questions I have:
1. Where can I see any information about the MTU? I've tried changing it while running a capture (showing the logs) but the camera feed got stuck and no MTU-related logs showed up
2. I keep getting an ERROR about the gige-daemon.lock file. Is this in order? If not, what is the reason and what can I do to remedy?
3. Please see other ERROR logs I'm getting (e.g. ` Property 'StrobeOperation' is null`). Are they ok as well?

Thanks,
Joel

yoelk, 2018-08-29:
Additional info:
I've tried setting the MTU to a few values, and checked the incoming packet sizes:
```
MTU = 1500 -> Packet size = 1630
MTU = 1550 -> Packet size = 1630
MTU = 1600 -> Packet size = 1630

MTU = 1700 -> Packet size = 1862
MTU = 2000 -> Packet size = 1862
MTU = 2300 -> Packet size = 1862
MTU = 9000 -> Packet size = 1862
```

This behavior suggests something isn't working well in the MTU negotiation process, or in the camera's MTU behavior. Or is this the planned behavior?
Ideally I'd like the packet sizes to be as big as possible for better network performance.

Thanks again,
Joel

TIS-Stefan, 2018-08-29:
Hello Joel

the maximum packet size supported by the camera is 2984 bytes. I do not know, what the 1862 means, but at least, the packet size negotiation is done by the Aravis library automatically in regard to the currently set MTU.

Stefan

TIS-Edgar, 2018-08-30:
To further Stefan's point TCAM_LOG will tell you the determined packet size. The log will contain a line similar to:

    Automatically set packet size to XXXX bytes

> I keep getting an ERROR about the gige-daemon.lock file. Is this in order? If not, what is the reason and what can I do to remedy?

Can you give me the output of:
`sudo systemctl status gige-daemon.service`
and
`ls -l /var/lock/gige-daemon.lock`

After installing the daemon should run automatically and the file should exist.
Do you experience lag when calling `tcam-ctrl -l`? Does `gige-daemon list` list your cameras?

> Property 'StrobeOperation' is null

This is something that should not happen. I am looking into it.

yoelk, 2018-09-02:
Thanks Edgar,

1. I think I'm missing something here.
How can I see these initialization prints you mention (e.g. `Automatically set packet size to XXXX bytes`)?
When I start the test code (with TCAM_LOG=DEBUG), the only logs I get are the ones I sent above.
I also tried setting this in the terminal and then reconnect the camera, but I don't see any logs on the screen. Where can I see them?

2. Here is the output of `sudo systemctl status gige-daemon.service`:
```
 ● gige-daemon.service - GigE indexing daemon
   Loaded: loaded (/lib/systemd/system/gige-daemon.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2018-09-01 15:02:49 CEST; 18h ago
 Main PID: 2257 (gige-daemon)
   CGroup: /system.slice/gige-daemon.service
           └─2257 /usr/bin/gige-daemon start --no-fork

Sep 01 15:02:49 joel-tinyinspektor systemd[1]: Started GigE indexing daemon.
```

3. Here is the output of `ls -l /var/lock/gige-daemon.lock` (it does exist):
```
-rw-r----- 1 root root 5 Sep  1 15:02 /var/lock/gige-daemon.lock
```

4. When I run `tcam-ctrl -l` the command prints immediately. But if the camera is not connected and then I connect it and keep calling `tcam-ctrl -l`, it only recognizes the camera after ~10 seconds. Is this what you meant? The same is with `gige-daemon list`. Also lists my camera and also takes ~10 seconds to update.

TIS-Edgar, 2018-09-03:
Concerning `The current value of Strobe Operation is NULL. Ignoring`:
This is an aravis bug. You can see the progress on that [here](https://github.com/AravisProject/aravis/issues/191).

1. `I think I'm missing something here.`
I am not sure what the both of us are missing here. After executing `export TCAM_LOG=DEBUG` you should get a complete log of the entire application run. You can see an example of this [here](https://gist.github.com/TIS-Edgar/c126b5621695a93c0dbc10a216f95ce5) for the command `gst-launch-1.0 tcambin serial=15510006 ! video/x-raw,format=BGRx,framerate=10/1 ! videoconvert ! ximagesink sync=false`.

2/3. Both are as expected. For some reason clients cannot connect to the daemon and query a device list. I am still on it.

4. The reason I am asking for this is to know if there is a connection issue to the gige-daemon or if there are other, bigger problems.
When executing tcam-ctrl -l with gige-daemon disabled you should get:
```
> TCAM_LOG=DEBUG tcam-ctrl -l
16970      <DEBUG> Logger:107: 
The following library versions are used:
	Tcam:	0.9.1_development/8d320e0_rev_2300
	Aravis:	0.6_version_5a2f97a
17767      <ERROR> /home/bvtest/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
17825      <ERROR> /home/bvtest/tiscamera/src/aravis/aravis_utils.cpp:465: Unable to connect to gige-daemon. Using internal methods
74260      <ERROR> /home/bvtest/tiscamera/src/aravis/aravis_utils.cpp:467: Aravis gave us 5
74375      <ERROR> /home/bvtest/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
74391      <ERROR> /home/bvtest/tiscamera/src/aravis/aravis_utils.cpp:465: Unable to connect to gige-daemon. Using internal methods
133460     <ERROR> /home/bvtest/tiscamera/src/aravis/aravis_utils.cpp:467: Aravis gave us 5
Available devices:
Model		Type	Serial

DMK Z12GX236	Aravis	08510294
DFK Z12GX236	Aravis	15510006
DMK 33GX264e	Aravis	23810274
DMK 33GX264e	Aravis	23810275
DFK 38GX304	Aravis	28819933
```

or something similar. Note that there is a waiting time between command start and command output. This is because tcam-ctrl has to manually query the network for devices (likely the ~10 seconds you are seeing). The reason the gige-daemon was introduced was to get rid of this waiting period by always having a local, up-to-date list of GigE devices.

yoelk, 2018-09-03:
1. I ran the following commands:
```
export TCAM_LOG=DEBUG
gst-launch-1.0 tcambin serial=20810358 ! video/x-raw,format=GRAY8,width=1920,height=1080,framerate=10/1 ! videoconvert ! ximagesink sync=false
```
and get this output (no image on the screen):
```
Setting pipeline to PAUSED ...
31795      <DEBUG> Logger:107: 
The following library versions are used:
	Tcam:	0.10.0_development/8d320e0_rev_2300
	Aravis:	0.6_version_5a2f97a
33630      <DEBUG> Logger:107: 
The following library versions are used:
	Tcam:	0.10.0_development/8d320e0_rev_2300
	Aravis:	0.6_version_5a2f97a
33665      <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
33809      <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
64949      <INFO> /src/tiscamera/src/aravis/AravisDevice.cpp:142: Automatically set packet size to 1616 bytes
65333      <WARNING> /src/tiscamera/src/aravis/aravis_utils.cpp:94: Unable to find std property. Passing raw property identifier through. WDR
65777      <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Trigger Selector
66025      <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Trigger Mode
66159      <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Trigger Source
66324      <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Trigger Activation
66723      <WARNING> /src/tiscamera/src/aravis/aravis_utils.cpp:94: Unable to find std property. Passing raw property identifier through. TimestampReset
66816      <WARNING> /src/tiscamera/src/aravis/aravis_utils.cpp:94: Unable to find std property. Passing raw property identifier through. TimestampLatch
66870      <WARNING> /src/tiscamera/src/aravis/aravis_utils.cpp:94: Unable to find std property. Passing raw property identifier through. TimestampLatchValue
67350      <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Offset Auto Center
68267      <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Strobe Enable
68418      <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Strobe Polarity
68554      <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:147: The current value of Strobe Operation is NULL. Ignoring
68575      <ERROR> /src/tiscamera/src/aravis/AravisDevice.cpp:943: Property 'StrobeOperation' is null
69043      <WARNING> /src/tiscamera/src/aravis/aravis_utils.cpp:94: Unable to find std property. Passing raw property identifier through. IRCutFilterEnableElement
69681      <DEBUG> /src/tiscamera/src/aravis/AravisDevice.cpp:1106: Adding format desc: Mono8 (30303859) 
69870      <DEBUG> /src/tiscamera/src/aravis/AravisDevice.cpp:1106: Adding format desc: Mono16 (20363159) 
70217      <DEBUG> /src/tiscamera/src/aravis/AravisDevice.cpp:183: Returning 31 properties
70268      <DEBUG> /src/tiscamera/src/PipelineManager.cpp:127: Received 2 formats.
70335      <DEBUG> /src/tiscamera/src/aravis/AravisDevice.cpp:183: Returning 31 properties
70473      <DEBUG> /src/tiscamera/src/PropertyHandler.cpp:273: Created 31 properties
70504      <DEBUG> Logger:107: 
The following library versions are used:
	Tcam:	0.10.0_development/8d320e0_rev_2300
	Aravis:	0.6_version_5a2f97a
71382      <INFO> /src/tiscamera/src/ImageSource.cpp:68: Source changed to state STOPPED
71448      <INFO> /src/tiscamera/src/aravis/AravisDevice.cpp:162: Destroying arvcamera
83545      <INFO> /src/tiscamera/src/CaptureDeviceImpl.cpp:157: Closed device DMK Z12GX236.
102607     <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
102722     <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
107934     <INFO> /src/tiscamera/src/aravis/AravisDevice.cpp:142: Automatically set packet size to 1616 bytes
108094     <WARNING> /src/tiscamera/src/aravis/aravis_utils.cpp:94: Unable to find std property. Passing raw property identifier through. WDR
108294     <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Trigger Selector
108412     <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Trigger Mode
108556     <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Trigger Source
108687     <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Trigger Activation
108909     <WARNING> /src/tiscamera/src/aravis/aravis_utils.cpp:94: Unable to find std property. Passing raw property identifier through. TimestampReset
108972     <WARNING> /src/tiscamera/src/aravis/aravis_utils.cpp:94: Unable to find std property. Passing raw property identifier through. TimestampLatch
109024     <WARNING> /src/tiscamera/src/aravis/aravis_utils.cpp:94: Unable to find std property. Passing raw property identifier through. TimestampLatchValue
109452     <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Offset Auto Center
110457     <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Strobe Enable
110623     <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Strobe Polarity
[... cut ...]
Setting pipeline to NULL ...
7003698    <INFO> /src/tiscamera/src/aravis/AravisDevice.cpp:162: Destroying arvcamera
7013861    <INFO> /src/tiscamera/src/CaptureDeviceImpl.cpp:157: Closed device DMK Z12GX236.
Freeing pipeline ...

```
I see there are missing packets. Can you please have a look and see if other log prints seem normal?

2/3. ok, waiting for updates.

4. So there seems to be something wrong with the GigE deamon in my PC, right?
Any idea how can I fix it? 

Thanks again for the support!
Joel

TIS-Arne, 2018-09-07:
@yoelk This packages are built from the latest development version. Could you please test them if they resolve the issues?

[tiscamera-build.zip](https://github.com/TheImagingSource/tiscamera/files/2361271/tiscamera-build.zip)

yoelk, 2018-09-09:
Thanks for the update!
Here is a log of what I've done:
```
sudo dpkg -i tiscamera_0.10.0_amd64.deb                            

(Reading database ... 317495 files and directories currently installed.)
Preparing to unpack tiscamera_0.10.0_amd64.deb ...
Removed symlink /etc/systemd/system/multi-user.target.wants/gige-daemon.service.
Unpacking tiscamera (0.10.0) over (0.10.0) ...
Setting up tiscamera (0.10.0) ...
Created symlink from /etc/systemd/system/multi-user.target.wants/gige-daemon.service to /lib/systemd/system/gige-daemon.service.
Processing triggers for gnome-menus (3.13.3-6ubuntu3.1) ...
Processing triggers for desktop-file-utils (0.22-1ubuntu5.1) ...
Processing triggers for bamfdaemon (0.5.3~bzr0+16.04.20160824-0ubuntu1) ...
Rebuilding /usr/share/applications/bamf-2.index...
Processing triggers for mime-support (3.59ubuntu1) ...
```

Then:
```
$ export TCAM_LOG=DEBUG
$ tcam-ctrl -l                                                                                                                                  

13957      <DEBUG> Logger:107: 
The following library versions are used:
	Tcam:	0.10.0_development/4a349b9_rev_2314
	Aravis:	0.6_version_5a2f97a
15129      <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
15164      <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:465: Unable to connect to gige-daemon. Using internal methods
24896      <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:467: Aravis gave us 1
24992      <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
25022      <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:465: Unable to connect to gige-daemon. Using internal methods
32635      <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:467: Aravis gave us 1
Available devices:
Model		Type	Serial

DMK Z12GX236	Aravis	20810358

```

and (sorry for the long log below):
```
$ gst-launch-1.0 tcambin serial=20810358 ! video/x-raw,format=GRAY8,width=1920,height=1080,framerate=10/1 ! videoconvert ! ximagesink sync=false
Setting pipeline to PAUSED ...
33419      <DEBUG> Logger:107: 
The following library versions are used:
	Tcam:	0.10.0_development/4a349b9_rev_2314
	Aravis:	0.6_version_5a2f97a
35168      <DEBUG> Logger:107: 
The following library versions are used:
	Tcam:	0.10.0_development/4a349b9_rev_2314
	Aravis:	0.6_version_5a2f97a
35202      <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
35232      <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:465: Unable to connect to gige-daemon. Using internal methods
45470      <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:467: Aravis gave us 1
45570      <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
45599      <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:465: Unable to connect to gige-daemon. Using internal methods
53562      <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:467: Aravis gave us 1
87563      <INFO> /src/tiscamera/src/aravis/AravisDevice.cpp:142: Automatically set packet size to 1616 bytes
87910      <WARNING> /src/tiscamera/src/aravis/aravis_utils.cpp:94: Unable to find std property. Passing raw property identifier through. WDR
88329      <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Trigger Selector
88571      <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Trigger Mode
88687      <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Trigger Source
88800      <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Trigger Activation
89163      <WARNING> /src/tiscamera/src/aravis/aravis_utils.cpp:94: Unable to find std property. Passing raw property identifier through. TimestampReset
89230      <WARNING> /src/tiscamera/src/aravis/aravis_utils.cpp:94: Unable to find std property. Passing raw property identifier through. TimestampLatch
89284      <WARNING> /src/tiscamera/src/aravis/aravis_utils.cpp:94: Unable to find std property. Passing raw property identifier through. TimestampLatchValue
89720      <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Offset Auto Center
90675      <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Strobe Enable
90799      <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Strobe Polarity
[... 2196 lines cut ...]

(gst-launch-1.0:10566): GStreamer-CRITICAL **: gst_structure_free: assertion 'structure != NULL' failed
751856     <INFO> /src/tiscamera/src/aravis/AravisDevice.cpp:162: Destroying arvcamera
764913     <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:467: Aravis gave us 1
764993     <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
765029     <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:465: Unable to connect to gige-daemon. Using internal methods
772198     <INFO> /src/tiscamera/src/CaptureDeviceImpl.cpp:157: Closed device DMK Z12GX236.
Freeing pipeline ...
Caught SIGSEGV
#0  0x00007f0f6928098d in pthread_join (threadid=139704011847424, 
#1  0x00007f0f670f9b97 in std::thread::join() ()
#2  0x00007f0f66b7f881 in tcam::DeviceIndex::~DeviceIndex() ()
#3  0x00007f0f68ee7ff8 in __run_exit_handlers (status=1, 
[... cut ...]
#5  0x00007f0f68ece837 in __libc_start_main (main=0x403130 <main>, argc=10, 
#6  0x0000000000403d19 in _start ()
Spinning.  Please run 'gdb gst-launch-1.0 10566' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
^C
```

Both times, I could see the image on the screen.
If you want me to run something else specifically, please tell.

I see there are still issues with `gige-daemon.lock`. Might it be that the package didn't install correctly?
I can't tell by looking at the versions (is there be a build date somewhere?)

TIS-Stefan, 2018-09-10:
Hello

Edgar is on holidays for two weeks and I try to take over again.

First of all, I do not really understand the problem, which is caused by my lack of knowing all about Linux. Sorry for that. Also the thread has become somewhat long-

>Please see other ERROR logs I'm getting (e.g. Property 'StrobeOperation' is null). Are they ok as well?`
I do not get this question. But I suggest to try the tcam-capture program and check, whether this property is there under "spcecial" tab.

> When I run tcam-ctrl -l the command prints immediately. But if the camera is not connected and then I connect it and keep calling tcam-ctrl -l, it only recognizes the camera after ~10 seconds. Is [...]

If the camera is set to DHCP and you have no DHCP server running on the network controller you use with the camera, then the camera tries to connect to a DHCP server for several seconds. 
In case, you use a persistent IP address in the camera and computer, the connection should be established much faster. 

Regarding the MTU: This becomes important only, in case you get frame drops, caused by missing packets. The Z12 series has no image memory on board, thus packet resend is impossible. The bigger network packets reduce network traffic and packet loss. However, as I wrote somewhere above, the possible packet size is bigger than the 1848 bytes Aravis negotiates.

Stefan

yoelk, 2018-09-10:
Thanks Stefan,
He sent me a new TIS build which I tested, so I sent him the test results.
Perhaps we should wait for Edgar to return and continue this thread.

Joel

TIS-Stefan, 2018-09-10:
Joel,
I would like to know, which is the blocking issue you encounter here. 
Stefan

TIS-Stefan, 2018-09-10:
Hi Joel

Thank you for the summary, that is really helpful.

> the gige-daemon doesn't seem to work correctly.
> As for your previous related question, we do open a DHCP server on the PC, so this is how the camera gets its address.

You can see, how fast the communication between camera, DHCP server is? This takes a few seconds. But from my experience with my "minidhcp" on my Windows computer, it is "felt"  5 seconds.

> The negotiated MTU seems lower than it should be.
Yes. Seems to be cause by Aravis.

> Some Aravis bug related to error logs about `Strobe.
I see. I tried this this morning with same camera mode and firmware version. I do not get these errors on my Ubuntu. However, Edgar said something about that, so lets wait for him.

> Generally look at the logs and see if they are as expected. As you see above, there are many logs titled CRITICAL, ERROR and WARNING. So I'd like to understand if something should be fixed or are these expected, to prevent future camera problems.

So it is a question of understand log output, but not a problem, that blocks working with the camera, correct?

yoelk, 2018-09-10:
Yes, none of these are currently blocking and we can work. But we want to pursue them to make sure we work in the best way with TIS cameras and to minimize future problems.

Thanks!

TIS-Stefan, 2018-09-10:
Ah, ok

So lets wait for two weeks, when Edgar is back.

Stefan

TIS-Edgar, 2018-09-25:
Hi Joel,

Sorry for keeping you waiting. I will try to address all concerns.

The gige-daemon problems should have disappeared with the latest packet, which you are running.
I will talk to @TIS-Arne and ask him to include the commit hash into the packet name that way you can see in the logs if the the corret library versions are used.

If in doubt you can always call `sudo dpkg -r tiscamera` to remove the libraries/tools before installing.
I assume the gige-daemon is running correctly?
Basically has the output of the queries in https://github.com/TheImagingSource/tiscamera/issues/212#issuecomment-417235035 changed?

All camera listings should be instant. If they are not there is a problem.

> The negotiated MTU seems lower than it should be.

Aravis tries to automatically determine the packet size. It does this by requesting test packets from the camera. The biggest packet size that gets through without being fragmented is used as the packet size. 
Since your logs state

    <INFO> /src/tiscamera/src/aravis/AravisDevice.cpp:142: Automatically set packet size to 1616 bytes

something is forcing a small packet size. What is your network setup like? Any switches, etc that might cause problems?
Is the packet size always 1616 bytes? 
It would be helpful to have two wireshark logs of two short camera streams to see what the negotiations look like and to see on what side the problem is.

You can send the logs to support@theimagingsource.com

> Some Aravis bug related to error logs about `Strobe.
I wanted to wait with integrating the aravis version update until a new aravis release is available.
Since aravis-0.6.0 is now available the next deb package you will get should not display that message any more.

Concerning the logs:

The criticals and warnings are certainly not good. I tried replicating them on multiple systems but to no avail. They are likely caused by the new meta information that are shipped with each buffer. I am looking into it.

-Edgar

yoelk, 2018-09-26:
Hi Edgar, and thanks for picking it up again,
I've reinstalled the version you sent on September 7th, and before that I made sure tiscamera is removed with `sudo dpkg -r tiscamera`.

I've then connected the camera, enabled DEBUG logs and ran `tcam-ctrl -l`. It takes a while and this is what I get:
```
➜  tiscamera-build tcam-ctrl -l         
9622       <DEBUG> Logger:107: 
The following library versions are used:
	Tcam:	0.10.0_development/4a349b9_rev_2314
	Aravis:	0.6_version_5a2f97a
10715      <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
10754      <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:465: Unable to connect to gige-daemon. Using internal methods
19619      <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:467: Aravis gave us 1
19712      <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
19748      <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:465: Unable to connect to gige-daemon. Using internal methods
24073      <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:467: Aravis gave us 1
Available devices:
Model		Type	Serial

DMK Z12GX236	Aravis	20810358
```
So it seems like the gige daemon is still not working correctly.

----------------------------------------
Regarding negotiated packet sizes, the camera is powered by its own power supply (so not via PoE) and is connected directly to my ethernet interface.
I've set the MTU in this interface to be 9000 (`sudo ip link set enp6s0 mtu 9000`) and ran a test console:
```
➜  tiscamera-build gst-launch-1.0 tcambin serial=20810358 ! video/x-raw,format=GRAY16_LE,width=1440,height=1080,framerate=18/1 ! videoconvert ! ximagesink sync=false
Setting pipeline to PAUSED ...
67131      <DEBUG> Logger:107: 
The following library versions are used:
	Tcam:	0.10.0_development/4a349b9_rev_2314
	Aravis:	0.6_version_5a2f97a
67852      <DEBUG> Logger:107: 
The following library versions are used:
	Tcam:	0.10.0_development/4a349b9_rev_2314
	Aravis:	0.6_version_5a2f97a
67868      <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
67884      <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:465: Unable to connect to gige-daemon. Using internal methods
74052      <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:467: Aravis gave us 1
74140      <ERROR> /src/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
74164      <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:465: Unable to connect to gige-daemon. Using internal methods
82788      <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:467: Aravis gave us 1
106050     <INFO> /src/tiscamera/src/aravis/AravisDevice.cpp:142: Automatically set packet size to 1848 bytes
106294     <WARNING> /src/tiscamera/src/aravis/aravis_utils.cpp:94: Unable to find std property. Passing raw property identifier through. WDR
106658     <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Trigger Selector
106875     <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Trigger Mode
107016     <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Trigger Source
107121     <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Trigger Activation
107396     <WARNING> /src/tiscamera/src/aravis/aravis_utils.cpp:94: Unable to find std property. Passing raw property identifier through. TimestampReset
107463     <WARNING> /src/tiscamera/src/aravis/aravis_utils.cpp:94: Unable to find std property. Passing raw property identifier through. TimestampLatch
107510     <WARNING> /src/tiscamera/src/aravis/aravis_utils.cpp:94: Unable to find std property. Passing raw property identifier through. TimestampLatchValue
108116     <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Offset Auto Center
109216     <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Strobe Enable
109370     <DEBUG> /src/tiscamera/src/aravis/aravis_utils.cpp:154: Returning EnumerationProperty for Strobe Polarity
109526     <ERROR> /src/tiscamera/src/aravis/aravis_utils.cpp:147: The current value of Strobe Operation is NULL. Ignoring
109554     <ERROR> /src/tiscamera/src/aravis/AravisDevice.cpp:945: Property 'StrobeOperation' is null
109890     <WARNING> /src/tiscamera/src/aravis/aravis_utils.cpp:94: Unable to find std property. Passing raw property identifier through. IRCutFilterEnableElement
110240     <DEBUG> /src/tiscamera/src/aravis/AravisDevice.cpp:1108: Adding format desc: Mono8 (30303859) 
110334     <DEBUG> /src/tiscamera/src/aravis/AravisDevice.cpp:1108: Adding format desc: Mono16 (20363159) 
110595     <DEBUG> /src/tiscamera/src/aravis/AravisDevice.cpp:183: Returning 31 properties
110663     <DEBUG> /src/tiscamera/src/PipelineManager.cpp:127: Received 2 formats.
110829     <DEBUG> /src/tiscamera/src/aravis/AravisDevice.cpp:183: Returning 31 properties
110952     <DEBUG> /src/tiscamera/src/PropertyHandler.cpp:273: Created 31 properties
110983     <DEBUG> Logger:107: 
The following library versions are used:
	Tcam:	0.10.0_development/4a349b9_rev_2314
	Aravis:	0.6_version_5a2f97a
111636     <INFO> /src/tiscamera/src/ImageSource.cpp:68: Source changed to state STOPPED
111703     <INFO> /src/tiscamera/src/aravis/AravisDevice.cpp:162: Destroying arvcamera
[... 403 lines cut ...]
[... cut ...]
#3  0x00007f8194155ff8 in __run_exit_handlers (status=1, 
#4  0x00007f8194156045 in __GI_exit (status=<optimized out>) at exit.c:104
#5  0x00007f819413c837 in __libc_start_main (main=0x403130 <main>, argc=10, 
#6  0x0000000000403d19 in _start ()
Spinning.  Please run 'gdb gst-launch-1.0 23069' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.

^C
```
It also took a while until I got an image, and after closing the window, I had to press ctrl-c to terminate it.
So as you see, still problems there as well.

Thanks for your help,
Joel

yoelk, 2018-09-26:
Oh, and one thing I forgot before:
Sometimes, it might happen that the driver doesn't start correctly (perhaps due to not closing it well in the last run).
It typically hangs in such situations on:
```
self.pipeline.set_state(Gst.State.PLAYING)
self.pipeline.get_state(Gst.CLOCK_TIME_NONE)
```

Is there some command to reset the driver state, so that it can start correctly? (other than reinstalling).
Thanks again,
Joel

TIS-Edgar, 2018-09-28:
Here is a new devel packet, that hopefully takes care of the criticals and the strobe issue:
[tiscamera_0.10.0.2341~development_b23050a_amd64.zip](https://github.com/TheImagingSource/tiscamera/files/2427931/tiscamera_0.10.0.2341.development_b23050a_amd64.zip)

Concerning the gige-daemon:
I have tried replicating the issue on multiple systems. On all fresh installs the issue does not seem to appear.

Could you try replicating the issue on a fresh install?
Does /var/lock maybe prevent readability?

Concerning MTU:

I will need a wireshark log. 

> Sometimes, it might happen that the driver doesn't start correctly (perhaps due to not closing it well in the last run).
> It typically hangs in such situations on:
> ``` [...]

What do you mean with reinstalling? GStreamer can hang itself under some circumstances. That can cause the application to hang. Other than that the camera itself could get enter an invalid state and thus not respond to requests. I would have to see the logs of such a case...

yoelk, 2018-09-30:
Concerning the new package: I couldn't install it. It throws an error:
```
sudo dpkg -i tiscamera_0.10.0.2321\~development_b23050a_amd64.deb
dpkg: error processing archive tiscamera_0.10.0.2321~development_b23050a_amd64.deb (--install):
 parsing file '/var/lib/dpkg/tmp.ci/control' near line 6 package 'tiscamera':
 'Depends' field, missing package name, or garbage where package name expected
Errors were encountered while processing:
 tiscamera_0.10.0.2321~development_b23050a_amd64.deb
```
Any idea?

About the rest, I'll write again soon

TIS-Edgar, 2018-10-01:
I'm terribly sorry, the generating script had a problem with the dependencies. This one should work. 
[tiscamera_0.10.0.2321~development_b23050a_amd64.zip](https://github.com/TheImagingSource/tiscamera/files/2432551/tiscamera_0.10.0.2321.development_b23050a_amd64.zip)

yoelk, 2018-10-01:
Thanks for the update!
I've just installed it and is seems that your dependencies have changed:
```
➜  dhcp_configuration git:(develop) tcam-ctrl -l         
tcam-ctrl: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.22' not found (required by /usr/lib/libtcam.so.0)
tcam-ctrl: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.22' not found (required by /usr/lib/libtcam-algorithms.so.0)
```

Did you build the package for Ubuntu 16.04?
If the dependencies have changed, please tell me which ones and I'll install them

Thanks!

TIS-Edgar, 2018-10-01:
I thought our build setup would automatically take care or library versioning. Seems like I was mistaken. This is built under 16.04 guaranteed.
[tiscamera_0.10.0.2322~development_d12419f_amd64.zip](https://github.com/TheImagingSource/tiscamera/files/2433538/tiscamera_0.10.0.2322.development_d12419f_amd64.zip)

yoelk, 2018-10-01:
this worked well, thanks!
I'll update on the other issues soon.

Joel

yoelk, 2018-10-02:
Hi again,
We've just found why we keep failing with the daemon.

Yesterday when I installed your .deb package, the daemon seemed to work perfect.
Today, I did again the following, and it made the daemon stop working properly:
```
sudo dpkg -r tiscamera 
sudo dpkg -i tiscamera_0.10.0.2322\~development_d12419f_amd64.deb 
```

It seems that uninstalling/reinstalling tiscamera creates the problem.
I've also tried now to purge the package and install it again, but the problem continued.
It occured to me that a reset might solve the problem and indeed it did.

Any idea why a reset is needed?

TIS-Edgar, 2018-10-02:
Hi Joel,

If this is true it means that there is an error in the instructions for systemd concerning starting/stopping the daemon.
I guess a reboot does not fix the issue?
Are there any useful information in the systemlog? `sudo journalctl -xe` should give all information for daemons, etc on your system.
I am trying to replicate it, but only semi successful. 
When you removed the package does /var/lock/gige-daemon.lock still exist?

yoelk, 2018-10-02:
reboot **did** help to solve the issue.
after removing the package, `/var/lock/gige-daemon.lock` still exists
Attached is my whole journal log:
[journalctl_log.txt](https://github.com/TheImagingSource/tiscamera/files/2437539/journalctl_log.txt)

This is before rebooting. The rebooting I told about was done in another PC.

TIS-Edgar, 2018-10-02:
Ok, it seems that systemd cannot shut down the service (for what ever reason), thus killing it. This causes the lock file to remain which causes later attempts to start the service to fail.

I have changed the systemd instructions in the following packet. This will hopefully remove this issues once and for all. Sorry this is taking so long.

[tiscamera_0.10.0.2322~development_d12419f_amd64.zip](https://github.com/TheImagingSource/tiscamera/files/2437713/tiscamera_0.10.0.2322.development_d12419f_amd64.zip)

yoelk, 2018-10-03:
Indeed, this sloves this problem. I've removed and reinstalled it, and the daemon still worked.
Thanks a lot for your continuous support!

yoelk, 2018-10-10:
Good morning Edgar,
Following up the latest version you gave us, there is an issue I need your help with.
You sent two packages of the same code (hash=d12419f):
A. on 2018/10/01
B. on 2018/10/02

version B solved the issue with the gigedaemon indeed.
The problem we're now seeing is that with version B, the framerate drastically decreased, due to many lost packets. As if the priority of the driver became lower and it isn't able to get enough run-time to deal with all of the packets. Could this have to do with the systemd instructions you changed in version B?

Attached are two log files, one when the MTU is set to 1500, and another where the MTU is set to 9000.
We thought it might be related but on both cases we get many missing frames.

[yoel_log_9000.txt](https://github.com/TheImagingSource/tiscamera/files/2464166/yoel_log_9000.txt)
[yoel_log_1500.txt](https://github.com/TheImagingSource/tiscamera/files/2464167/yoel_log_1500.txt)

Thanks a lot!
Joel

TIS-Edgar, 2018-10-11:
[tiscamera_0.10.0.2325~development_df15c44_amd64.deb.zip](https://github.com/TheImagingSource/tiscamera/files/2468606/tiscamera_0.10.0.2325.development_df15c44_amd64.deb.zip)

This package should fix the MTU related problems. The package size should now be between 8000 and 9000 bytes. Please let me know if this fixes the issue or if you can still observe it.

yoelk, 2018-10-11:
Thanks Edgar!
We'll test it the coming days.
I'm confused though about your comment regarding the MTU. In our cameras (DMK_Z12GX236) it's not possible to get more than ~2000 bytes MTU due to the lack of inner memory right? You probably related to other camera types.

yoelk, 2018-10-11:
Hi again,
I've just tested it and with this version, I can't get any frames from the camera.
When I run gst-launch-1.0, it simply hangs and I have to kill it with ctrl+c.
I've reinstalled the last version you gave us and it immediately solved the problem.
Please see the following terminal log:
```
➜  ~ gst-launch-1.0 tcambin serial=20810358 ! video/x-raw,format=GRAY8,width=1640,height=1080,framerate=18/1 ! videoconvert ! ximagesink sync=false
Setting pipeline to PAUSED ...
32144      <DEBUG> Logger:107: 
The following library versions are used:
	Tcam:	0.10.0_development/df15c44_rev_2325
	Aravis:	0.6_version_3891b1a
^C
```

Thanks,
Joel

TIS-Edgar, 2018-10-12:
You are right the package size maximum for the Z12GX236 is lower than 9000. 

The patch I included in the last debian package solved the package size negotiation for most cameras.
For others, it seems to have had the opposite effect....
I am looking into how to address it permanently in aravis. In the mean time I am going to prepare a workaround for you to enable higher package sizes. Expect that monday-tuesday. Sorry for the delay.

TIS-Edgar, 2018-10-16:
Ok, this package will exhibit the following behavior:
When you simply open the device you will get the old established behavior with a low packet size.
When you execute `export TCAM_GIGE_PACKET_SIZE=2984` before opening the device you will get a manual overwrite that ignores the auto negotiation.

Getting this permanently right is taking longer than anticipated, so this will at least allow you to work around this issue.

[tiscamera_0.10.0.2329~development_1e0b465_amd64.deb.zip](https://github.com/TheImagingSource/tiscamera/files/2482166/tiscamera_0.10.0.2329.development_1e0b465_amd64.deb.zip)

yoelk, 2018-10-16:
That's great Edgar! Thanks again!
We'll start using it right away and write back if we see any issues.

Joel

yoelk, 2018-10-21:
Hi Edgar,
As I wrote before, it works fine with my DMK_Z12GX236.
I apologize but I forgot to mention before, that we're also testing these versions on GIGE29_WUXGA_M_Z12 cameras.

It uses the same hardware but has this different name.
Could you please make this patch (with TCAM_GIGE_PACKET_SIZE) work for both of these camera types (DMK_Z12GX236 and GIGE29_WUXGA_M_Z12)?

Thanks and sorry for the hassle,
Joel

TIS-Edgar, 2018-10-22:
Hi Joel,

The patch is hardware independent and simply overwrites the packet size with your setting no matter what camera you are using.

- Edgar

yoelk, 2018-11-07:
Hello Edgar,
I've noticed that the extra tools are missing from the package (such as the tcam-gigetool for setting the camera's IP).
Could you please build it again with this flag? `BUILD_TOOLS:BOOL=ON`

Thanks a lot and have a nice day!
Joel

TIS-Edgar, 2018-11-07:
There you are: [tiscamera_0.10.0.2357~development_e3da47d_amd64.deb.zip](https://github.com/TheImagingSource/tiscamera/files/2556879/tiscamera_0.10.0.2357.development_e3da47d_amd64.deb.zip)

Seems like a change in the install instructions caused the gigetool to not be included. I apologize for the inconvenience.

yoelk, 2018-11-07:
Thanks Edgar!
I just tried it but couldn't find `tcam-gigetool` anywhere.
I see that `tcam-ctrl` is in /usr/bin but not `tcam-gigetool`.

I do see however `camera-ip-conf`. Is this the tool for setting IP addresses of camera? or is tcam-gigetool supposed to do that?

Thanks again,
Joel

yoelk, 2018-11-07:
btw, I see that `camera-ip-conf` can be used for setting the IP address, but that a power cycle is necessary.
Is it possible to somehow send a software reset instead?

Thanks again,
Joel

TIS-Edgar, 2018-11-07:
For some reason our installation tests missed the tcam-gigetool problems...
We have since discovered another issue with the tool which we will have to fix before publishing it, so I will have to ask for some time, so that we can provide a new package tomorrow.

camera-ip-conf and tcam-gigetool are somewhat similar. 
You would have to ask @TIS-Arne about the differences. 
Both are able to set the ip configuration of the camera.

Powercycling is necessary when uploading a new firmware.
When adjusting the IP settings, it is recommended to ensure that the new values are correct 
and everything behaves as expected (so that the camera will come up in an expected manner should a powerloss occur).

It is not possible to powercycle the camera via command. 
A rereading of the eeprom is possible. For this you would have to write a certain value to a certain register:
Register: 0xef00004  
Value: 0xb007b007

We currently do not have a tool that allows this. I will have to refer you to aravis directly. 
Their arv-tool allows writing registers.
Please be aware that many of our cameras do not have this reset feature. The DMK Z12GX236 is one of the few that has it.

yoelk, 2018-11-07:
ok, thanks for the info!
It's lucky that the DMK supports it (and I assume also the GIGE29_WUXGA_M_Z12).
I'll give it a try and see if I can avoid the power cycle.
We'd like to set the camera automatically, without user intervention.

Thanks, and waiting for your inputs regarding the tcam-gigetool,
Joel

yoelk, 2018-11-08:
@TIS-Arne
Hello Arne,
May I ask about the differences between the tcam-gigetool and the camera-ip-conf tool?
We've decided to try and work with static IP addresses instead of DHCP, and are trying to set the IP to the cameras.
The tcam-gigetool isn't available for me at the moment, as I could not compile it myself nor is it built with the tiscamera package (as Edgar mentioned).
Uisng the camera-ip-conf tool, I can only connect to cameras that already have an IP address, so DHCP is needed to define the static IP addess. Is this any different with the tcam-gigetool?
Is there a way to probe cameras before they have a same-subnet IP address?

Thanks, and sorry if it's a stupid question,
Joel

TIS-Arne, 2018-11-20:
@yoelk The tcam-gigetool is meant to supersede the camera-ip-conf tool. It will be part of the tiscamera package in the next release. Both tools should list all cameras that respond to broadcast messages and both tools are able to configure those cameras without the need of DHCP. It could be possible though that the response packets of the cameras could be filtered by the kernel RP filter so this one should be turned of when configuring the cameras.

closed 2019-11-29 by TIS-Stefan