TheImagingSource/tiscamera #213: uvcvideo:  errors when using cpp tcamcamera example with two devices connected

midjji, 2018-08-30:
Problem, various errors such as " No such property (Exposure)" 
stalls and hangs: Most common problem is frame_ready_callback never beeing called. 

dmesg 
[1899467.626716] uvcvideo: Failed to query (SET_CUR) UVC control 5 on unit 3: -110 (exp. 1).
[1899467.644886] uvcvideo: Failed to set UVC probe control : -71 (exp. 34).
[1899467.669377] uvcvideo: Non-zero status (-71) in video completion handler.

Cannot replicate with just one camera connected. 
Note the issue occurs even if just one TcanCamera is created
DMK33 UX252

midjji, 2018-08-30:
problem also occurs in the tcam-stereo-capture.zip example.

midjji, 2018-08-30:
if the cameras are disconnected and reconnected then the program will work, sometimes several times in a row for a few hundred to a few k images, after which one of them stalls while the other keeps working fine.

TIS-Stefan, 2018-08-30:
Could you please try lower frame rates for both cameras?

midjji, 2018-08-30:
previously: Framerate is 60hz at 640x480.
Cant replicate at 5hz

midjji, 2018-08-30:
Issue only occurs when two physically adjacent usb3 ports are used. 
I guess two underlying usb busses, makes it work, in this case even at 600hz 640x480 or 1920x1080x180.

BananaHemic, 2019-04-10:
Has this issue been fixed, or is the only solution to use two non-adjacent usb ports? I've been experiencing a very similar problem with my TIS cameras.

TIS-Stefan, 2019-04-10:
Hello

From reading this thread again, I think, it is an USB controller bandwidth issue. Running both cameras with more than 50% of USB bandwidth will lead to frame drops.
But not to the issue about "non such property". 
However, I do not know, which cameras and computer models are in use. I also do not know, whether there is an extra USB hub in use and whether it is powered.

Therefore, for a may be better diagnosis:
- Which computer model is in use?
- Wich camera models are in use?
- Which video formats are in use?
- Does dmesg output indicate an error?

Thank you in advance.

Stefan

midjji, 2019-04-14:
Partially, while reducing load can alleviate the problem there are still
strange behaviours if the computer load is high. Regardless of specific
load on the usb buss.

There is also the issue of disconnecting the cameras. If you only have one
it's very likely to work to run the demo programs over and over, but if you
simultaneously connect to both, every three times or so, you'll need to
physically disconnect and reconnect them or they will hang on connect.

midjji, 2019-04-14:
The exposure property issue seems to be operating system and tcam version
dependent.

DMK 33ux252.
Connection problem is common to all frame rates and resolutions. Regardless
of shared controller or separate line.
No error in dmesg in general.

 Adjacent ports is irrelevant but if the two physical ports correspond to
one line/controller, there is trouble. You can check this in linux.

TIS-Stefan, 2019-04-15:
Please please do not let me guess into the blue! 

Which computer model do you use?

TIS-Arne, 2019-04-17:
@midjji 
Can you provide a full dmesg output at the time the error occurs? This would be very helpful.

midjji, 2019-04-17:
dmesg1 is cleared, then connect and it works fine.
dmesg2 is shut down program, then start it again and it stalls on connect.

[435952.186047] usb 4-1: new SuperSpeed USB device number 4 using xhci_hcd
[435952.208024] usb 4-1: LPM exit latency is zeroed, disabling LPM.
[435952.211273] usb 4-1: New USB device found, idVendor=199e, idProduct=9005
[435952.211279] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[435952.211282] usb 4-1: Product: DMK 33UX252
[435952.211286] usb 4-1: Manufacturer: The Imaging Source Europe GmbH
[435952.211289] usb 4-1: SerialNumber: 27810458
[435952.213127] uvcvideo: Found UVC 1.10 device DMK 33UX252 (199e:9005)
[435952.216958] uvcvideo 4-1:1.3: Entity type for entity Extension 3 was not initialized!
[435952.216963] uvcvideo 4-1:1.3: Entity type for entity Processing 2 was not initialized!
[435952.216966] uvcvideo 4-1:1.3: Entity type for entity Camera 1 was not initialized!
[435957.910418] usb 4-2: new SuperSpeed USB device number 5 using xhci_hcd
[435957.932294] usb 4-2: LPM exit latency is zeroed, disabling LPM.
[435957.935439] usb 4-2: New USB device found, idVendor=199e, idProduct=9005
[435957.935445] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[435957.935449] usb 4-2: Product: DMK 33UX252
[435957.935453] usb 4-2: Manufacturer: The Imaging Source Europe GmbH
[435957.935457] usb 4-2: SerialNumber: 27810456
[435957.937617] uvcvideo: Found UVC 1.10 device DMK 33UX252 (199e:9005)
[435957.941682] uvcvideo 4-2:1.3: Entity type for entity Extension 3 was not initialized!
[435957.941688] uvcvideo 4-2:1.3: Entity type for entity Processing 2 was not initialized!
[435957.941692] uvcvideo 4-2:1.3: Entity type for entity Camera 1 was not initialized!
[435970.958291] usb 1-5: new full-speed USB device number 59 using xhci_hcd
[435971.106724] usb 1-5: New USB device found, idVendor=16c0, idProduct=0483
[435971.106730] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[435971.106734] usb 1-5: Product: USB Serial
[435971.106738] usb 1-5: Manufacturer: Teensyduino
[435971.106742] usb 1-5: SerialNumber: 5339770
[435971.120357] cdc_acm 1-5:1.0: ttyACM0: USB ACM device
[435971.120498] usbcore: registered new interface driver cdc_acm
[435971.120499] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[436060.101226] uvcvideo: Non-zero status (-71) in video completion handler.
[436072.068827] xhci_hcd 0000:04:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.

[435952.186047] usb 4-1: new SuperSpeed USB device number 4 using xhci_hcd
[435952.208024] usb 4-1: LPM exit latency is zeroed, disabling LPM.
[435952.211273] usb 4-1: New USB device found, idVendor=199e, idProduct=9005
[435952.211279] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[435952.211282] usb 4-1: Product: DMK 33UX252
[435952.211286] usb 4-1: Manufacturer: The Imaging Source Europe GmbH
[435952.211289] usb 4-1: SerialNumber: 27810458
[435952.213127] uvcvideo: Found UVC 1.10 device DMK 33UX252 (199e:9005)
[435952.216958] uvcvideo 4-1:1.3: Entity type for entity Extension 3 was not initialized!
[435952.216963] uvcvideo 4-1:1.3: Entity type for entity Processing 2 was not initialized!
[435952.216966] uvcvideo 4-1:1.3: Entity type for entity Camera 1 was not initialized!
[435957.910418] usb 4-2: new SuperSpeed USB device number 5 using xhci_hcd
[435957.932294] usb 4-2: LPM exit latency is zeroed, disabling LPM.
[435957.935439] usb 4-2: New USB device found, idVendor=199e, idProduct=9005
[435957.935445] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[435957.935449] usb 4-2: Product: DMK 33UX252
[435957.935453] usb 4-2: Manufacturer: The Imaging Source Europe GmbH
[435957.935457] usb 4-2: SerialNumber: 27810456
[435957.937617] uvcvideo: Found UVC 1.10 device DMK 33UX252 (199e:9005)
[435957.941682] uvcvideo 4-2:1.3: Entity type for entity Extension 3 was not initialized!
[435957.941688] uvcvideo 4-2:1.3: Entity type for entity Processing 2 was not initialized!
[435957.941692] uvcvideo 4-2:1.3: Entity type for entity Camera 1 was not initialized!
[435970.958291] usb 1-5: new full-speed USB device number 59 using xhci_hcd
[435971.106724] usb 1-5: New USB device found, idVendor=16c0, idProduct=0483
[435971.106730] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[435971.106734] usb 1-5: Product: USB Serial
[435971.106738] usb 1-5: Manufacturer: Teensyduino
[435971.106742] usb 1-5: SerialNumber: 5339770
[435971.120357] cdc_acm 1-5:1.0: ttyACM0: USB ACM device
[435971.120498] usbcore: registered new interface driver cdc_acm
[435971.120499] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[436060.101226] uvcvideo: Non-zero status (-71) in video completion handler.
[436072.068827] xhci_hcd 0000:04:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[436096.091672] xhci_hcd 0000:04:00.0: WARN Event TRB for slot 3 ep 2 with no TDs queued?
[436096.696537] uvcvideo: Non-zero status (-71) in video completion handler.

TIS-Arne, 2019-04-18:
@midjji 
These lines:

`[436072.068827] xhci_hcd 0000:04:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[436096.091672] xhci_hcd 0000:04:00.0: WARN Event TRB for slot 3 ep 2 with no TDs queued?`

suggest there are issues with your USB host controller, either with the hardware or with the driver. 

Googling the log message brings up numerous reports, for example this Ubuntu bug report:
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1749961

There does not seem to be a definite fix for this other than to try a more recent kernel. It does not look like the root cause is in the tiscamera software though. 

The issue with the USB controller is unlikely to lead to missing properties though. I think this should only lead to hangs/stalls, ie. the new_frame_callback not being called. If you still have the issue of missing properties, it might be helpful to set the following environment variable before launching your application:

`export TCAM_LOG=DEBUG`

This will cause the application to create extensive debug output. Please post the output here so we can investigate.

midjji, 2019-04-22:
I cannot currently reproduce the change of available properties.
I'll try to update to the latest kernels and see what happens, its not a
real solution though,
 as I need cuda in the same program and that does not play nice with all
kernel versions.

closed 2019-11-29 by TIS-Stefan