TheImagingSource/tiscamera #214: [DFK33UX264] Can't get full manual operation

blazini36, 2018-09-02:
Using these cameras in a custom program under linux controlling them with v4l2-ctl. Due to the overlapping menu items and some that don't seem to have an "off" option I get some automatic intervention. For example:   

whitebalance_temperature_preset has 8 menu items for presets and one must be selected, can't find a controlling parameter to just use the whitebalance_temperature integer instead. I'm able to see dynamic changes with the camera through my GUI and with the current settings changing just the exposure_time_us causes the whitebalance components to automatically change.
Current settings:
 ~ $ v4l2-ctl -d /dev/topcam -L
                     brightness (int)    : min=0 max=4095 step=1 default=240 value=240
                     saturation (int)    : min=0 max=255 step=1 default=64 value=64
                            hue (int)    : min=-180 max=180 step=1 default=0 value=0
   white_balance_component_auto (bool)   : default=1 value=0
    white_balance_red_component (int)    : min=0 max=255 step=1 default=64 value=120
   white_balance_blue_component (int)    : min=0 max=255 step=1 default=64 value=134
                          gamma (int)    : min=1 max=500 step=1 default=100 value=100
                           gain (int)    : min=0 max=480 step=1 default=0 value=0
                  exposure_auto (menu)   : min=0 max=3 default=3 value=1
				1: Manual Mode
				3: Aperture Priority Mode
              exposure_absolute (int)    : min=1 max=40000 step=1 default=3 value=1
               exposure_time_us (int)    : min=25 max=4000000 step=1 default=333 value=104
                   auto_shutter (bool)   : default=1 value=0
        auto_exposure_reference (int)    : min=0 max=255 step=1 default=128 value=128
                      gain_auto (bool)   : default=1 value=0
                   trigger_mode (bool)   : default=0 value=0
               software_trigger (button) : flags=write-only
                  trigger_delay (int)    : min=0 max=1000000 step=1 default=0 value=31
                  strobe_enable (bool)   : default=0 value=0
                strobe_polarity (bool)   : default=0 value=0
                strobe_exposure (bool)   : default=1 value=1
                strobe_duration (int)    : min=0 max=32767 step=1 default=100 value=100
                   strobe_delay (int)    : min=0 max=32767 step=1 default=0 value=0
                          gpout (bool)   : default=0 value=0
                           gpin (bool)   : default=0 value=0
                   roi_offset_x (int)    : min=0 max=2192 step=2 default=0 value=0
                   roi_offset_y (int)    : min=0 max=2044 step=2 default=0 value=0
                roi_auto_center (bool)   : default=1 value=0
               trigger_polarity (menu)   : min=0 max=1 default=0 value=0
				0: RisingEdge
				1: FallingEdge
          trigger_exposure_mode (menu)   : min=0 max=1 default=0 value=0
				0: FrameStart
				1: ExposureActive
            trigger_burst_count (int)    : min=1 max=1000 step=1 default=1 value=1
       trigger_debounce_time_us (int)    : min=0 max=1000000 step=1 default=0 value=0
           trigger_mask_time_us (int)    : min=0 max=1000000 step=1 default=0 value=0
  trigger_noise_supression_time (int)    : min=0 max=1000000 step=1 default=0 value=0
     auto_functions_roi_control (bool)   : default=1 value=0
        auto_functions_roi_left (int)    : min=0 max=2432 step=1 default=0 value=0
         auto_functions_roi_top (int)    : min=0 max=2032 step=1 default=0 value=0
       auto_functions_roi_width (int)    : min=16 max=2448 step=1 default=0 value=16
      auto_functions_roi_height (int)    : min=16 max=2048 step=1 default=0 value=16
              whitebalance_mode (menu)   : min=0 max=1 default=0 value=0
				0: Gray World
				1: Temperature
      white_balance_auto_preset (menu)   : min=0 max=4 default=0 value=0
				0: Any
				1: Warm White
				2: Cool White
				3: Daylight
				4: Deep Shade
  white_balance_green_component (int)    : min=0 max=255 step=1 default=64 value=64
whitebalance_temperature_preset (menu)   : min=0 max=8 default=6500 value=7
				0: Sodium-Vapur Lamp
				1: Warm Light
				2: Halogen Light
				3: Neutral White
				4: Cool White Fluorescent Light
				5: Cool White/Daylight (CFL)
				6: Daylight
				7: Cool White LED
				8: Blue Sky
      white_balance_temperature (int)    : min=2500 max=10000 step=100 default=6500 value=7500
                flip_horizontal (bool)   : default=0 value=0
                  flip_vertical (bool)   : default=0 value=0
            highlight_reduction (bool)   : default=0 value=0
 exposure_auto_upper_limit_auto (bool)   : default=1 value=1
      exposure_auto_lower_limit (int)    : min=25 max=1000000 step=1 default=333 value=333
      exposure_auto_upper_limit (int)    : min=25 max=1000000 step=1 default=333 value=333
         override_scanning_mode (int)    : min=1 max=16 step=1 default=1 value=1
      auto_functions_roi_preset (menu)   : min=0 max=5 default=1 value=1
				0: Full Sensor
				1: Center 50%
				2: Center 25%
				3: Bottom Half
				4: Top Half
				5: Custom Rectangle

TIS-Stefan, 2018-09-03:
Hello

I use the latest tcam-capture from master branch in Ubuntu 18.04 in order to reproduce your issue. The program offers Whitebalance Temperatur as integer values as well as White Balance Temperature Preset.

The integer value for the temperature is listed as
> white_balance_temperature (int) : min=2500 max=10000 step=100 default=6500 value=7500
Manual white balance can be set only, if the white balance automatic has been disable, but I guess, you know that already. 
These are all white balance parameters shown on my computer:
`v4l2-ctl --all | grep white
   white_balance_component_auto 0x0098090c (bool)   : default=1 value=0
    white_balance_red_component 0x0098090e (int)    : min=0 max=255 step=1 default=64 value=88
   white_balance_blue_component 0x0098090f (int)    : min=0 max=255 step=1 default=64 value=201
              whitebalance_mode 0x0199e246 (menu)   : min=0 max=1 default=0 value=0
      white_balance_auto_preset 0x0199e247 (menu)   : min=0 max=4 default=0 value=0
  white_balance_green_component 0x0199e248 (int)    : min=0 max=255 step=1 default=64 value=64
whitebalance_temperature_preset 0x0199e249 (menu)   : min=0 max=8 default=6500 value=3
      white_balance_temperature 0x0199e250 (int)    : min=2500 max=10000 step=100 default=6500 value=3500
`
Can you give some more details about your issue or a method, how to reproduce it?

Stefan

blazini36, 2018-09-03:
I think that the (menu) items are just not labeled intuitively but if you have the settings just as you have above in tcam-capture and you move the "white_balance_red_component" slider it will revert back to the original setting even though "white_balance_component_auto" is off=0. 

I figured out that "whitebalance_mode" must be set to 1=temperature as 0=grey world is some sort of auto function as well. In tcam capture the "whitebalance" toggle controls this.

To me it seems that  "white_balance_auto_preset" menu=0 should be labeled "none or off" rather than "any" and "whitebalance_mode" menu=0 should have a name that implies it automatically controls the whitebalance components rather than "gray world"

BTW I just found tcam capture for linux the other day, it's good to see that people are spending time on Linux software, it works well.

blazini36, 2018-09-03:
The preset does not have an effect on this. It seemed that Whitebalance_mode had to be set to 1=temperature but trying to replicate that does not seem to be the case now.

I'm checking for inconsistencies and trying to replicate the issue I was having earlier and I can't. The one thing I am noticing while trying to replicate is that in tcam-capture if you toggle "Exposure auto" it works as it should. But if you go into the terminal and check "v4l2-ctl -d /dev/video0 -L" it does not toggle "exposure_auto" as would be expected. That parameter lists:
 exposure_auto (menu)   : min=0 max=3 default=3 value=1
				1: Manual Mode
				3: Aperture Priority Mode
Setting it to either mode in terminal does not make a difference in tcam capture as it would in guvcview so I'm not sure what to make of this. tcam may set exposure_auto in some other way as may be the case with some other settings so I'm not sure it's good to use tcam capture as a reference for controlling the cameras completely through v4l2-ctl.

TIS-Stefan, 2018-09-04:
`Do we agree that there are some inconsistencies regarding this matter in the usb3.xml file?`

Yes.

closed 2018-09-04 by TIS-Stefan