TheImagingSource/tiscamera #215: problem in reading a frame in tcam_demo.cpp

moonanjum26, 2018-09-04:
OS: Ubuntu 18.04
Camera type: DMK 23U618
test code: https://github.com/TheImagingSource/tiscamera/blob/master/examples/cpp/examples/tcam_demo.cpp.
1) I built and ran the code and it was working fine. The problem is that I just want to count the number of frames it has read in 60 seconds and save each frame to disk using this example, but I am not getting the results. I have searched for the documentation but it saying to use GRABBER library. I don't know how to import that library in C++ code. Can you please share a sample code to read and write frames in C++. 
2) We have IC utility in windows for TIS camera , but there is no such utility for Linux systems. If we want to check at how much FPS our camera is acquiring frames and writing to disk, how can we check that without 
[image]
 IC utility?
3) If we are unable to attain the 120 FPS , is there any inbuilt function which can adjust the same using C++?
I have attached my sample code which I was trying to count the frames in 60 seconds.

Thanks in advance

TIS-Stefan, 2018-09-04:
Hello

The Windows libraries differ extremely from the approach we us in Linux. In Linux, we simply use, what is already there, e.g GStreamer.
I suppose, the following program does, what you want to do
[tcamopencvsaveimage.zip](https://github.com/TheImagingSource/tiscamera/files/2348979/tcamopencvsaveimage.zip)
It is based on the tcamcamera class of our tiscamera repository. There is only a loop taking 10 images, but I think you can enhance this.
Saving images to hard disc takes far more time, then receiving them from the camera. Therefore, for speed measurement, you better do not save the images and comment out the saving code.

Stefa

moonanjum26, 2018-09-05:
Thanks for the reply. I tried the sample code which you have given but it is reading the frames at 60fps maximum. The camera specification says at 120fps. I have changed the following command
cam.set_capture_format("BGRx", FrameSize{640,480}, FrameRate{120,1}); to make it read at 120 fps,but it is not working, and not even for 65 fps. 
Can I get a sample python code also so that I can differentiate between the two languages which reads or writes fast.

TIS-Stefan, 2018-09-05:
Hello

does tcam-ctrl -c <serial> offer 120 fps? If yes, the camera is connected to USB 3.0. If 60 fps is the maximum, the camera is connected to an USB 2.0 controller or the camera thinks, it is connected to an USB 2.0 controller
You can also enter "lsusb -t" in a terminal and check, where the camera is connected too. 480M indicates USB 2.0, 5000M USB 3.

moonanjum26, 2018-09-05:
Thanks !. Its working now. But, still it is reading at 117fps even setting it to 120 fps. I checked "lsusb -t" it is connected to USB 3 now. 
Can I get a sample code for python as well to read and write the frames.

Thanks in advance

TIS-Stefan, 2018-09-06:
Hello

Frame drops can happen, due to high CPU load, incomplete image data transfer and so on.

A python sample should be available at 
https://github.com/TheImagingSource/tiscamera/blob/master/examples/python/opencv.py

Stefan

closed 2018-09-06 by TIS-Stefan