TheImagingSource/tiscamera #216: Ubuntu crash in our software, related to tcam-ctrl

yoelk, 2018-09-09:
- OS: Ubuntu 16.04
- python: 3.6
- TIS version: 0.9.1
- TIS version source: tag `v-tiscamera-0.10.0` from this GitHub
- Camera type: DMK Z12GX236

We're running our own software (Can't share code unfortunately) which uses a TIS camera to stream video on the screen, and sometimes set camera properties (Zoom, Focus, etc.).

We ran the software for a long time and it crashed out of the blue.
An Ubuntu crash report was created, signaling that it's a TIS-related issue. Perhaps a memory leak.
It seems that the offending command was `tcam-ctrl -p 20810357`, which in other cases 

Can someone please have a look?
Attached are two screenshots. The crash file's content (without the core-dump) can be seen here:
```
ProblemType: Crash
Architecture: amd64
CrashCounter: 1
CurrentDesktop: Unity
Date: Thu Sep  6 16:10:08 2018
DistroRelease: Ubuntu 16.04
ExecutablePath: /usr/bin/tcam-ctrl
ExecutableTimestamp: 1535892322
ProcCmdline: tcam-ctrl -p 20810357
ProcCwd: /home/igor/Desktop/logs_for_TIS_detection_issues
ProcEnviron:
 LC_PAPER=he_IL.UTF-8
 LC_ADDRESS=he_IL.UTF-8
 LC_MONETARY=he_IL.UTF-8
 TERM=xterm-256color
 SHELL=/bin/bash
 LD_PRELOAD=<set>
 LC_NUMERIC=he_IL.UTF-8
 LD_LIBRARY_PATH=<set>
 LC_TELEPHONE=he_IL.UTF-8
 PATH=(custom, user)
 LC_IDENTIFICATION=he_IL.UTF-8
 LANG=en_US.UTF-8
 LC_MEASUREMENT=he_IL.UTF-8
 LANGUAGE=en_US
 XDG_RUNTIME_DIR=<set>
 LC_TIME=he_IL.UTF-8
 LC_NAME=he_IL.UTF-8
ProcMaps:
 00400000-004b4000 r-xp 00000000 103:02 12858193                          /usr/bin/tcam-ctrl
 006b3000-006b4000 r--p 000b3000 103:02 12858193                          /usr/bin/tcam-ctrl
 006b4000-006b5000 rw-p 000b4000 103:02 12858193                          /usr/bin/tcam-ctrl
 01ac9000-01b4c000 rw-p 00000000 00:00 0                                  [heap]
 7f239c000000-7f239c021000 rw-p 00000000 00:00 0
 7f239c021000-7f23a0000000 ---p 00000000 00:00 0
 7f23a4000000-7f23a4021000 rw-p 00000000 00:00 0
 7f23a4021000-7f23a8000000 ---p 00000000 00:00 0
 7f23a8d46000-7f23a8d47000 ---p 00000000 00:00 0
 7f23a8d47000-7f23a9547000 rw-p 00000000 00:00 0
 7f23a9547000-7f23a9548000 ---p 00000000 00:00 0
 7f23a9548000-7f23a9d48000 rw-p 00000000 00:00 0
 7f23a9d48000-7f23a9d49000 ---p 00000000 00:00 0
 7f23a9d49000-7f23aa549000 rw-p 00000000 00:00 0
 7f23ac000000-7f23ac0ba000 rw-p 00000000 00:00 0
 7f23ac0ba000-7f23b0000000 ---p 00000000 00:00 0
 7f23b0576000-7f23b058d000 r-xp 00000000 103:02 11538919                  /lib/x86_64-linux-gnu/libresolv-2.23.so
 7f23b058d000-7f23b078d000 ---p 00017000 103:02 11538919                  /lib/x86_64-linux-gnu/libresolv-2.23.so
 7f23b078d000-7f23b078e000 r--p 00017000 103:02 11538919                  /lib/x86_64-linux-gnu/libresolv-2.23.so
 7f23b078e000-7f23b078f000 rw-p 00018000 103:02 11538919                  /lib/x86_64-linux-gnu/libresolv-2.23.so
 7f23b078f000-7f23b0791000 rw-p 00000000 00:00 0
 7f23b0791000-7f23b07b0000 r-xp 00000000 103:02 11539059                  /lib/x86_64-linux-gnu/libselinux.so.1
 7f23b07b0000-7f23b09af000 ---p 0001f000 103:02 11539059                  /lib/x86_64-linux-gnu/libselinux.so.1
 7f23b09af000-7f23b09b0000 r--p 0001e000 103:02 11539059                  /lib/x86_64-linux-gnu/libselinux.so.1
 7f23b09b0000-7f23b09b1000 rw-p 0001f000 103:02 11539059                  /lib/x86_64-linux-gnu/libselinux.so.1
 7f23b09b1000-7f23b09b3000 rw-p 00000000 00:00 0
[... 485 lines cut ...]
         atfct = <optimized out>
         onfct = <optimized out>
         cxafct = <optimized out>
 #4  0x00007f23b3e3a045 in __GI_exit (status=<optimized out>) at exit.c:104
 No locals.
 #5  0x00007f23b3e20837 in __libc_start_main (main=0x457f7d <main>, argc=3, argv=0x7ffeb71a9818, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7ffeb71a9808) at ../csu/libc-start.c:325
         result = <optimized out>
         unwind_buf = {cancel_jmp_buf = {{jmp_buf = {0, 3632004766945694828, 4552848, 140731970394128, 0, 0, -3631564102908586900, -3737954424815692692}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x3, 0x457f7d <main>}, data = {prev = 0x0, cleanup = 0x0, canceltype = 3}}}
         not_first_call = <optimized out>
 #6  0x00000000004578b9 in _start ()
 No symbol table info available.
 .
 Thread 1 (Thread 0x7f23b2b0a700 (LWP 25856)):
 #0  0x00007f23b19af5b5 in ?? ()
 No symbol table info available.
 #1  0x0000000000000000 in ?? ()
 No symbol table info available.
Title: tcam-ctrl crashed with SIGSEGV
UnreportableReason: This package does not seem to be installed correctly
UpgradeStatus: No upgrade log present (probably fresh install)
_MarkForUpload: True
```

[image]
[image]

TIS-Stefan, 2018-09-10:
Hello

The error message seems to be "stack memory exhausted".

I need to know, which steps are necessary to reproduce the issue. What does your software do?

Thank you in advance!

Stefan

yoelk, 2018-09-10:
Perhaps it's a memory issue not related to TIS, so closing for now

closed 2018-09-10 by yoelk