TheImagingSource/tiscamera #217: erroneous pipeline: no element "tcambin"

rbrown-acertaralabs, 2018-09-20:
1. Running Debian 9.5 (stretch)
2. Followed all instructions at https://github.com/theimagingsource/tiscamera
3. Build appears successful...see screenshot below.
4. Trying `gst-launch-1.0 tcambin ! videoconvert ! ximagesink` as indicated at URL from (2) above -- both as normal user and as sudo.
5. Getting error message:  `erroneous pipeline: no element "tcambin"`     (see screenshot below)
6. Note: I have NO cameras connected to my system...I am simply checking if the built binaries are running...I figured if they run with no cameras they would at least execute and report "no cameras found" or something along those lines.  Is that true?  

[image]

TIS-Stefan, 2018-09-21:
Hi 

Unfortunately you did not follow the instructions completele.
You forgot the "make" and the "sudo make install" step.

Stefan

rbrown-acertaralabs, 2018-09-21:
My fault...I overlooked that.  Thank you!

closed 2018-09-21 by rbrown-acertaralabs