TheImagingSource/tiscamera #219: Empty property list DMK33GP2000e

Shadberrow, 2018-09-23:
Hello!

I'am trying to use DMK33GP2000e camera with NanoPi NEO Plus2.
NanoPi running Ubuntu 16.04.5 LTS Xenial.
By executing command  `tcam-ctrl -l` I've got list of connected cameras successfully:
```
Available devices:
Model           Type    Serial

DMK 33GP2000e   Aravis  17714168
```
or `tcam-gigetool list`
```
------------------------ Interface: eth0 [192.168.2.199] ------------------------

     Model Name      |  Serial Number   |  User Defined Name   |     Current IP
DMK 33GP2000e        | 17714168         | ?                    | 169.254.220.143
```

But when I'am trying to get list of available properties using  `tcam-ctrl -p 17714168` I got:
```
Found 0 propert(y/ies)
```
or `tcam-ctrl -f 17714168` got empty list:
```
Available format settings:
```

I updated my NanoPi using `sudo apt-get update`, reinstalled tiscamera followed by instructions in README.md file and additionally installed GStreamer from [gstreamer.freedesktop.org](https://gstreamer.freedesktop.org/documentation/installing/on-linux.html)

Please suggest me something. Thanks!

TIS-Stefan, 2018-09-24:
Hello

If your network controller has IP 192.168.2.199 and the camera has IP 169.254.220.143, then you wont be able to get a video from it, because the IP addresses do not match.
"69.254.220.143" indicates, the camera runs on DHCP, did not find a DHCP server and thus used a auto link local IP address.
I suggest to give the camera a matching fixed IP address 192.168.2.n where n can be a value of 1-198 and 200-255.
Some more details can be found at https://s1-dl.theimagingsource.com/api/2.5/packages/publications/whitepapers-cameras/wpgigecamerastartup/4df6c9cb-9fb5-50d0-8b42-82259fd2124b/wpgigecamerastartup_1.0.en_US.pdf

Stefan

Shadberrow, 2018-09-24:
Oh, I see. 
Thank you so much, now it works.

closed 2018-09-24 by Shadberrow

Shadberrow, 2018-09-25:
Hello!

Now using `tcam-ctrl -p <serial>` I can get a list of camera properties, but before this I got a list of messages like this:
```
[  417.855585] dwmac-sun8i 1c30000.ethernet eth0: len 1982 larger than size (1536)
[  417.876524] dwmac-sun8i 1c30000.ethernet eth0: len 1982 larger than size (1536)
[  417.895585] dwmac-sun8i 1c30000.ethernet eth0: len 1982 larger than size (1536)
[  417.915333] dwmac-sun8i 1c30000.ethernet eth0: len 1750 larger than size (1536)
[  417.934401] dwmac-sun8i 1c30000.ethernet eth0: len 1750 larger than size (1536)
[  417.954877] dwmac-sun8i 1c30000.ethernet eth0: len 1750 larger than size (1536)
[  417.976868] dwmac-sun8i 1c30000.ethernet eth0: len 1634 larger than size (1536)
[  417.996066] dwmac-sun8i 1c30000.ethernet eth0: len 1634 larger than size (1536)
[  418.015725] dwmac-sun8i 1c30000.ethernet eth0: len 1634 larger than size (1536)
[  418.037684] dwmac-sun8i 1c30000.ethernet eth0: len 1574 larger than size (1536)
[  418.057998] dwmac-sun8i 1c30000.ethernet eth0: len 1574 larger than size (1536)
[  418.078507] dwmac-sun8i 1c30000.ethernet eth0: len 1574 larger than size (1536)
[  418.100443] dwmac-sun8i 1c30000.ethernet eth0: len 1546 larger than size (1536)
[  418.121022] dwmac-sun8i 1c30000.ethernet eth0: len 1546 larger than size (1536)
[  418.141562] dwmac-sun8i 1c30000.ethernet eth0: len 1546 larger than size (1536)
Found 44 propert(y/ies)
```
Is it problem with NanoPi ethernet driver or because of some additional camera settings?

TIS-Stefan, 2018-09-25:
Hello

I searched for this message in the internet and found https://serverfault.com/questions/749110/why-can-i-receive-ethernet-frames-bigger-than-my-current-mtu-size

I do not know, why your get these packet size warning messages, but as long as the camera works, I would ignore them.

Stefan