TheImagingSource/tiscamera #220: Select binning mode?

midjji, 2018-09-25:
Camera supports it? but cant find how to activate it.

TIS-Stefan, 2018-09-25:
Which camera model do you use?

The binning / skipping property is name "Override Scanning Mode". Before it can be used a video format of half size must be set.
Some of the possible values in Override Scanning Mode do not fit to cameras. I suggest to try tcam-capture for testing this property.

closed 2018-09-25 by TIS-Stefan

midjji, 2018-09-25:
Using dmk 33ux252

midjji, 2018-09-25:
The dmk 33ux252
hmm there is only a single parameter, between 1,16 if I read it right,

shouldnt there be two, one for each binning factor, row col?

How do I compute what happens?

TIS-Stefan, 2018-09-25:
> hmm there is only a single parameter, between 1,16 if I read it right,
This is correct, the values enable the different possible binning / skipping modes.

> How do I compute what happens?
Try them the in the tcam-capture program. Set a 640x480 video format first. Currently I do not know the valid values.