TheImagingSource/tiscamera #221: Camera device time and trigger id values

midjji, 2018-09-25:
In windows we can access the device time and the current trigger index since last the camera was set to trigger mode. (unless we read the docs wrong). How do I access these values in linux?

TIS-Stefan, 2018-09-25:
You may let me know, which camera you use.

TIS-Stefan, 2018-09-25:
I now know, which camera you use.

The additional data for each image as you mention above, is currently not available. First of all, the V4L2 does not support this frame information. Secondly , the tcambin does not have this implemented, also not for GigE cameras.
But if you use Aravis with USB3Vision directly, you could get access to this properties. I did not try this on my own.

midjji, 2018-09-26:
Hi,
Isnt Aravis ethernet only? Any examples for something like what you mean?

midjji, 2018-09-26:
Nevermind I found it,
You say that isnt tested? which is the most stable ?

Den ons 26 sep. 2018 kl 12:55 skrev Mikael Persson <
mikael.p.persson@gmail.com>:

TIS-Stefan, 2018-09-26:
> You say that isnt tested? which is the most stable ?
I said, I did not test that on my own. Aravis is not our project.

midjji, 2018-09-26:
Do you fully support the genicam standard?

TIS-Stefan, 2018-09-26:
> Do you fully support the genicam standard?

As far as I know, yes.

closed 2018-11-27 by midjji