TheImagingSource/tiscamera #222: camera-ip-conf make failing -- best practice for includes

rbrown-acertaralabs, 2018-09-26:
Trying to build `camera-ip-conf` at `/home/user/ansel/tiscamera/tools/camera-ip-conf` per the TIS Wiki here.

During the build, it cannot find various `.h` files -- there's likely a best-practice, here, but I am forgetting.  How should I setup the environment so that make and g++ properly find all needed `.h` files?

For example, I've done this, but it seems clunky and wrong (although it would eventually work):

First build/make error (no such file `Camera.h`):
[image]

Manually forcing it within `ConsoleManager.h` by suppliying a relative path to `Camera.h` does work but it just bubbles to the next `.h` that it cannot find:

[image]

[image]

I'm sure there is a better way, so I am simply asking for the best practice here to make the build/make work when it cannot find '.h' files.

rbrown-acertaralabs, 2018-09-26:
Neither `CPATH` nor `CPLUS_INCLUDE_PATH` seem to work...

rbrown-acertaralabs, 2018-09-26:
WAIT!  I was overcomplicating things!  I of course did all the steps under "Building tiscamera" and I just realized that all of the tools were built as part of that and placed under `builds`.  Well, I feel silly.  Closing this ticket.

closed 2018-09-26 by rbrown-acertaralabs

TIS-Stefan, 2018-09-26:
Hello 

Please do not feel silly. When I started with Linux coming from many years using Windows, I made the same errors as you. 

Stefan