TheImagingSource/tiscamera #224: GigE -- Best Way to Capture Single Images?

rbrown-acertaralabs, 2018-09-28:
My GigE TIS camera is working well and the `live-stream.sh` script works great:

`.../tiscamera/examples/gstreamer-1.0/live-stream.sh`

Having navigated the Wiki and the tickets on GitHub, I am a bit lost about what to do next for integrating the camera into my final application:
* I am on Debian 9.5; using TIS GigE camera DFM 25GP031-ML; planning to implement in Python (C if needed)
* I am interested in triggering/capturing single images to files (no video stream)
* The following script works well:
    `.../tiscamera/examples/python/trigger-images.py`
* The following script also works well:
    `.../tiscamera/examples/python/properties`
* The above two scripts, and the `live-stream.sh` script seem very close to what I need, but now I want to capture a single image from inside my own Python script.  Which approach would be best?
    1. Use `gst-launch-1.0` called from `subprocess.call()` from Python, using something like `gst-launch-1.0 tcamsrc serial=12345 num-buffers=1 ! jpegenc ! filesink location=test.jpg` (this does not work, still debugging), or...
    2. Use `snap_image` method as shown in `.../tiscamera/examples/python/properties` ... this subsequently calls method `close_valve` and subsequently sets property "drop" -- this seems a bit convoluted...or...
    3. Use the structure in `trigger_images.py` which uses the object `source = Gst.ElementFactory.make("tcambin")` to set property `source.set_tcam_property("Trigger Mode"...)` which fires the trigger, but what would the next step be to get the image data and save it to a file?  Is there another method in the `Gst.ElementFactory` object/class that would do this?  I suppose I should read the documentation on `Gst.ElementFactory`, but perhaps there is a quick answer to this?

Just looking for some guidance.  Thank you!

TIS-Stefan, 2018-09-28:
Hello

I prefer to use a small Pyhton class, that contains all the GStreamer code, the property handling and has a nice function like "GetImage", so I get a numpy data type with the image and have OpenCV for saving. More or less like this sample:
[save-single-image.zip](https://github.com/TheImagingSource/tiscamera/files/2427621/save-single-image.zip)

Stefan

rbrown-acertaralabs, 2018-09-28:
Thank you for the example...this is definitely closer to what I need.  I will modify your example to further experiment to become familiar with the TIS module.

I do have some additional questions...

I want my app to be as lean, fast, and efficient as possible.  For example, I would prefer to avoid the overhead of OpenCV, so I will use PIL combined with Tk (this seems straightforward).  What other software components can I ignore in order to achieve very fast camera initialization and very low latency for single-image captures (GigE)?  For example, my impression is that an open pipeline will constantly stream images at the frame rate specified, and your example `TIS.Get_image` invokes `get_property('last-sample')`.  So it looks like the camera is constantly streaming and we are just grabbing the last sample.  I don't need a stream...can I capture single images without opening a pipeline?   I'm looking for recommendations for lowest code overhead and footprint for simply capturing occasional still snapshots.

Thank you...I appreciate your time.

TIS-Stefan, 2018-10-08:
Hello rbrown,

I wonder, whether we already had email contact, in which I have sent the 
[trigger-save-image.zip](https://github.com/TheImagingSource/tiscamera/files/2455591/trigger-save-image.zip) to you. 
Documentation is in the readme files. 

Stefan

rbrown-acertaralabs, 2018-10-08:
Hello Stefan,

Yes, I did receive your `trigger-save-image.zip` code.  It works very well and I continue to use it as the core of my application: I replaced the OpenCV GUI components with a leaner Tk approach, verified network traffic with the cameras is low while they are "idle", and continue to build on the code to produce objects that can be integrated with my application.  Thank you again very much for this...it really provided the solution I was looking for -- the documentation you provided is also very good.

Sincerely,
Rob
--
Robert Brown
Lead Software Engineer
Acertara Acoustic Laboratories, LLC
1900 South Sunset Street, Unit F
Longmont, CO 80501
(303)834-8413 x310
rbrown@acertaralabs.com<mailto:rbrown@acertaralabs.com>

-----Original Message-----
From: Robert Brown
Sent: Tuesday, October 2, 2018 6:36 PM
To: 'Stefan Geissler, TIS Support' <support@theimagingsource.com>
Cc: 'steve.connor@theimagingsource.com' <steve.connor@theimagingsource.com>
Subject: RE: Guidance for Image Capturing <case:458338>

Hello Stefan,

Thank you very much for the sample code!  This is excellent.  I experimented with it and it is working well...I removed the dependency on OpenCV and I now render images directly in Tk (faster performance and fewer window lock ups under X) and I'm off to a good start -- this will be a very good beginning approach for the framework of my application.

Sincerely,
Rob

LukeSkywalker92, 2018-10-18:
Hi @TIS-Stefan,

i also tried to use your `trigger-save-image.zip` code, and the Program.py works as expected. But when i change the time between two images (time.sleep(10)) in the while loop (should be line 100), i get only one image an then nothing happens anymore.
Do you have an idea why this is happening?
In my application i need to trigger images every minute.

Best Luke

TIS-Stefan, 2018-10-18:
Hello Luke

I have no idea, why this behaves different. time.sleep(10) should simply wait for 10 seconds. This has no effect on the camera.
Does the sample work again with 2 or 5 seconds as sleeping time?

Stefan

BTW: The sample is now at https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/tree/master/Save%20an%20image%20on%20trigger%20in%20Python#save-an-image-on-trigger

LukeSkywalker92, 2018-10-18:
I now used the sample from this link and it's the same.
I then tried it for different sleeping times and it works up to 6 seconds. 
With 7 seconds it stops working.

TIS-Stefan, 2018-10-18:
I tested that on my computer. No problems with 10 second.  Also not with 65 seconds (to see the seconds hand on my watch moving).
I suggest to add a "print("Trigger")" and check, whether your program does not hang at the sleep (which is improbably. Do you see the "No image received" output?

``` Python
try:
        while lastkey != 27 and error < 5:
                time.sleep(65)
                print("Trigger")
                Tis.Set_Property("Software Trigger",1) # Send a software trigger
```
I used full resolution and hightest frame rate:
``` Python
Tis = TIS.TIS("28510360", 2592, 1944, 15, True)
```
Maybe that causes your issue, something goes sleeping... So you may try lower frame rates:
``` Python
Tis = TIS.TIS("28510360", 2592, 1944, 8, True)
```

rbrown-acertaralabs, 2018-10-18:
Maybe on a related note, I found that this code (trigger-save-image, or `trigger-save-camera.py`) would often skip an image capture...it would be off by one.  For example, in a scenario where you capture image A then change the image subject (e.g. by pointing the camera elsewhere) and then do another capture B...very often capture B was the same image as capture A.  I found a couple of items that cleared this up:

Inside `on_new_image`, move `newImageReceived` below the call to `GetImage`:

<pre>
# (1) original code
userdata.busy = True
userdata.newImageReceived = True
userdata.image = tis.Get_image()
userdata.busy = False
</pre>

<pre>
# (2) modified
userdata.busy = True
userdata.image = tis.Get_image()
userdata.newImageReceived = True
userdata.busy = False
</pre>

...this seems important because the main capture loop is waiting on `CD.newImageReceived` which is being updated asynchronously in the callback, so the original code (1) could set `newImageReceived` prematurely before the call to `GetImage`.  It might be good to put additional error checks around `GetImage`, too.

Also, in the original code, I believe there was a very subtle typo that could have caused capture synchronization issues:

<pre>
# (1) original code
# Check, whether there is a new image and handle it.
if CD.newImageReceived is True:
    CD.newImageRecevied = False
    cv2.imshow('Window', CD.image)
</pre>

...notice above "Received" is spelled wrong at `CD.newImageRecevied = False`, so this is setting an additional property in the CD object named `newImageRecevied` (mis-spelled).  It's valid code, thus no syntax errors, but it's not changing the intended flag.

closed 2019-11-29 by TIS-Stefan

sudhirsun, 2020-12-18:
How we get the seral number of a embedded webcam in ubuntu? 
Tis = TIS.TIS("**28510360**", 2592, 1944, 15, True) to create TIS object this is a mandatory parameter right?

How we capture an image from our embedded webcam in below format ?
640x480 60 fps, yuv422, 8bit

TIS-Stefan, 2020-12-18:
Hello

I do not know, how to handle third party devices, because we do not support them. But I think, the laptop cameras have no serial numbers at all. Or no serial number property.

Your embedded webcam is not supported by tiscamera. You may try the v4l2src module and a gstreamer pipeline instead.