TheImagingSource/tiscamera #225: ROS example code does not work under ROS-melodic/ubuntu18.04

sevangelatos, 2018-09-28:
I have an DMM 37UX273-ML under Ubuntu 18.04.1 LTS using ROS melodic. The included gstreamer  version is 1.14.1.
I am using tag: v-tiscamera-0.10.0.

I was unable to make the ROS example run out of the box just by changing the tiscamera line in tis_rosstarter.

Sometimes the output would be:
```
./tis_rosstarter 
tcambin serial="38810198" name=source ! video/x-raw,format=GRAY8,width=1440,height=1080,framerate=30/1 ! videoconvert ! video/x-raw,format=RGB ,width=1440,height=1080,framerate=30/1! shmsink socket-path=/tmp/ros_mem
Press Enter to end program[ INFO] [1538129648.537871642]: Using gstreamer config from env: "shmsrc socket-path=/tmp/ros_mem ! video/x-raw-rgb, width=1440,height=1080,framerate=30/1,bpp=24,depth=24,blue_mask=16711680, green_mask=65280, red_mask=255 ! ffmpegcolorspace"
[ INFO] [1538129648.540020936]: using default calibration URL
[ INFO] [1538129648.540054078]: camera calibration URL: file:///home/sevangelatos/.ros/camera_info/camera.yaml
[ INFO] [1538129648.540618289]: Loaded camera calibration from 

(gscam:31388): GStreamer-WARNING **: 13:14:08.548: 0.10-style raw video caps are being created. Should be video/x-raw,format=(string).. now.
[ INFO] [1538129648.548485783]: Time offset: 1538118462.104
[ INFO] [1538129648.653965030]: Publishing stream...
[ INFO] [1538129648.654091288]: Started stream.

(gscam:31388): GStreamer-CRITICAL **: 13:14:08.654: gst_caps_get_structure: assertion 'GST_IS_CAPS (caps)' failed

(gscam:31388): GStreamer-CRITICAL **: 13:14:08.654: gst_structure_get_int: assertion 'structure != NULL' failed

(gscam:31388): GStreamer-CRITICAL **: 13:14:08.654: gst_structure_get_int: assertion 'structure != NULL' failed

Program ended
```

In some cases, (randomly, in some invocations) I would get the following message repeated:
```
[ WARN] [1538129659.023345965]: GStreamer image buffer underflow: Expected frame to be 2095596416 bytes but got only 4665600 bytes. (make sure frames are correctly encoded)
```
In any case, no images were published from the camera.

The problem seems to be related to changes in the gstreamer pipeline syntax. In my case, I was able to make it work by simplifying the gstreamer pipeline as seen in the [attached patch file](https://github.com/TheImagingSource/tiscamera/files/2427924/tis_rosstarter.txt).

Could you please update the example code to work correctly under 18.04?

TIS-Stefan, 2018-09-28:
Hello

2095596416 bytes seems to be somewhat too big for an image.

The ROS sample code is based on an older ROS GSTreaner Plugin version, which uses GStreamer 0.10. Is that still available in Ubuntu 18.04?
Maybe it makes sense to update the ROS plugin, if I am right, there was a GStreamer 1.0  using version, right?

Does the standard pipeline 
`gst-launch-1.0 tcambin ! video/x-raw,format=RGB ,width=1440,height=1080,framerate=30/1! videoconvert ! ximagesink`
give an image?
Which computer model do you use?

sevangelatos, 2018-09-28:
Indeed, 2Gbytes is insane for an image. In 18.04 only gstreamer1.0 is available. The pipeline 
```
gst-launch-1.0 tcambin ! video/x-raw,format=RGB ,width=1440,height=1080,framerate=30/1 ! videoconvert ! ximagesink
```
works fine. Similarly, I was able to get the example to work using the patch I attached. 

PS: I am using a [Lenovo Y50-70](https://github.com/TheImagingSource/tiscamera/files/2428606/lenovo_hardware_info.txt) (i7-4720HQ).

TIS-Stefan, 2018-09-28:
Thank you. I will read the changes you made.

closed 2019-11-29 by TIS-Stefan