TheImagingSource/tiscamera #227: ValueError: Namespace Tcam not available

syedmustafa54, 2018-10-05:
I am on Ubuntu 18.04 when running python opencv.py i am getting error 
```Traceback (most recent call last):
  File "imaging_source_opencv.py", line 26, in <module>
    from list_devices import select_camera
  File "/home/bintex/Downloads/list_devices.py", line 24, in <module>
    gi.require_version("Tcam", "0.1")
  File "/home/bintex/anaconda3/lib/python3.6/site-packages/gi/__init__.py", line 130, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace Tcam not available
```

TIS-Stefan, 2018-10-08:
Hello

did you follow the readme at https://github.com/TheImagingSource/tiscamera (scroll down a little bit) and made and installed the built GStreamer modules? If not, the  Tcam namespace wont be available.

Stefan

gilzimmerman, 2019-12-27:
I am having the same issue and followed all the instructions on the readme. PYthon 3.7. Please advise.

TIS-Edgar, 2020-01-07:
Hi,

I am assuming you did not change any installation paths during configuration. Is that correct?

Can you give me the output of the following commands:

`sudo find /usr -name Tcam-0.1`
`sudo find /usr -name *.gir`
`sudo find /usr -name *.typelib`

charlypradel, 2020-01-27:
Hi, 

I have the same problem. I followed the installation guide and use python 3.7.3. I entered the three commands and these are my outputs:

python3 00-list-devices.py 
Traceback (most recent call last):
  File "00-list-devices.py", line 24, in <module>
    gi.require_version("Tcam", "0.1")
  File "/usr/lib/python3/dist-packages/gi/__init__.py", line 129, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace Tcam not available

pi@raspberrypi:~/Documents/tiscamera/examples/python $ sudo find /usr -name Tcam-0.1
pi@raspberrypi:~/Documents/tiscamera/examples/python $ sudo find /usr -name *.gir
/usr/share/gir-1.0/cairo-1.0.gir
/usr/share/gir-1.0/GstCheck-1.0.gir
/usr/share/gir-1.0/DBusGLib-1.0.gir
/usr/share/gir-1.0/GIRepository-2.0.gir
/usr/share/gir-1.0/libxml2-2.0.gir
/usr/share/gir-1.0/GstVideo-1.0.gir
/usr/share/gir-1.0/GstApp-1.0.gir
/usr/share/gir-1.0/GstRtsp-1.0.gir
/usr/share/gir-1.0/GstRtp-1.0.gir
/usr/share/gir-1.0/xfixes-4.0.gir
/usr/share/gir-1.0/GstAudio-1.0.gir
/usr/share/gir-1.0/GstSdp-1.0.gir
/usr/share/gir-1.0/GstNet-1.0.gir
/usr/share/gir-1.0/GModule-2.0.gir
/usr/share/gir-1.0/GstBase-1.0.gir
/usr/share/gir-1.0/xlib-2.0.gir
/usr/share/gir-1.0/fontconfig-2.0.gir
/usr/share/gir-1.0/win32-1.0.gir
/usr/share/gir-1.0/xrandr-1.3.gir
/usr/share/gir-1.0/GLib-2.0.gir
/usr/share/gir-1.0/Gio-2.0.gir
/usr/share/gir-1.0/GstPbutils-1.0.gir
/usr/share/gir-1.0/GL-1.0.gir
/usr/share/gir-1.0/GstTag-1.0.gir
/usr/share/gir-1.0/GstAllocators-1.0.gir
/usr/share/gir-1.0/GObject-2.0.gir
/usr/share/gir-1.0/freetype2-2.0.gir
/usr/share/gir-1.0/xft-2.0.gir
/usr/share/gir-1.0/GstGL-1.0.gir
/usr/share/gir-1.0/DBus-1.0.gir
/usr/share/gir-1.0/GstController-1.0.gir
/usr/share/gir-1.0/Gst-1.0.gir
pi@raspberrypi:~/Documents/tiscamera/examples/python $ sudo find /usr -name *.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GstCheck-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GstAllocators-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/fontconfig-2.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GstRtsp-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/Gst-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/Atk-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GstVideo-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/PangoFT2-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GIRepository-2.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GObject-2.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/xrandr-1.3.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/DBus-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GstSdp-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GstRtp-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GMenu-3.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GdkPixbuf-2.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/Gio-2.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/libxml2-2.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/xfixes-4.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/Gdk-3.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GstPbutils-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/xlib-2.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GL-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GstApp-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GstAudio-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GdkX11-3.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GstGL-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/PangoXft-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GstNet-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/Pango-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/DBusGLib-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/xft-2.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/cairo-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GdkPixdata-2.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GLib-2.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GstController-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/Gtk-3.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/freetype2-2.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GstTag-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GModule-2.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/PangoCairo-1.0.typelib
/usr/lib/arm-linux-gnueabihf/girepository-1.0/GstBase-1.0.typelib

3242433424, 2020-02-13:
Hi, 
i have the same Problem. Trying to run my TIS camera on my Raspberrypi. 
Just getting the ValueError: Namespace Tcam not available.

Pls Help ist urgent

TIS-Stefan, 2020-02-13:
There is a similar issue in which Anaconda was the reason:
https://github.com/TheImagingSource/tiscamera/issues/255

Maybe that helps?

closed 2021-02-19 by TIS-Stefan