TheImagingSource/tiscamera #229: Install Error occurred on Ubuntu 16.04

Mactarvish, 2018-10-31:
Hi, I'm trying to install the tiscamera for my DMK 72BUC02 , and I followed the installation informations step by step, but when I ran the "sudo make install" an error occurred:

wzj@ubuntu:~/tiscamera/build$ sudo make install
[  1%] Built target pugixml_obj
Scanning dependencies of target tcam-base
[  1%] Building CXX object src/CMakeFiles/tcam-base.dir/logging.cpp.o
[ 10%] Built target tcam-base
[ 11%] Linking CXX shared library libtcam.so
[ 19%] Built target tcam
[ 20%] Linking C shared library libtcamprop.so
[ 21%] Built target tcamprop
[ 22%] Generating Tcam-0.1.gir
[ 23%] Generating Tcam-0.1.typelib
[ 23%] Built target create_gobject
Scanning dependencies of target tcam-algorithms
[ 23%] Building CXX object src/algorithms/CMakeFiles/tcam-algorithms.dir/whitebalance.cpp.o
[ 24%] Building CXX object src/algorithms/CMakeFiles/tcam-algorithms.dir/biteater.cpp.o
[ 25%] Building CXX object src/algorithms/CMakeFiles/tcam-algorithms.dir/parallel.cpp.o
[ 26%] Linking CXX shared library libtcam-algorithms.so
[ 31%] Built target tcam-algorithms
Scanning dependencies of target tcamgstbase
[ 32%] Building CXX object src/gstreamer-1.0/CMakeFiles/tcamgstbase.dir/tcamgstbase.cpp.o
[ 33%] Linking CXX shared library libtcamgstbase.so
[ 33%] Built target tcamgstbase
Scanning dependencies of target gsttcamautoexposure
[ 34%] Building CXX object src/gstreamer-1.0/CMakeFiles/gsttcamautoexposure.dir/gsttcamautoexposure.cpp.o
[ 35%] Linking CXX shared library libgsttcamautoexposure.so
[ 35%] Built target gsttcamautoexposure
Scanning dependencies of target gsttcamautofocus
[ 36%] Building CXX object src/gstreamer-1.0/CMakeFiles/gsttcamautofocus.dir/gsttcamautofocus.cpp.o
[ 37%] Linking CXX shared library libgsttcamautofocus.so
[ 37%] Built target gsttcamautofocus
Scanning dependencies of target gsttcamwhitebalance
[ 37%] Building CXX object src/gstreamer-1.0/CMakeFiles/gsttcamwhitebalance.dir/gsttcamwhitebalance.cpp.o
[ 38%] Linking CXX shared library libgsttcamwhitebalance.so
[ 38%] Built target gsttcamwhitebalance
Scanning dependencies of target gsttcamsrc
[ 39%] Building CXX object src/gstreamer-1.0/CMakeFiles/gsttcamsrc.dir/gsttcamsrc.cpp.o
[ 40%] Linking CXX shared library libgsttcamsrc.so
[ 40%] Built target gsttcamsrc
Scanning dependencies of target gsttcambiteater
[ 41%] Building CXX object src/gstreamer-1.0/CMakeFiles/gsttcambiteater.dir/gsttcambiteater.cpp.o
[ 42%] Linking CXX shared library libgsttcambiteater.so
[ 42%] Built target gsttcambiteater
Scanning dependencies of target gsttcambin
[ 42%] Building CXX object src/gstreamer-1.0/CMakeFiles/gsttcambin.dir/gsttcambin.cpp.o
/home/wzj/tiscamera/src/gstreamer-1.0/gsttcambin.cpp: In lambda function:
/home/wzj/tiscamera/src/gstreamer-1.0/gsttcambin.cpp:644:104: warning: format not a string literal and no format arguments [-Wformat-security]
 g_error_new(GST_CORE_ERROR, GST_CORE_ERROR_MISSING_PLUGIN, msg_string.c_str());
                                                                              ^
[ 43%] Linking CXX shared library libgsttcambin.so
[ 43%] Built target gsttcambin
[ 45%] Built target tcam-dfk73
[ 46%] Linking CXX shared library libtcam-v4l2.so
[ 49%] Built target tcam-v4l2
[ 50%] Linking CXX shared library libtcam-libusb.so
[ 58%] Built target tcam-libusb
Scanning dependencies of target tcam-ctrl
[ 59%] Building CXX object tools/tcam-ctrl/CMakeFiles/tcam-ctrl.dir/main.cpp.o
[ 60%] Building CXX object tools/tcam-ctrl/CMakeFiles/tcam-ctrl.dir/properties.cpp.o
[ 60%] Building CXX object tools/tcam-ctrl/CMakeFiles/tcam-ctrl.dir/formats.cpp.o
[ 61%] Linking CXX executable tcam-ctrl
[ 61%] Built target tcam-ctrl
[ 91%] Built target 33update
[ 98%] Built target firmware-update
[100%] Built target dfk73udev
Install the project...
-- Install configuration: ""
-- Installing: /usr/lib/libtcam.so.0.10.0
-- Up-to-date: /usr/lib/libtcam.so.0
-- Up-to-date: /usr/lib/libtcam.so
-- Set runtime path of "/usr/lib/libtcam.so.0.10.0" to "/usr/lib/tcam-0"
-- Installing: /usr/lib/libtcamprop.so.0.1
-- Up-to-date: /usr/lib/libtcamprop.so
-- Set runtime path of "/usr/lib/libtcamprop.so.0.1" to ""
-- Installing: /usr/share/gir-1.0/Tcam-0.1.gir
-- Installing: /usr/lib/x86_64-linux-gnu/girepository-1.0/Tcam-0.1.typelib
-- Up-to-date: /usr/include/tcamprop.h
-- Installing: /usr/lib/libtcamgstbase.so
-- Set runtime path of "/usr/lib/libtcamgstbase.so" to ""
-- Installing: /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamsrc.so
-- Set runtime path of "/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamsrc.so" to ""
-- Installing: /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamwhitebalance.so
-- Set runtime path of "/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamwhitebalance.so" to ""
-- Installing: /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamautoexposure.so
-- Set runtime path of "/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamautoexposure.so" to ""
-- Installing: /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamautofocus.so
-- Set runtime path of "/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamautofocus.so" to ""
-- Installing: /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcambiteater.so
-- Set runtime path of "/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcambiteater.so" to ""
-- Installing: /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcambin.so
-- Set runtime path of "/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcambin.so" to ""
-- Installing: /usr/lib/libtcam-algorithms.so.0.10.0
-- Up-to-date: /usr/lib/libtcam-algorithms.so.0
-- Up-to-date: /usr/lib/libtcam-algorithms.so
-- Installing: /usr/lib/tcam-0/libtcam-v4l2.so.0.10.0
-- Up-to-date: /usr/lib/tcam-0/libtcam-v4l2.so.0
-- Up-to-date: /usr/lib/tcam-0/libtcam-v4l2.so
-- Set runtime path of "/usr/lib/tcam-0/libtcam-v4l2.so.0.10.0" to ""
-- Up-to-date: /usr/lib/libtcam-dfk73.so.0.10.0
-- Up-to-date: /usr/lib/libtcam-dfk73.so.0
-- Up-to-date: /usr/lib/libtcam-dfk73.so
-- Installing: /usr/lib/tcam-0/libtcam-libusb.so.0.10.0
-- Up-to-date: /usr/lib/tcam-0/libtcam-libusb.so.0
-- Up-to-date: /usr/lib/tcam-0/libtcam-libusb.so
-- Set runtime path of "/usr/lib/tcam-0/libtcam-libusb.so.0.10.0" to ""
-- Up-to-date: /usr/lib/pkgconfig/tcam.pc
-- Up-to-date: /usr/share/tiscamera/images
-- Up-to-date: /usr/share/tiscamera/images/23uc256.png
-- Up-to-date: /usr/share/tiscamera/images/logo.jpg
-- Up-to-date: /usr/share/uvcdynctrl/data/199e/tisEUVC.xml
-- Up-to-date: /usr/share/uvcdynctrl/data/199e/usb3.xml
-- Up-to-date: /usr/lib/udev/rules.d/80-theimagingsource-cameras.rules
-- Installing: /usr/bin/tcam-ctrl
-- Set runtime path of "/usr/bin/tcam-ctrl" to ""
-- Up-to-date: /usr/bin/tcam-capture
CMake Error at tools/tcam-capture/cmake_install.cmake:40 (file):
  file INSTALL given no DESTINATION
Call Stack (most recent call first):
  tools/cmake_install.cmake:38 (include)
  cmake_install.cmake:40 (include)
  

Makefile:115: recipe for target 'install' failed
make: *** [install] Error 1

How to solve it? Thanks in advance~

TIS-Stefan, 2018-11-01:
Hello

I just tested that on a freshly installed Ubuntu and there was no problem with installing tcam-capture.

If you did not change the cmake parameters, then tcam-capture should be installed into /usr/bin. "/usr" is the preffix passed by "-DCMAKE_INSTALL_PREFIX=/usr" to cmake. 

I suggest to remove the build directory and repeat the build steps, making sure, you copied the cmake command line (readme.md) completely (with the to dots (..) at the end.

Stefan

Mactarvish, 2018-11-05:
Thank you, I'll try your suggestion.

closed 2018-11-05 by Mactarvish

veuron2018, 2019-01-21:
Hi Stefan,

I got the same error at a fresh Ubuntu 18.04.
-----------------------------------------------------------------------------------
CMake Error at tools/tcam-capture/cmake_install.cmake:45 (file):
  file INSTALL given no DESTINATION
-----------------------------------------------------------------------------------

line 45 "file(INSTALL DESTINATION "" TYPE DIRECTORY FILES "/home/qiang/tiscamera/to    ols/tcam-capture/tcam_capture" REGEX "/\\_\\_pycache\\_\\_$" EXCLUDE REGEX "    /image\\_dir[^/]*$" EXCLUDE)"

I did see /usr/bin/tcam-capture but not sure what the meaning of this error above.
When I tried to run tcam-capture, it said "ModuleNotFoundError: No module named 'tcam_capture'". It seems it could not import /tools/tcam_capture.

The only different thing I did is before I build based on your instructions, I clicked the "Download" button under the link https://www.theimagingsource.com/support/downloads-for-linux/install/tiscamera/ to install tiscamera. Then I uninstalled it and followed the instruction in github instead.

Regards,

veuron2018, 2019-01-22:
The culprit is python.
We installed Anaconda3 for machine learning. 
python in Anaconda3 and the default one under /usr/bin may be messed up in apt-get install, build or run time.
After removing Anaconda3 from PATH, your instructions works and tcam-capture runs well.
Thanks for help.

sopper08, 2019-02-02:
> The culprit is python.
> We installed Anaconda3 for machine learning.
> python in Anaconda3 and the default one under /usr/bin may be messed up in apt-get install, build or run time.
> After removing Anaconda3 from PATH, your instructions works and tcam-capture runs well.
> Thanks for help.

Thanks for your help.
I meet the same problem, and it be solved now after your suggestion.