TheImagingSource/tiscamera #233: Problem when using gscam in ROS

Crisliu7, 2018-12-05:
I worked with DMM 42BUC03-ML and tried to launch the camera in ROS.
Every time when I launch it, it's fine, when I put some image or objects in front of the camera, the exception will be thrown. It gives me:
> ** (gscam:11048): WARNING **: ffmpegcsp0: size 1212416 is not a multiple of unit size 1228800.

Here is my launch file:
```
<launch>

  <arg name="cam_name" value="creative_cam" />

  <env name="GSCAM_CONFIG" value="v4l2src device=/dev/video3 ! video/x-raw-gray,framerate=30/1,width=1280,height=960 ! ffmpegcolorspace " />
  <node pkg="gscam" type="gscam" name="$(arg cam_name)">
    <param name="camera_name" value="$(arg cam_name)" />
    <!-- <param name="camera_info_url" value="package://localcam/calibrations/${NAME}.yaml" /> -->
    <remap from="camera/image_raw" to="/sensors/scanner/camera/image_raw" />
  </node> 

</launch>```

TIS-Stefan, 2018-12-06:
Hello

The problem is caused by missing data packages not picked up by the USB controller from the camera.  1228800 is the result of 1280x960, which is number of pixels. You got some less pixels, therefore ffmpeg complains. You may try a lower frame rate or you may try our tiscamera / tcambin modules instead of v4l2src. The tcambin discards incomplete images. However, the tcambin needs GStreamer 1.0. As far as I know, there should be a newer GSCAM, which  uses GStreamer 1.0 too. 

Stefan

Crisliu7, 2018-12-06:
> Hello
>
> The problem is caused by missing data packages not picked up by the USB controller from the camera. 1228800 is the result of 1280x960, which is number of pixels. You got some less pixels, [...]

Thanks, Stefan. However, when I connect the camera to NVIDIA TX2 and set the framerate as 30 fps, the real framerate is just 10 fps. Btw, any instruction for using tcambin to launch the camera?

TIS-Stefan, 2018-12-06:
The TX2 is ARM based, the graphic CPU is not used for the camera. Therefore, it is not faster, than a PI3 or Odroid. 10 fps is a frame rate, where you can get reliable images.

Do you have a GSCam, that is based on GStreamer 1.0? If not, you can not use tcambin.

Stefan

Crisliu7, 2018-12-06:
> The TX2 is ARM based, the graphic CPU is not used for the camera. Therefore, it is not faster, than a PI3 or Odroid. 10 fps is a frame rate, where you can get reliable images.
> 
> Do you have a GSCam, that is based on GStreamer 1.0? If not, you can not use tcambin.
> 
> Stefan

So it means the highest framerate for my camera model can only be 10fps on TX2?

I have gscam. do I need to uninstall the gstreamer0.10 and install the gstreamer1.0?
and set following line in my ros luanch file?
`<env name="GSCAM_CONFIG" value="tcambin device=/dev/video3 ! video/x-raw-gray,framerate=30/1,width=744,height=480 ! ffmpegcolorspace " />`

TIS-Stefan, 2018-12-07:
Hello

> So it means the highest framerate for my camera model can only be 10fps on TX2?

Unfortunately, yes.

>I have gscam. do I need to uninstall the gstreamer0.10 and install the gstreamer1.0?
and set following line in my ros luanch file?

Well, we do not recommend to use the totally outdated gstreamer0.10. Please check for a newer version of gscam or request one. It is, as far as I know, a github project.
In case, you can not get a newer version, you may refer to https://github.com/TheImagingSource/tiscamera/tree/master/examples/ROS. The sample there builds a GStreamer 1.0 pipeline for the camera and allows to access the camera properties. The images are sinked into a memory sink. The GSCAM pipeline reads the images from the memory sink and forwards them to ROS.

Stefan

closed 2019-11-29 by TIS-Stefan