TheImagingSource/tiscamera #235: DFG USB2Pro on Xubuntu

avstrallen, 2018-12-21:
I'm having an awful time getting a DFG USB2Pro to work on Linux. Nobody else is writing about this particular problem, so I guess I must be doing something incredibly dim.

I can enumerate the device from USB:

adam:~$ tcam-ctrl -l
Available devices:
Model		Type	Serial

DFG/USB2pro	V4L2	47510665

I can list and set properties:

adam:~$ tcam-ctrl -p 47510665
Found 9 propert(y/ies)
Brightness                              (int) min=0 max=255 step=1 default=128 value=115 category=Exposure
Contrast                                (int) min=0 max=127 step=1 default=64 value=64 category=Exposure
Saturation                              (int) min=0 max=127 step=1 default=64 value=64 category=Color
Hue                                     (int) min=-128 max=127 step=1 default=0 value=0 category=Color
Volume                                  (int) min=0 max=65535 step=655 default=60928 value=60928 category=Unknown
Balance                                 (int) min=0 max=65535 step=655 default=32768 value=32768 category=Unknown
Bass                                    (int) min=0 max=65535 step=655 default=32768 value=32768 category=Unknown
Treble                                  (int) min=0 max=65535 step=655 default=32768 value=32768 category=Unknown
Mute                                    (bool) default=false value=false category=Unknown

However, when I try to list available formats, an empty list is returned:

adam:~$ tcam-ctrl -f 47510665
Available format settings:
adam:~$

Any advice very welcome!
Many thanks,
Adam

TIS-Stefan, 2018-12-21:
Hello Adam

In latest Ubuntu Kernel is the driver for DFG/USB2pro built in. A user made this. 

Merry Christmas!
Stefan

closed 2019-11-29 by TIS-Stefan

0x3333, 2020-08-19:
Sorry to revisit this issue, but I'm unable to read frames from a DFG/USB2pro on linux Debian/Ubuntu.

I know that they use the CX231XX driver, but I can't get the frames using V4L. In your comment you say:

     the DFG/USB2pro uses a proprietary protocol, therefore v4l2 and gstreamer wont handle it

How to handle?