TheImagingSource/tiscamera #238: Strobe Duration/delay issue 33UX264

blazini36, 2019-01-01:
I use the cameras exclusively with a strobe controlled by the GPIO. I originally did so without much issue as the strobe I previously set up had extremely little delay. I was able to achieve <100us exposure with the camera set at "strobe_exposure=1" As nice as it was it was not practical for my design.

I've designed something more practical but it has a slight delay in switching and requires more "on-time". I find it's not possible to sync the strobe whatsoever unless the exposure is set at 10200us. Ideally I should be able to set "strobe_exposure=0" and be able to sync the start of expose with the duration/delay controls but I've found several issues. 

1) the strobe duration and strobe delay must match exactly or nothing happens.  Neither will change unless they are both set to the same setting. I need a good 7000us of on time for my opto to switch and my LEDs to hit full bright. Unfortunately this means I have 7000us of delay which the exposure must be set past to catch the strobe;

2)there should be some method of preempting the strobe output. I do not believe the trigger delays and such operate in framerate mode (trigger_mode=0), and even in trigger_mode it looks like the strobe_out follows the start of expose, not the trigger rise/fall.

Being able to set stobe_delay to a negative value or something  would be ideal since it would preempt the strobe output. Of course this would only work if the duration could be changed independently. So far the only way I've found to get anything going with the current strobe hardware is to invert the strobe_polarity (can expose at minimum. 25us), but then it would be on 90% of the time and probably overheat. Am I missing something or is the only way to get a useable strobe is to use the fastest switching method possible?

TIS-Tim, 2019-01-02:
I don't quite understand what you are seeing there.

Could you

(a) tell us exactly what settings you are using (resolution, framerate, exposure time, all trigger settings, all strobe settings)
(b) try using the Windows software and check whether the behavior is the same?
(c) capture the trigger/strobe signals with an oscilloscope and show us what signals you get depending on the settings you configure in the camera?

The camera does not support a negative strobe delay right now. You could work around that by making the external trigger pulse as long as you need the strobe pulse to be, and directly use that pulse as your strobe signal. You would then set the trigger delay so that the exposure time starts when your light is on.

blazini36, 2019-01-02:
This is what happens with the cameras running in trigger-mode=0 and strobe_exposure=0:
 I can have strobe delay and duration set to 0 and  I will receive no strobe output because my strobe needs longer on-times, . I can change either of those values and no actual change occurs, Listing the current values with either v4l2 or tcam will verify that the parameters have changed but the camera does not actually change the strobe behavior at all. It is only when I set both duration and delay to the exact same number that they both do actually change. I can set the duration long enough that it will turn my strobe on sufficiently, but the problem is that I have also had to set the delay the exact amount to make that work and I have now missed the exposure event unless the exposure is set for long times which defeats the purpose of setting strobe_exposure=0. I can set any strobe delay and duration value but the camera will not actually make the change unless they are both identical values. The camera will report back the value changes have occured, but the change only occurs in reality if they are both exactly the same value.

The changes I would have to make to the machine controlling the cameras to accomodate those tests would not at all be trivial. I also cannot use windows to test anything as my FPGA board Is controlled by the same Linux PC that the camera's work with. It will not work under windows. I designed the strobe to accommodate the open drain output from the camera whereas my FPGA outputs V+, making them incompatible.

I can try to capture the signals with a scope but it is visually obvious what happens. Is this not something you can verify on your side? I updated the firmwares from 78x to 94x (found on Github) with a windows PC and I will test that later.

blazini36, 2019-01-03:
with the new firmware (942) I am noticing a couple of new glitches as well as slightly different behavior with the above mentioned glitch. 

As a test (for my hardware) I wired my original solid state relay that I previously used as my strobe switching device to see if I could improve the delay time since I cannot properly tune it out with software. This did get the exposure time back down to 250us but only if I use strobe_exposure=1 as I cannot tune the strobe duration and delay properly. I notice the issue with this firmware (possibly existed in previous firmware but I've only had to make these adjustments recently) that the cameras do not allow the strobe_exposure=1 setting to actually take effect when strobe_duration and strobe_delay are set to the same value unless it is 0. UVC reports the set values but the camera operates differently. This issue effects both of my cameras but I have opened another issue regarding only one of the cameras, a problem I've had for some time but have just realized the actual problem.

blazini36, 2019-01-04:
Here's my experience:

I would agree on using strobe_exposure=0 to tune the delay out of my strobe if the duration, delay were working, that would be ideal.

When I switch back to the strobe switching method that has very minimal delay I can achieve exposure times of 250us or better but only if using strobe_exposure=1. I understand what you are saying about strobe_exposure=1 setting the duration of the strobe pulse, but it seems to do so without effecting the cameras delay setting. The actual issue is that I cannot set the strobe_duration without matching the strobe_delay to it when strobe_exposure is set to 0. This renders the ability to increase the duration completely pointless. So strobe_exposure=0 is not likely to help me at all unless this issue gets resolved. At least using Strobe_exposure=1 allows me to have a good exposure as long as I have a fast strobe.

The systems major intention is to be used in trigger mode. I allow a framerate mode for certain uses. High framerates with 2 cameras at full resolution eat up alot of processor and bandwidth. I really have no use for high framerates. The cameras must take quick pictures of a fast moving object with a trigger sync, which they are capable of doing well when tuned properly. 5fps is faster than I trigger in trigger mode.

TIS-Tim, 2019-01-04:
I can confirm there is definitely something wrong when trying to set strobe timing parameters in linux. I also get Strobe Duration/Delay only being changed when both have the same value.

I will investigate this.

TIS-Tim, 2019-01-07:
Turns out there is an issue in the firmware that prevents setting most combinations of strobe parameters through UVC.

Please create a support case at https://www.theimagingsource.com/support/contact/ so that I can send you a firmware update.

blazini36, 2019-01-25:
Has the updated firmware been created yet? I haven't received any info.

TIS-Stefan, 2019-01-28:
@Tim 
Hi Tim

das hattest Du bearbeitet.

Stefan

Yours sincerely / Mit freundlichen Grüßen 

Stefan Geissler / Stefan Geißler 

-------------------------------------------------------------------- 
The Imaging Source Europe GmbH 
Überseetor 18 
D-28217 Bremen 
Germany 
Tel   *49/421/33591-0 
Fax   *49/421/33591-80 
Registergericht: Amtsgericht Bremen HRB 11 927 
Geschäftsführer: Rolf Bollhorst  

-------------------------------------------------------------------- 

 On Fri, 25th Jan 2019 at 3:06 am, TheImagingSource/tiscamera <reply@reply.github.com> wrote: 

Has the updated firmware been created yet? I haven't received any info. 
—

  

View and manage this ticket online: https://theimagingsource.deskpro.com/tickets/KKNU-3245-RRZE 

  The Imaging Source [https://www.theimagingsource.com] · https://theimagingsource.deskpro.com/ 

 (#BCWY27RCKZSM3HACD5)

TIS-Stefan, 2019-01-28:
Hello

Did you create a support case as Tim mentioned?

Stefan

blazini36, 2019-01-28:
I sent the email through the website but did not get a response

blazini36, 2019-01-28:
You're correct, I missed that email.

I uploaded the new firmware to both cameras and I can now adjust the strobe duration independently of the delay. My method of driving the strobes is a bit different now since I reworked my strobe driver board to use gate driven mosfets rather than optos but I can get the exposure down to about 200us with the current light source