TheImagingSource/tiscamera #240: DFK33UX264 whitebalance mode settings stuck?

blazini36, 2019-01-03:
I've had this issue with this camera since I received it, I was only able to narrow it down to the camera itself now. I have 2 DFK33UX264's that I run typically under a custom program. I had previously assumed this was an issue with my program and needed to be weeded out but it only presented itself with 1 camera. Inside the programI typically execute a v4l2 control file with some initial parameters and then execute a program stored file that sets color correction parameters with v4l2.

1 Of the camera's no matter what seemed to be stuck in an auto white balance function, despite listing UVC parameters that all auto functions are disabled. I just spent some time with this camera in TCAM-capture and I realized that it does not behave like the other camera. It will constantly try to set the whitebalance temperature back to 4000 and the preset to "cool white flourescent light". Now I can play with the whitebalance toggles and eventually get it to stop doing this and allow me to set it to the desired whitebalance temperature of 7500. The problem is as soon as I shutdown the camera it will revert back to the original settings and have the same issue making it impossible for me to operate it by setting controls. I do not have any of these issues with the other camera. 

I also tried this camera with the windows software and had the same behavior but was able to get it unstuck there as well. The Technical manual States that there are 2 user settings that can be stored directly in the camera's memory but I cannot figure out how to access that to see if something is already stored for some reason, or if I can fix this issue by storing a setting. I saw nothing in the windows version on how to do this either. I just updated the firmware on both cameras from 781 to 942 and I've had the same issue on both firmwares, though I did not figure out the problem until after upgrading.

TIS-Tim, 2019-01-03:
Saving camera settings is not available though UVC, it is only available for USB3 Vision applications.

It is generally a good idea to always configure all required settings programmatically when opening a camera, because when users start other applications before yours, the settings might be changed from what the camera starts with.

Did you happen to save the current settings as startup settings for one of those cameras in IC Capture? You can undo that there by clicking Device->Properties->Startup->Restore Factory Default.

blazini36, 2019-01-04:
As I said, saving and implementing all settings is done at program load time and while running.

The thought of having a user set in hardware was an idea as a work around for the issue I'm experiencing. This camera powers on in la la land and I have to toggle controls to get it to snap out of it. Obviously I cannot do that programmatically.  Has anyone looked into kernel driver issues when running multiple cameras? This of course does not explain why I seen the same issue in IC Capture but then again I cannot say for certain the other camera does not do this in IC capture as I do not really use Windows at all. All I know is that the other camera does not do this at all.

As I said, I do not use Windows so I do not use IC capture so the startup settings in that program are not my issue. I only tried it because I'm sure the TIS member here would ask me to do so.

blazini36, 2019-01-06:
Can you do me a favor and plug 2 DFK33XU264's in and launch TCAM-capture? See if you have issues with whitebalance coming out of auto on one of the cameras

TIS-Stefan, 2019-01-09:
Hello

Unfortunately I have a pair of DFK 33UX249 only, but it uses the same firmware as the 264 camera.

It tried them in Windows and Linux and there is no issue disabling white balance on both of them. Changing the red value on one and green on the other camera in order to make sure white balance auto is turned off, showed, that the automatic is turned off on both cameras.

I ran one IC Capture instance with both cameras and in Linux two tcam-capture instances for each camera in parallel.

What do I miss?

How does your Linux code address the cameras? Are you sure you address the correct cameras? (Sorry for asking,)

Stefan

blazini36, 2019-01-09:
This is the typical scenario of what I am doing:
Udev rules are created to symlink the camera's serial number to a name. The kernel's "video0" link will never be reliable with multiple cameras. Say "mycam1" I don't have any issue addressing the correct camera for listing v4l2 controls or sending them, it's symlinked to the serial attribute.

The udev symlinks are the links my program reference. All v4l2 shellcalls go out to /dev/mycam1, /dev/mycam2.

I run a bash file for each camera at program start with v4l2 controls that I want to keep statically set, every control in that file has a 20ms delay before executing the next. I then run a set of user saved controls that are settable/savebale from inside the program, 20ms between the controls when that loads.

The delays are for piece of mind, I think I've exposed just about every bug I can with the cameras so I'm playing it safe. It is at the point where I only really see the whitebalance bug when cold booting the PC, I'll typically see it in 1 camera, with TCAM or my program.

Being that I see it in Tcam after a cold boot, prior to starting my program rules out my udev names or the command methods I use, Tcam doesn't use my symlinks and my control files haven't been executed yet. 

Now if I unplug both cameras and plug them back in while the PC is on, I can pretty much start either program and not see this issue. My Windows testing was very short so I can't comment much on that other than I have seen the issue. I have been weeding down the causes, because it almost seems like there are multiple things that aggravate it, but right now the above is exactly what I see

I also enable 2 sets of UVC quirks because the camera's are unstable without them, I can't keep them from dropping out otherwise. I have seen this issue without the quirks enabled as well. The cameras are very stable with these 2 quirks enabled but I still have this issue with whitebalance, usually just from cold boot.
UVC_QUIRK_FIX_BANDWIDTH
UVC_QUIRK_IGNORE_SELECTOR_UNIT
Are what I use. I just started using the selector unit quirk last week but I've had the whitebalance issue for months.

blazini36, 2019-01-17:
Honestly I've been trying to recreate the problem and lately I'm having a hard time getting it to happen. Figuring out the correct UVC quirks to apply and sticking a short delay in between the list of v4l2 commands seems to have helped immensely overall. It almost seems like in some circumstance the FPGA thinks it came out of whitebalance auto and reports back that it did so but it did not. The FPGA seems completely non-volatile so powering down or unplugging it doesnt really shake it out.

closed 2019-11-29 by TIS-Stefan