TheImagingSource/tiscamera #241: Raspberry PI 2 not able to access some camera properties, cannot set Trigger mode

CenekAlbl, 2019-01-14:
When trying to access some properties of the camera on a Raspberry PI 2, they are not available. One example is the Trigger mode, which I can't access. When I run tcam-capture I can see the camera, the videofeed and set several options, like exposure and white balance. However on my laptop I can control many more properties, including partial readout modes and trigger modes. I compiled the code on the Pi from the newest source.

This is a message I get in the console after starting tcam-capture.

libEGL warning: DRI2: failed to authenticate
qt5ct: using qt5ct plugin
14-01-2019:15:57:09,744 WARNING  [CapsDesc.py:167] Could not interpret structure. Omitting. video/x-raw, format=(string){ RGBx, xRGB, BGRx, xBGR, RGBA, ARGB, BGRA, ABGR }, width=(int)640, height=(int)480, framerate=(fraction){ 60/1, 30/1, 15/1, 5/1, 1/1 };
14-01-2019:15:57:09,746 WARNING  [CapsDesc.py:167] Could not interpret structure. Omitting. video/x-raw, format=(string){ RGBx, xRGB, BGRx, xBGR, RGBA, ARGB, BGRA, ABGR }, width=(int)1920, height=(int)1080, framerate=(fraction){ 15/1, 5/1, 1/1 };
14-01-2019:15:57:09,747 WARNING  [CapsDesc.py:167] Could not interpret structure. Omitting. video/x-raw, format=(string){ RGBx, xRGB, BGRx, xBGR, RGBA, ARGB, BGRA, ABGR }, width=(int)2048, height=(int)2048, framerate=(fraction){ 5/1, 1/1 };
14-01-2019:15:57:09,748 WARNING  [CapsDesc.py:167] Could not interpret structure. Omitting. video/x-raw, format=(string){ RGBx, xRGB, BGRx, xBGR, RGBA, ARGB, BGRA, ABGR }, width=(int)3072, height=(int)2048, framerate=(fraction){ 5/1, 1/1 };
14-01-2019:15:57:11,351 WARNING  [TcamView.py:482] get_tcam_property failed for 'Trigger Mode'
inotify_add_watch("/home/pi/.config/qt5ct") failed: "No such file or directory"

My main goal is to be able to enable Trigger mode and trigger the camera externally.

When trying to run the example Save an image on trigger in Python I get the following error:

tcambin serial="48810687" name=source ! video/x-raw,format=BGRx,width=640,height=480,framerate=30/1 ! videoconvert ! appsink name=sink
Traceback (most recent call last):
  File "Program.py", line 75, in <module>
    print("Gain Auto : %s " % Tis.Get_Property("Gain Auto").value)
  File "/home/pi/dev/Linux-tiscamera-Programming-Samples-bak/Save an image on trigger in Python/TIS.py", line 160, in Get_Property
    return CameraProperty(*self.source.get_tcam_property(PropertyName))

Other examples which use trigger work, but the Trigger mode True does not seem to apply, they just stream the frames continuously regardless of the trigger being sent.

TIS-Stefan, 2019-01-14:
Hello

usually the uvc extension xml file and is installed with the "sudo make install" process. I would like you to check, whether the xml is correcly installed. A documentation about this can be found at
https://github.com/TheImagingSource/tiscamera/wiki/UVC-Extension-Units

I would like you to send me the output of
`tcam-ctrl -p 48810687`

also

`tcam-ctrl -c 48810687`

would be nice.

Stefan

CenekAlbl, 2019-01-14:
Hi,

thanks for the quick reply.

The ouput of tcam-ctrl -p 48810678:
Found 7 propert(y/ies)
Brightness                              (int) min=0 max=4095 step=1 default=-8193 value=240 category=Exposure
Whitebalance Auto                       (bool) default=true value=true category=Color
Whitebalance Red                        (int) min=0 max=255 step=1 default=64 value=64 category=Color
Whitebalance Blue                       (int) min=0 max=255 step=1 default=64 value=64 category=Color
Gain                                    (int) min=0 max=480 step=1 default=57343 value=0 category=Exposure
Exposure Auto                           (bool) default=false value=true category=Exposure
Exposure                                (int) min=100 max=60000000 step=100 default=300 value=300 category=Exposure

and of tcam-ctrl -c 48810687:

Available gstreamer-1.0 caps:
video/x-bayer, format=(string)rggb, width=(int)640, height=(int)480, framerate=(fraction){ 60/1, 30/1, 15/1, 5/1, 1/1 };
video/x-bayer, format=(string)rggb, width=(int)1920, height=(int)1080, framerate=(fraction){ 15/1, 5/1, 1/1 };
video/x-bayer, format=(string)rggb, width=(int)2048, height=(int)2048, framerate=(fraction){ 5/1, 1/1 };
video/x-bayer, format=(string)rggb, width=(int)3072, height=(int)2048, framerate=(fraction){ 5/1, 1/1 };
video/x-bayer, format=(string)rggb16, width=(int)640, height=(int)480, framerate=(fraction){ 30/1, 15/1, 5/1, 1/1 };
video/x-bayer, format=(string)rggb16, width=(int)1920, height=(int)1080, framerate=(fraction){ 7/1, 5/1, 1/1 };
video/x-bayer, format=(string)rggb16, width=(int)2048, height=(int)2048, framerate=(fraction){ 2/1, 1/1 };
video/x-bayer, format=(string)rggb16, width=(int)3072, height=(int)2048, framerate=(fraction){ 2/1, 1/1 }

I will check the documentation about the xml file.

Thanks,

Cenek

TIS-Stefan, 2019-01-14:
Hi Cenek

thank you. You are right, many properties are missing. I wait for your result regarding the uvc extension.

Stefan

CenekAlbl, 2019-01-17:
Hi Stefan,

I tried the procedure described at https://github.com/TheImagingSource/tiscamera/wiki/UVC-Extension-Units without success. Unfortunately I cannot produce any meaningful results now, because both our cameras refuse to send any pictures due to broken connector. I need to wait for new cameras to arrive.

TIS-Stefan, 2019-01-17:
I already answered to you by email. I do not think, the connector is broken, because the cameras worked fine, when they left the office. 

Plugin the plug into the camera, until there was faint click sound. You may need some power doing that.

Stefan

closed 2019-11-29 by TIS-Stefan