TheImagingSource/tiscamera #242: Overexposure in trigger mode for GigE modules

haiwangcat, 2019-01-20:
I found that using the software trigger in the trigger mode gives me pictures with higher exposure than the video mode (non-trigger mode). I got this problem with the model DFK 33GX290e and DFK 33GX236. The issue can be directly reproduced in Tcam-Capture. The OS I use is Ubuntu 18.04. So any solutions for this issue?

TIS-Stefan, 2019-01-21:
Hello

If the IMX209 and IMX236 sensor is set to trigger mode, it runs in "Global Reset Shutter" mode. Therefore, the exposure time does not matter and the image becomes overexposed to the bottom.

This is standard behavior of this sensor, which can not be changed. Therefore, I close this ticket. 

If this is a ko-criteria for you using the cameras, then please contact our sales department for finding a solution.

Stefan

closed 2019-01-21 by TIS-Stefan

allentangff, 2019-01-25:
Hi Stefan,

How is the video mode able to overcome this problem (the Global Reset Shutter), and is there a way to snap images from the live video without having to display the actual video (no videosink)?