TheImagingSource/tiscamera #243: Record video approach

dernster, 2019-01-29:
Hi,

What would be a good approach for supporting live streaming and at the same time being able to record a video based on a user action? I'm wondering if the only way to achieve this would be by building a specific gstreamer pipeline.

My camera model is DFK Z12G445.

Thanks!

TIS-Stefan, 2019-01-30:
Hello 

You are right, building a suitable pipeline is the first step. Unfortunately I do not have such a pipeline sample, therefore (I do not really dare to suggest this) is searching in the internet for appropriate samples.
After the tcambin you add a "tee" filter, so you can split the pipeline into display and avi capture path.
I would not directly create a AVI capture directly after the "tee" module, because this is somewhat hard to control by user interaction. I would add an appsink. In the Appsink handler the images are forward to an "appsrc", if the user wants to capture AVI. I thought, we had a sample for that, but I can not find it any more. 

However, maybe your questions are more basic, so please feel free to ask.

Stefan

dernster, 2019-01-30:
Thanks for the response @TIS-Stefan, so basically the idea would be to split the main live streaming pipeline using a tee that goes to an appsink. From there, if I receive a "start capturing" event from the user I would start feeding another pipeline (just for recording) which its src element is an appsrc element right?

TIS-Stefan, 2019-01-30:
Hi

Yes, that would be the general approach I would use.

Which programming language do you prefer? C++ or Pyhton?

Stefan

TIS-Stefan, 2019-01-30:
Ok,

you may start with the samples https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples

Stefan

dernster, 2019-01-30:
Thank you @TIS-Stefan

closed 2019-01-30 by dernster

TIS-Stefan, 2019-01-30:
Come back to me, when we shall go into the coding details.