TheImagingSource/tiscamera #244: uvc camera not detected in tcamsrc plugin

kirtikk, 2019-01-30:


kirtikk, 2019-01-30:
When i try to execute the below pipeline on ubuntu 16.04 with gstreamer 1.0 installed, camera is not being detected by tcamsrc plugin. 
1.gst-launch-1.0 tcamsrc ! tcamautoexposure ! videoconvert ! xvimagesink
DeviceIndex::update_device_list ()NUmber of connected devices is :63093312
NUmber of connected devices is :63093312
BackendLoader::get_device_list_from_backend
retrieving list for libtcam-v4l2.so
Amount of devices: 0
BackendLoader::get_device_list_from_backend
retrieving list for libtcam-libusb.so
Amount of devices: 0
Number of found devices: 0Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
Setting pipeline to NULL ...
Freeing pipeline ...

2.Also tried with the v4l2src using the below mentioned pipeline:
sudo GST_DEBUG=3 gst-launch-1.0 v4l2src device=/dev/video0 ! tcamautoexposure ! videoconvert ! xvimagesink

Output :
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
0:00:00.225910409 21069      0x10800a0 WARN          v4l2bufferpool gstv4l2bufferpool.c:546:gst_v4l2_buffer_pool_set_config:<v4l2src0:pool:src> libv4l2 converter detected, disabling CREATE_BUFS
0:00:01.734688312 21069      0x10800a0 ERROR                default tcamgstbase.cpp:53:tcam_gst_find_camera_src: Camera source not set!
0:00:01.734706431 21069      0x10800a0 ERROR       tcamautoexposure gsttcamautoexposure.cpp:1798:gst_tcamautoexposure_transform_ip: Could not find source element
0:00:01.734725157 21069      0x10800a0 WARN                 basesrc gstbasesrc.c:2939:gst_base_src_loop:<v4l2src0> error: Internal data stream error.
0:00:01.734729911 21069      0x10800a0 WARN                 basesrc gstbasesrc.c:2939:gst_base_src_loop:<v4l2src0> error: streaming stopped, reason error (-5)
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
gstbasesrc.c(2939): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason error (-5)
Execution ended after 0:00:01.631537060
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

3.Output of  "tcam-ctrl -l"  
Output :-
DeviceIndex::update_device_list ()
NUmber of connected devices is :-1485387264
NUmber of connected devices is :-1485387264
BackendLoader::get_device_list_from_backend
retrieving list for libtcam-v4l2.so
Amount of devices: 0
BackendLoader::get_device_list_from_backend
retrieving list for libtcam-libusb.so
Amount of devices: 0
Number of found devices: 0
No devices found.

Could anyone please give your inputs...

kirtikk, 2019-01-31:
Hi Stefan,
I am using an UVC compliant USB camera from a different manufacturer called logitec..

TIS-Stefan, 2019-02-01:
Hi 

our modules are made for use with our cameras, not with cameras of other manufacturers. Therefore, I am sorry, but your nice Logitec can not be used with out modules. You may start with the v4l2src instead,

Stefan

closed 2019-02-01 by TIS-Stefan