TheImagingSource/tiscamera #245: Camera software not working on Raspberry Pi

userinv, 2019-01-30:
Hi Stefan, 

I've been working on a project which involves a lidar and a DMX 33UX252.
When the lidar detects a distance greater than 4m, a software trigger is
triggered and executes a call back function to take pictures.

The code that has been written for the camera works as expected on a laptop
running `Ubuntu 18.04.1 LTS (Bionic Beaver)`. However, when I try running the same program on the Raspberry Pi running on a `Raspbian GNU/Linux 9 (stretch)`, the camera starts taking pictures immediately and the function "" which returns TRUE if the software trigger has been correctly asserted returns FALSE. I am assuming that the issue is related to the system the software is running on. Maybe I need additional packages to make the software compatible with the Pi?

Moreover, when the camera is initializing, the following error is printed on the console: `(vod_proj:2152): GStreamer-CRITICAL **: gst_object_unref: assertion 'object != NULL' failed`

The initialization phase is just ensuring the trigger mode and other properties are set to the appropriate setting.

I have also included a print out of all the properties of the camera when it is printed from the laptop and the Raspberry Pi.

**Here is the code which sets the software trigger:**
```
while(true)
{
	double distance = lidar.getDistance();
	std::cout << distance <<std::endl;
	buffer[arrayIndex] = distance;
	bool valuesAreSame = allEqual(buffer);

	if(valuesAreSame)
	{
		std::cout << "Values are all same" <<std::endl;
		break;
	}
	else if(distance > 4.0 && distance < 9000.0 && !valuesAreSame)
	{
		//Initiate a software trigger to capture a frame of data
		bool initiateTrigger = camera.setPropertyValue("Software Trigger", softwareTrigger, bool(TRUE));

	       if(!initiateTrigger)
		{
			std::cout << "Could not trigger" << std::endl;
		}
	}

	if(arrayIndex == (ARRAY_SIZE - 1))
	{
		arrayIndex = 0;
	}
	else
	{
		arrayIndex++;
	}
}
```
**Camera properties outputted from the laptop**
```
Properties:
integer property 'Brightness': value = 240 default = 240 min = 0 max = 4095 step_size = 1
integer property 'Gamma': value = 100 default = 100 min = 1 max = 500 step_size = 1
integer property 'Gain': value = 0 default = 0 min = 0 max = 480 step_size = 1
integer property 'Exposure': value = 333 default = 333 min = 20 max = 4000000 step_size = 1
boolean property 'Exposure Auto': value = false default_value = true
integer property 'Exposure Auto Reference': value = 128 default = 128 min = 0 max = 255 step_size = 1
boolean property 'Gain Auto': value = true default_value = true
boolean property 'Trigger Mode': value = true default_value = false
button property 'Software Trigger': value = false default_value = false
integer property 'Trigger Delay (us)': value = 31 default = 0 min = 0 max = 1000000 step_size = 1
boolean property 'Strobe Enable': value = true default_value = false
boolean property 'Strobe Polarity': value = false default_value = false
boolean property 'Strobe Exposure': value = true default_value = true
integer property 'Strobe Duration': value = 100 default = 100 min = 0 max = 32767 step_size = 1
integer property 'Strobe Delay': value = 0 default = 0 min = 0 max = 32767 step_size = 1
integer property 'GPOut': value = 0 default = 0 min = 0 max = 1 step_size = 1
integer property 'GPIO': value = 0 default = 0 min = 0 max = 1 step_size = 1
integer property 'Offset X': value = 0 default = 0 min = 0 max = 1792 step_size = 2
integer property 'Offset Y': value = 0 default = 0 min = 0 max = 1532 step_size = 2
boolean property 'Offset Auto Center': value = true default_value = true
enum property 'Trigger Polarity': value = 'RisingEdge' default_value = 'RisingEdge' values = {FallingEdge, RisingEdge, }
enum property 'Trigger Exposure Mode': value = 'FrameStart' default_value = 'FrameStart' values = {ExposureActive, FrameStart, }
integer property 'Trigger Burst Count': value = 10 default = 1 min = 1 max = 1000 step_size = 1
integer property 'Trigger Debounce Time (us)': value = 0 default = 0 min = 0 max = 1000000 step_size = 1
integer property 'Trigger Mask Time (us)': value = 0 default = 0 min = 0 max = 1000000 step_size = 1
integer property 'Trigger Noise Surpression Time (us)': value = 0 default = 0 min = 0 max = 1000000 step_size = 1
boolean property 'Reverse X': value = false default_value = false
boolean property 'Reverse Y': value = false default_value = false
```
**Camera properties outputted from the Pi**
```
Properties:
integer property 'Brightness': value = 240 default = -8193 min = 0 max = 4095 step_size = 1
integer property 'Gamma': value = 100 default = 57343 min = 1 max = 500 step_size = 1
integer property 'Gain': value = 0 default = 57343 min = 0 max = 480 step_size = 1
boolean property 'Exposure Auto': value = false default_value = true
boolean property 'Gain Auto': value = true default_value = true
integer property 'Brightness Reference': value = 128 default = 128 min = 0 max = 255 step_size = 1
integer property 'Exposure Min': value = 0 default = 0 min = 0 max = 0 step_size = 1
integer property 'Exposure Max': value = 0 default = 0 min = 0 max = 0 step_size = 1
double property 'Gain Min': value = 0.000000 default = 0.000000 min = 0.000000 max = 480.000000 step_size = 1.000000
double property 'Gain Max': value = 480.000000 default = 480.000000 min = 0.000000 max = 480.000000 step_size = 1.000000
integer property 'Exposure ROI Left': value = 0 default = 0 min = 0 max = 631 step_size = 1
integer property 'Exposure ROI Width': value = 640 default = 640 min = 8 max = 640 step_size = 1
integer property 'Exposure ROI Top': value = 0 default = 0 min = 0 max = 471 step_size = 1
integer property 'Exposure ROI Height': value = 480 default = 480 min = 8 max = 480 step_size = 1
```

TIS-Stefan, 2019-01-31:
Hello Userinv

I appreciate, that you gave the lists of available properties, they confirm your suspect of a missing packet. I suppose, there is the UVC Extension missing, so you do not get the Software Trigger Property.  I would like you to follow the steps documented at https://github.com/TheImagingSource/tiscamera/wiki/UVC-Extension-Units

With tcam-ctrl -p <serialnumber> you will get also a list camera properties. If all is correct, then "Software Trigger" should be listed. 

I used a Rasperian on the PI had always had the software trigger. Maybe the uvcdynctrl package is not installed by default on Debian 9.

Stefan
--

userinv, 2019-01-31:
The UVC extension was already installed and I followed the instructions as written in the link you provided. I ran the both commands `sudo uvcdynctrl -i tisEUVC.xml -d /dev/video0` and `sudo uvcdynctrl -i usb3.xml -d /dev/video0` but here are the errors it outputted.

**Executed `sudo uvcdynctrl -i tisEUVC.xml -d /dev/video0`**
```
pi@raspberrypi:~/Documents/projects/tiscamera/data/uvc-extensions $ sudo uvcdynctrl -i tisEUVC.xml -d /dev/video0

[libwebcam] Unknown V4L2 control ID encountered: 0x0199E201

[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E201, name = 'Exposure Time (us)'

[libwebcam] Unknown V4L2 control ID encountered: 0x0199E202

[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E202, name = 'Auto Shutter'

[libwebcam] Unknown V4L2 control ID encountered: 0x0199E203

[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E203, name = 'Auto Exposure Reference'

[libwebcam] Unknown V4L2 control ID encountered: 0x0199E205

[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E205, name = 'Gain Auto'

[libwebcam] Unknown V4L2 control ID encountered: 0x0199E208

[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E208, name = 'Trigger Mode'

[libwebcam] Unknown V4L2 control ID encountered: 0x0199E209

[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E209, name = 'Software Trigger'

[libwebcam] Unknown V4L2 control ID encountered: 0x0199E210

[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E210, name = 'Trigger Delay'

[libwebcam] Unknown V4L2 control ID encountered: 0x0199E211

[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E211, name = 'Strobe Enable'

[libwebcam] Unknown V4L2 control ID encountered: 0x0199E212

[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E212, name = 'Strobe Polarity'

[libwebcam] Unknown V4L2 control ID encountered: 0x0199E213

[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E213, name = 'Strobe Exposure'

[libwebcam] Unknown V4L2 control ID encountered: 0x0199E214

[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E214, name = 'Strobe Duration'

[libwebcam] Unknown V4L2 control ID encountered: 0x0199E215

[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E215, name = 'Strobe Delay'

[libwebcam] Unknown V4L2 control ID encountered: 0x0199E216

[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E216, name = 'GPOUT'

[libwebcam] Unknown V4L2 control ID encountered: 0x0199E217

[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E217, name = 'GPIN'

[libwebcam] Unknown V4L2 control ID encountered: 0x0199E218

[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E218, name = 'ROI Offset X'

[libwebcam] Unknown V4L2 control ID encountered: 0x0199E219

[libwebcam] Invalid or unsupported V4L2 control encountered: ctrl_id = 0x0199E219, name = 'ROI Offset Y'

[... 112 lines cut ...]
usb3.xml: error: video0: unable to map 'Auto Functions ROI Top' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)

usb3.xml: error: video0: unable to map 'Auto Functions ROI Width' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)

usb3.xml: error: video0: unable to map 'Auto Functions ROI Height' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)

usb3.xml: error: video0: unable to map 'Whitebalance Mode' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)

usb3.xml: error: video0: unable to map 'White Balance Temperature' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)

usb3.xml: error: video0: unable to map 'Whitebalance Temperature Preset' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)

usb3.xml: error: video0: unable to map 'White Balance Green Component' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)

usb3.xml: error: video0: unable to map 'White Balance Auto Preset' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)

usb3.xml: error: video0: unable to map 'Exposure Auto Lower Limit' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)

usb3.xml: error: video0: unable to map 'Exposure Auto Upper Limit' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)

usb3.xml: error: video0: unable to map 'Exposure Auto Upper Limit Auto' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)

usb3.xml: error: video0: unable to map 'Override Scanning Mode' control. ioctl(UVCIOC_CTRL_MAP) failed with return value -1 (error 2: No such file or directory)
```
What are the next steps from here?

TIS-Stefan, 2019-01-31:
The next step is simple:
Ignore the shown properties, that are not supported by your camera. These message above a usual.

Then do tcam-ctrl -p <serialnumber> and check, whether Software Trigger is avialabe.

userinv, 2019-01-31:
When I manually run the .xml file for usb3, it works fine but I followed the instructions to make the .xml file run automatically but it does not work when i restart the Pi with the camera connected. Is there anything else I can try?

userinv, 2019-01-31:
I added `sudo uvcdynctrl -i /usr/share/uvcdynctrl/data/199e/usb3.xml -d /dev/video0` to the end of .bashrc so the .xml file runs at startup and it works now. Thank you for the help!

TIS-Stefan, 2019-01-31:
Very well!
I close the case now. If you have any more questions, please feel free to ask, I try to help.

closed 2019-01-31 by TIS-Stefan