TheImagingSource/tiscamera #246: Fresh Ubuntu install - tcam-capture not working (or any other code)

rrittgarn, 2019-02-05:
Hello, working with a DMK 33GX178e that works properly from windows, however linux code is not working. 

Installed a fresh image of Ubuntu 18.04 on my laptop, installed required packages from readme. Installed OpenCV from source with extras and when running the CPP example Saving images using OpenCV I get these errors:

https://pastebin.com/z8WzZfup

I have manually set the GST Plugins directory (which resolved errors about gst binaries not found). Attempted both packaged version (per README) and built GStreamer from source. I am getting issues with properties being read from the camera in all applications, including tcam capture. 

tcam-capture errors: https://pastebin.com/WchYC7uK

Guessing there is a missing dependency or required version of something missing from documentation as I had this issue before re-installing Ubuntu.

TIS-Stefan, 2019-02-05:
Hello Ryan

I am sorry for asking, but did you follow the the building steps documented on our Readme at https://github.com/TheImagingSource/tiscamera?

Stefan

rrittgarn, 2019-02-05:
Yes, as stated in original post, I used the readme. I copied and pasted all required packages. I believe that list is incomplete because I know for some of the examples you require GTK-3.0, however that library is not included in your required packages. 

Please let me know what information i can provide for further troubleshooting. Again, this was a fresh install of Ubuntu that I copied and pasted the instructions from the readme and it does not work. 

Thanks

TIS-Stefan, 2019-02-05:
Ryan

one test please
Does
gst-launch-1.0 tcambin ! videoconvert ! ximagesink

work on the terminal?

Stefan

rrittgarn, 2019-02-05:
No, i get a blank output with the following messages in console: 

$ gst-launch-1.0 tcambin ! videoconvert ! ximagesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
INFO:
Working with src caps: video/x-raw, framerate=(fraction)19/2, width=(int)3072, height=(int)2048, format=(string)GRAY16_LE
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source: Device lost
Additional debug info:
/home/ryan/projects/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp(1045): gst_tcam_src_device_lost_callback (): /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source
Execution ended after 0:00:21.265382708
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

TIS-Stefan, 2019-02-05:
Which computer model do you use?
Does the green LED light on the camera?

rrittgarn, 2019-02-05:
The Camera works from windows using your IC Capture app. It is a GigE camera, so the PC is irrelevant. 

It is an XPS13 Developer edition running Ubuntu 18.04.

TIS-Stefan, 2019-02-05:
Please excuse me asking. I asked, because the reported error happens, if the camera is connected to a 100mBit network. Therefore, I asked for the green LED, that indicates 1GB speed.

rrittgarn, 2019-02-05:
Yes, it is plugged into a juniper EX-2200-24P

It shows green currently.

TIS-Stefan, 2019-02-05:
ok, ruled one possible problem out.

You should not get a device lost, when starting the live video, obviously, even if 16 bit video formats are used.
So next step....

`gst-launch-1.0 tcambin ! video/x-raw, format=BGRx, width=640, height=480, framerate=30/1 !  videoconvert ! ximagesink
`
Makes a small colored video format. I am not sure, whether BGRx can be exchanged against GRAY8.

Please let me know the result.

Sorry for all this guessing around.

rrittgarn, 2019-02-06:
small blank box - no image. 

Here's the console output:  https://pastebin.com/ivubGQFi

Thanks

rrittgarn, 2019-02-06:
I was able to get a C++ program to at least give me additional information: 

double property 'DeviceTemperature': value = 58.500000 default = 58.500000 min = -179769313486231570814527423731704356798070567525844996598917476803157260780028538760589558632766878171540458953514382464234321326889464182768467546703537516986049910576551282076245490090389328944075868508455133942304583236903222948165808559332123348274797826204144723168738177180919299881250404026184124858368.000000 max = 179769313486231570814527423731704356798070567525844996598917476803157260780028538760589558632766878171540458953514382464234321326889464182768467546703537516986049910576551282076245490090389328944075868508455133942304583236903222948165808559332123348274797826204144723168738177180919299881250404026184124858368.000000 step_size = 0.001000

I believe this is what tcam-capture is choking on, but I still don't have image output. 

Thanks

rrittgarn, 2019-02-06:
Console output on your next suggestion:  https://pastebin.com/HecWpx27

Still blank box. 

To verify that my gstreamer is working, I tested with my webcam and was successful with this pipeline: gst-launch-1.0 v4l2src ! videoconvert ! autovideosink  

Tried replacing ximagesink with autovideosink and no luck there either.

TIS-Edgar, 2019-02-06:
There seems to be a negotiation error in the tcambin. This prevents GRAY8/GRAY16_LE -> BGRx, which is something tcam-capture implicitly expects to work.

    gst-launch-1.0 tcamsrc ! video/x-raw, format=GRAY16_LE, width=640, height=480, framerate=30/1 ! videoconvert ! ximagesink

This pipeline should work. If not I would require the output of 

```
export TCAM_LOG=DEBUG
gst-launch-1.0 --gst-debug=tcamsrc:5 tcamsrc ! video/x-raw, format=GRAY16_LE, width=640, height=480, framerate=30/1 ! videoconvert ! ximagesink
```

With a 33 mono camera all auto-adjustmenst are in-camera so the bin offers little/to no advantage for you.
As a potential workaround you could try the development branch of the upcoming 0.11.0 release, as this issue seems to be resolved there.

rrittgarn, 2019-02-07:
top pipeline did not work. 

Without Sudo getting permissions errors on the gige service pid file: 
https://pastebin.com/YzaSPL4z

With Sudo: https://pastebin.com/wmuhYqgZ

Is there anything special I need to do to try out the development branch?

TIS-Edgar, 2019-02-07:
Simply call `git checkout development`. 
Please uninstall tiscamera before doing that. It is also recommended to use a new build folder.

Can you give me your procedure for building gstreamer?
All our tests show that a GRAY16 pipeline should be running, and with gstreamer installed from the repositories it definitely runs. The custom gstreamer _may_ cause the issues, but I would like to verify that before saying anything with certainty.

rrittgarn, 2019-02-07:
I actually re-installed Ubuntu again thinking the same thing. I only have Gstreamer from packages now - no custom build. OpenCV is installed and working (as far as i can tell thus far). 

I did the following: 

git checkout development
git pull
cd tiscamera/build
sudo make uninstall 
cd ..
rm -rf build
mkdir build
cd build
cmake -DBUILD_ARAVIS=ON -DBUILD_GST_1_0=ON -DBUILD_TOOLS=ON -DBUILD_V4L2=ON -DCMAKE_INSTALL_PREFIX=/usr ..
make 
sudo make install

then trying the
gst-launch-1.0 tcamsrc ! video/x-raw, format=GRAY16_LE, width=640, height=480, framerate=30/1 ! videoconvert ! ximagesink
Pipeline I am still seeing errors, but I see video for the first time in this whole lovely ordeal. 

I still cannot launch tcam-capture successfully - complains of disconnect.

TIS-Edgar, 2019-02-07:
Good to hear, that at least a basic stream is working,

tcam-capture is a bit aggressive when it comes to device-lost events. Those are typically triggered by errors reported by the backend such as too many missing images, missing heartbeats, etc. 

A common cause may be a low mtu. You can change it with
`sudo ip link set dev eth0 mtu 9000` where eth0 should be replaced with your interface in question.

If that does not resolve the problems I would require a new log of the current situation.

```
export TCAM_LOG=DEBUG
gst-launch-1.0 --gst-debug=tcamsrc:5 tcamsrc ! video/x-raw,format=GRAY16_LE,width=640,height=480,framerate=30/1 ! videoconvert ! ximagesink
```

tcam-capture should give a somewhat identical log, but stop the stream after the first error.

rrittgarn, 2019-02-08:
Looking into MTU, my switch is not configured for jumbo frames, and I was not aware it was a requirement of the GIGE cameras. I have verified proper MTU and am using a different USB NIC on my laptop successfully for tcam-capture only on the Development Branch though. master does not yield images.

rrittgarn, 2019-02-08:
Additionally, now trying to build the latest linux examples, i am getting YAML related errors. 

What version of yaml-cpp should be installed?

rrittgarn, 2019-02-08:
ryan@ryan-XPS-13-9343:~/projects/Linux-tiscamera-Programming-Samples/Simple Snap Image in C++/build$ make
[ 25%] Linking CXX executable simple-snapimage
CMakeFiles/simple-snapimage.dir/home/ryan/projects/tiscamera/examples/cpp/common/tcamcamera.cpp.o: In function `gsttcam::TcamCamera::store_device_state(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
/home/ryan/projects/tiscamera/examples/cpp/common/tcamcamera.cpp:690: undefined reference to `YAML::operator<<(std::ostream&, YAML::Node const&)'
CMakeFiles/simple-snapimage.dir/home/ryan/projects/tiscamera/examples/cpp/common/tcamcamera.cpp.o: In function `gsttcam::TcamCamera::restore_device_state(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
/home/ryan/projects/tiscamera/examples/cpp/common/tcamcamera.cpp:697: undefined reference to `YAML::LoadFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
CMakeFiles/simple-snapimage.dir/home/ryan/projects/tiscamera/examples/cpp/common/tcamcamera.cpp.o: In function `YAML::detail::memory_holder::create_node()':
/usr/include/yaml-cpp/node/detail/memory.h:39: undefined reference to `YAML::detail::memory::create_node()'
CMakeFiles/simple-snapimage.dir/home/ryan/projects/tiscamera/examples/cpp/common/tcamcamera.cpp.o: In function `YAML::detail::node_ref::mark_defined()':
/usr/include/yaml-cpp/node/detail/node_ref.h:28: undefined reference to `YAML::detail::node_data::mark_defined()'
CMakeFiles/simple-snapimage.dir/home/ryan/projects/tiscamera/examples/cpp/common/tcamcamera.cpp.o: In function `YAML::detail::node_ref::set_null()':
/usr/include/yaml-cpp/node/detail/node_ref.h:33: undefined reference to `YAML::detail::node_data::set_null()'
CMakeFiles/simple-snapimage.dir/home/ryan/projects/tiscamera/examples/cpp/common/tcamcamera.cpp.o: In function `YAML::detail::node_ref::set_scalar(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
/usr/include/yaml-cpp/node/detail/node_ref.h:34: undefined reference to `YAML::detail::node_data::set_scalar(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
CMakeFiles/simple-snapimage.dir/home/ryan/projects/tiscamera/examples/cpp/common/tcamcamera.cpp.o: In function `YAML::detail::node_ref::begin()':
/usr/include/yaml-cpp/node/detail/node_ref.h:43: undefined reference to `YAML::detail::node_data::begin()'
CMakeFiles/simple-snapimage.dir/home/ryan/projects/tiscamera/examples/cpp/common/tcamcamera.cpp.o: In function `YAML::detail::node_ref::end()':
/usr/include/yaml-cpp/node/detail/node_ref.h:48: undefined reference to `YAML::detail::node_data::end()'
CMakeFiles/simple-snapimage.dir/home/ryan/projects/tiscamera/examples/cpp/common/tcamcamera.cpp.o: In function `YAML::Node::Scalar[abi:cxx11]() const':
/usr/include/yaml-cpp/node/impl.h:152: undefined reference to `YAML::detail::node_data::empty_scalar[abi:cxx11]'
CMakeFiles/simple-snapimage.dir/home/ryan/projects/tiscamera/examples/cpp/common/tcamcamera.cpp.o: In function `YAML::Node::AssignData(YAML::Node const&)':
/usr/include/yaml-cpp/node/impl.h:250: undefined reference to `YAML::detail::memory_holder::merge(YAML::detail::memory_holder&)'
CMakeFiles/simple-snapimage.dir/home/ryan/projects/tiscamera/examples/cpp/common/tcamcamera.cpp.o: In function `YAML::as_if<bool, void>::operator()() const':
/usr/include/yaml-cpp/node/impl.h:116: undefined reference to `YAML::convert<bool>::decode(YAML::Node const&, bool&)'
CMakeFiles/simple-snapimage.dir/home/ryan/projects/tiscamera/examples/cpp/common/tcamcamera.cpp.o: In function `YAML::detail::node& YAML::detail::node_data::get<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, boost::shared_ptr<YAML::detail::memory_holder>)':
/usr/include/yaml-cpp/node/detail/impl.h:112: undefined reference to `YAML::detail::node_data::convert_to_map(boost::shared_ptr<YAML::detail::memory_holder>)'
/usr/include/yaml-cpp/node/detail/impl.h:126: undefined reference to `YAML::detail::node_data::insert_map_pair(YAML::detail::node&, YAML::detail::node&)'
CMakeFiles/simple-snapimage.dir/home/ryan/projects/tiscamera/examples/cpp/common/tcamcamera.cpp.o: In function `YAML::detail::node& YAML::detail::node_data::convert_to_node<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, boost::shared_ptr<YAML::detail::memory_holder>)':
/usr/include/yaml-cpp/node/detail/impl.h:171: undefined reference to `YAML::detail::memory_holder::merge(YAML::detail::memory_holder&)'
collect2: error: ld returned 1 exit status
CMakeFiles/simple-snapimage.dir/build.make:161: recipe for target 'simple-snapimage' failed
make[2]: *** [simple-snapimage] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/simple-snapimage.dir/all' failed
make[1]: *** [CMakeFiles/simple-snapimage.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

TIS-Edgar, 2019-02-08:
The package libyaml-cpp-dev should contain all required dependencies for that example.

rrittgarn, 2019-02-08:
That is the package I have installed and am now getting the YAML issues and cannot compile them. Is there a bug related to this in the development branch?

TIS-Edgar, 2019-02-11:
It is not. The example compiles as expected on Ubuntu 18.04. 
Could you try a new build directory for the example? I am currently suspecting a caching issue in cmake.

rrittgarn, 2019-02-12:
Followed the same steps as above for cleaning and recreating - i cannot build the example files without a YAML related error message

TIS-Edgar, 2019-02-12:
I found the problem. It was that the example failed to link against yaml-cpp. 
The fixes have been pushed to the development branch of the Samples repository.
```
cd ~/projects/Linux-tiscamera-Programming-Samples/Simple Snap Image in C++/build
git pull
git checkout development
make
```

closed 2019-11-29 by TIS-Stefan