TheImagingSource/tiscamera #248: DFK72BUC02 in Ubuntu 16.04

NarSB, 2019-02-11:
I want to use the camera DFK 72BUC02 in ubuntu 16.04. I followed the instructions and the installation seems correct but:

When I tried to change the camera firmware to dfk72uc02_3012.euvc says:
Firmware Size: 19968 EEPROM Size: 16384
Firmware does not fit in EEPROM. Aborting!

Using the default firmawe and using the command tcam-ctrl -l , I got the message:
No device found.

What should I do to use the camera?

TIS-Edgar, 2019-02-15:
Hi,

Sorry for the late reply. 
There are two types of dfk72 cameras in circulation. One has a bigger EEPROM.
You will have to use the previous firmware file dfk72uc02_158.euvc. 
The rest of the instructions remains identical.

NarSB, 2019-02-15:
Thank you for your replay.

I tried the firmware and I can use the camera but, with the tcam-capture command shows the image in gray. The camera in Windows has color. Is there another firmware that preserves de color?

TIS-Edgar, 2019-02-15:
To verify you used the dfk72uc02_158.euvc file and not the dmk72uc02_158.euvc right?
Do applications like guvcview show a color image?

NarSB, 2019-02-15:
I have tested the firmware with two cameras:

./firmware-update -l
Found 2 device(s).

Name           - ID        - Serialnumber
DFx 72BUC02    - 199e:8207 - 49010288
DFx 72BUC02    - 199e:8307 - 49010289

In the 49010288 tested with the guvcview, it shows the image in gray. When I change the output to RGB3 or BGB3, it returns the following error

libv4l2: error set_fmt gave us a different result then try_fmt!
Segmentation fault (core dumped)

The 49010289 is not detected with this firmware.

TIS-Edgar, 2019-02-18:
What does `uname -a` say? 

Can you give me the output of `sudo firmware-update -i -d <SERIAL>` for both cameras?

NarSB, 2019-02-18:
sudo ./firmware-update -i -d 49010289 
Device manufacturer: The Imaging Source Europe GmbH
Product name:        DFx 72BUC02
Serial number:       49010289
VendorID:ProductID:  199e:8207
Firmware version:    158
UVC mode is:         off
Camera EEPROM size:  16384

sudo ./firmware-update -i -d 49010288 
Device manufacturer: The Imaging Source Europe GmbH
Product name:        DFx 72BUC02
Serial number:       49010288
VendorID:ProductID:  199e:8207
Firmware version:    158
UVC mode is:         off
Camera EEPROM size:  16384

TIS-Edgar, 2019-02-19:
You need to toggle the uvc flag for them as is described in the instructions.

sudo firmware-update -d <SERIAL> -m uvc

Then both cameras should be listed.

With the current 4.15 kernel for Ubuntu 16.04 a color image is available. 
Are you using an older kernel? What does 'uname -a ' say?

NarSB, 2019-02-21:
That was it. Thank you now they work perfectly.

TIS-Edgar, 2019-02-21:
That is great to hear. If you have any further questions, please let us know.

closed 2019-02-21 by TIS-Edgar