TheImagingSource/tiscamera #250: Docker container

dernster, 2019-02-14:
Hello! I'm running a container that successfully compiles everything but when I'm try to see the cameras available using `camera-ip-config -l` it doesn't show up, whereas running it outside the container (on the actual machine) it works. Do you see any limitation on the way the cameras are discovered when using it from a docker container?

Thanks!

dernster, 2019-02-14:
Maybe I need to expose a specific port on the container?

dernster, 2019-02-14:
Ok, I answered my own question. Running the container using the parameter `--net = "host"` solves the issue. Probably because when using this parameter there's no need to expose specific ports, and the discovering process of the cameras probably uses a specific incoming port or even a random one?

TIS-Edgar, 2019-02-15:
Hi,

Sorry for the late reply.

Ports are random. Be it for discovery, configuration or streaming. Should you have multiple application that act independently from eachother you would see some conflicts there if they would all use the same ports.

closed 2019-02-15 by dernster