TheImagingSource/tiscamera #251: Multiple docker containers issue

dernster, 2019-02-21:
Hello,

I have a host machine running ubuntu 16.04 and docker containers based on ubuntu 18.04. In my local network there are two cameras DFK Z12G445 connected (50810617 & 50810620).

1. If I run both instances of my program on the host machine, it works just fine. I get stream from both cameras (with and without the gige-daemon.service running).
2. If I run just a single camera on one docker container. It works, no matter which camera I choose.
3. If I run two containers with one camera on each one, I'm able to get stream only from one of them (it seems to be consistently from 50810617).

On each container I'm setting the network mode as "host" which means the container will share the networking layer with the host machine. I had to do that because of random ports used by the SDK (I cannot just expose a couple of fixed ports).

Another note to consider could be that docker containers don't run services/daemons by design (not sure if this helps to determine the issue).

I've attached the logs (using TCAM_LOG=DEBUG) [here](https://gist.github.com/dernster/f530b81d98c08c859cdb844cac706cd7)

It seems that the first difference between both cameras is:
```
camera_50810617_1  | 573641     <INFO> /tiscamera/src/PipelineManager.cpp:502: Pipeline creation successful.
camera_50810617_1  | 573663     <INFO> /tiscamera/src/PipelineManager.cpp:512: source ->  sink
camera_50810617_1  | 573680     <INFO> /tiscamera/src/ImageSink.cpp:51: Pipeline started playing
camera_50810617_1  | 575880     <INFO> /tiscamera/src/aravis/AravisDevice.cpp:588: Starting actual stream...

camera_50810620_1  | 378766     <INFO> /tiscamera/src/PipelineManager.cpp:502: Pipeline creation successful.
camera_50810620_1  | 378770     <INFO> /tiscamera/src/PipelineManager.cpp:512: source ->  sink
camera_50810620_1  | 378773     <INFO> /tiscamera/src/ImageSink.cpp:51: Pipeline started playing
camera_50810620_1  | 378909     <ERROR> /tiscamera/src/aravis/AravisDevice.cpp:555: Unable to create ArvStream.
camera_50810620_1  | 378914     <ERROR> /tiscamera/src/ImageSource.cpp:61: Unable to start stream from device.
```

Could it be related to the random ports selection on both containers? Is there any chance that both containers choose the same ports in this dockerized setup? 

Thank you!

TIS-Edgar, 2019-02-22:
Hi,
I am not the familiar with docker setups, but since both container instances ask the host system for ports I would suspect, that the ports will be different since they come from the same source.

An easy way to see what is happening is to create a wireshark log showing us exactly what is happening. If both container use the same port it will be visible there.

If you are not comfortable posting the log here send it directly to support@theimagingsource.com and reference this issue.

dernster, 2019-02-26:
Hi @TIS-Edgar, thanks for the response. I've sent you an email :)

closed 2019-07-04 by TIS-Edgar