TheImagingSource/tiscamera #254: DFG USB2Pro can not be detected on Ubuntu 16.04 with the newest kernel version 4.15

Xiaoyu0601-Wang, 2019-03-04:
I followed the instructions to install the tiscamera. 
I plugged in the DFG USB2Pro. Then I used the following command to test:
tcam-ctrl -l
But the system told me "No devices found.".

Is there any other configuration that I should do to make it work?

Michael.

TIS-Stefan, 2019-03-04:
Hello Michael

You need Kernel 4.18 or 4.19 (not 4.20). These kernel versions have the driver for DFG/USB2.pro built in.

You tiscamera modules can show the DFG/USB2pro, but unfortunately the video norm and input channel properties are missing. 

Stefan

labqw123, 2019-03-04:
We have same issue on NVIDIA Jetson TX2, Ubuntu 16.04, kernel version: 4.4. If I understand you right we can't make it work on that device?

Xiaoyu0601-Wang, 2019-03-05:
> Hello Michael
> 
> You need Kernel 4.18 or 4.19 (not 4.20). These kernel versions have the driver for DFG/USB2.pro built in.
> 
> You tiscamera modules can show the DFG/USB2pro, but unfortunately the video norm and input channel properties are missing.
> 
> Stefan

@TIS-Stefan 

I updated the kernel to 4.19 in Ubuntu 16.04, the DFG USB2pro can be found now.
> Missing input properties is fine.
However, after DFG USB2pro is connected in UI "tcam-capture", I still can not get image of camera from the UI in Ubuntu.
btw: The hardwre works fine in windows.

Do you have any idea about this issue?

Many thanks,
Michael.

Xiaoyu0601-Wang, 2019-03-05:
> Hello Michael
> 
> You need Kernel 4.18 or 4.19 (not 4.20). These kernel versions have the driver for DFG/USB2.pro built in.
> 
> You tiscamera modules can show the DFG/USB2pro, but unfortunately the video norm and input channel properties are missing.
> 
> Stefan

@TIS-Stefan 

I updated the kernel to 4.19 in Ubuntu 16.04, the DFG USB2pro can be found now.
Missing input properties is fine.
However, after DFG USB2pro is connected in UI "tcam-capture", I still can not get image of camera from the UI in Ubuntu.
btw: The hardwre works fine in windows.

Do you have any idea about this issue?

Many thanks,
Michael.

Xiaoyu0601-Wang, 2019-03-05:
@TIS-Stefan 

And another issue:
hidetoshi@hidetoshi-ThinkPad:~$ gst-launch-1.0 tcambin ! video/x-raw, format=BGRA,width=640,height=480,framerate=30/1 ! videoconvert ! ximagesink 
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
Setting pipeline to NULL ...
Freeing pipeline ...

Here, I also have tried with sudo, but had the same result. How can I make this work?

Michael

TIS-Stefan, 2019-03-05:
Hello Michael

You may try guvcview for the DFG/USB2pro. 
In tcam-capture you may see a blue screen, which is generated by the DFG/USB2pro, if no camera is connected or synchronization failed, because e.g. the video norm does not fit to the camera.

I do not recommend to use the tcambin. Better is the v4l2src.

Xiaoyu0601-Wang, 2019-03-13:
@TIS-Stefan 

I tried guvcview, it didn't work.
I'm trying to use ros video_stream_opencv package to open the video.
Here, if the analog video is not connected, by running the ros package, I can get the blue screen.
However, if I connect the analog video, it gives me an error "select timeout", and I can not get the video stream.

Is there any solution to make it work by running ros package?

Michael

TIS-Stefan, 2019-03-13:
Hello Michael

`Is there any solution to make it work by running ros package?`

The NVidia TX2 is somewhat tricky to use, since they use an own Kernel and own drivers. I would like to suggest to try the DFG/USB2pro on a desktop computer first.

Also my experiences with ROS are extremely limited, so you may ask the ROS comunity. I do not have ROS installed on my Linux computers.

Stefan

TIS-Stefan, 2019-03-13:
Hi Michael

I should apologize for my last answer, because it was not really helpful.
Therefore, I connected a DFG/USB2pro to my desktop computer, searched an analogue (oh oh) camera and tried again. This time, I used qv4l2. All is working fine. qv4l2 needs QT to be installed, so you may do not do this on your TX2, if you do not want to use QT.

Next step is checking OpenCV, the cv::Videocapture class in particular. I will write a small C++ program for testing, but I think, setting video norm and input channel is not supported. However, I will see, how far I can get.  If this works, you may go in contact with the ROS poeple and ask, how to implement the needed properties of DFG/USB2pro.

In worst case, we have to write an own v4l2 based program, that gives access to all camera properties, but this is not too hard.

So, I get a coffee now and start hacking afterwards.

Stefan

TIS-Stefan, 2019-03-13:
Very nice. While trying to get libavcodec-ffmpeg.so.56, which is missing somehow, I destroyed my Linux. Thus, you may need to wait a few days for an answer.

TIS-Stefan, 2019-03-13:
So, got it working. I had too much OpenCV mismatch on my computer.

I did not find any settings for input channel (Composite and S-VHS) and video norm (NTSC and PAL-B) in the documentation of cv::VideoCapture. But you can set these properties with v4l2-ctl (paket v4l-utils), before starting your OpenCV application.

The possible intput channel settings are
v4l2-ctl --set-input 0   // that is Compsite
v4l2-ctl --set-input 1  // that is S-VHS

The possible video norms, depending on your analogue camera are:
v4l2-ctl -s PAL-B
v4l2-ctl -s NTSC

After configuring this, you can start an OpenCV cv::VideoCapture using program and you should see the live video.
Now I am curios to know, whether ros video_stream_opencv shows the video.

Stefan

Xiaoyu0601-Wang, 2019-03-13:
@TIS-Stefan 

Thank you so much for your information.
Sorry for not making it clear, I'm also using a desktop PC with Ubuntu 16.04, not TX2.

>>I did not find any settings for input channel (Composite and S-VHS) and video norm (NTSC and PAL-B) in the documentation of cv::VideoCapture. But you can set these properties with v4l2-ctl (paket v4l-utils), before starting your OpenCV application.

This is exactly the part that stuck me for a long time. Thanks a lot.

I will test DFG USB2Pro with the analog video again tomorrow according to your comments.
Since the ros package is also based on OpenCV, I think with a little modification, the ros video_stream_opencv package will also work. 

I will let you know my result after the test. 

Many thanks,
Michael.

TIS-Stefan, 2019-03-13:
Michael
I would simply set the input channel and video norm in a terminal and start the ROS package. Maybe this works, if so, the ROS package does not need to be modified at all.

Stefan

Xiaoyu0601-Wang, 2019-03-14:
@TIS-Stefan 

I have tested the DFG USB2Pro with a NTSC analog camera. It works well with ros video_stream_opencv package. only need to change the video port according to the /dev/video*.

>>The possible intput channel settings are
>>v4l2-ctl --set-input 0 // that is Compsite
>>v4l2-ctl --set-input 1 // that is S-VHS
>>The possible video norms, depending on your analogue camera are:
>>v4l2-ctl -s PAL-B
>>v4l2-ctl -s NTSC

Here, the full commands are:
v4l2-ctl -d * --set-input 0 // that is Compsite
v4l2-ctl -d *--set-input 1 // that is S-VHS
v4l2-ctl -d * -s PAL-B
v4l2-ctl -d * -s NTSC
where * is the video port number.

By typing "v4l2-ctl -d /dev/video* --all", it give:
hidetoshi@hidetoshi-ThinkPad:~$ v4l2-ctl -d /dev/video2 --all
Driver Info (not using libv4l2):

So I think this may be the reason that ubuntu guvcview and qv4l2 UI can't be used for test the DFG USB2Pro directly.

I think the content in this issue can be put in the Readme to help others with the same problem.

Really thanks for the help recently.
Michael.

TIS-Stefan, 2019-03-14:
Hi Michael

Thank you for the good news about ROS.

Usually qv4l2 and guvcview allow to select the /dev/video* files, except there is something wrong with one of them. 

Stefan

closed 2019-03-22 by Xiaoyu0601-Wang

0x3333, 2020-08-25:
I know this is an old issue, but I'd like to leave here information that may help others. The DFG/USB2pro sometimes doesn't work on Linux because if you don't initialize the companion audio device before starting getting frames from the video device, it will not work. Just init the audio device(Alsa) and read a single frame, then, start streaming from the video.

This is somewhat a quirk in the Conexant CX231XX driver in Linux. I will work on understanding more clearly this issue, but it's already a start.

Here is a command that will work flawless:

```
mencoder \
-noskip \
tv:///0 \
-tv norm=NTSC\
:device=/dev/video0\
:driver=v4l2\
:outfmt=yuv2\
:width=720:height=480\
:fps=29.97\
:alsa\
:adevice=hw.0,0\
:amode=1\
:buffersize=200 \
-ffourcc divx \
-oac mp3lame \
-lameopts cbr:preset=128 \
-ovc lavc \
-lavcopts vcodec=mpeg4\
:vbitrate=1400\
:threads=2\
:autoaspect \
-vf dsize=4/3 \
-vf-add scale \
-vf-add kerndeint=10:0:0:1:1 \
-vf-add softskip,harddup \
-endpos 02:10:00 \
-o ~/output.avi
```

Jeffson-wu, 2023-01-20:
@0x3333 @Xiaoyu0601-Wang 
I am facing the same problem on ubutu 18.04 that qv4l2 UI is not working with DFG/USB2pro, how should I try for next?

0x3333, 2023-01-20:
> @0x3333 @Xiaoyu0601-Wang
> I am facing the same problem on ubutu 18.04 that qv4l2 UI is not working with DFG/USB2pro, how should I try for next?

As I said in my comment. If you don't initialize the audio device it will not work. There are  some apps that does initialize, but some don't. I believe that that is the case in your app.

Jeffson-wu, 2023-01-20:
how to initialize the audio device?qv4l2 UI does not initialize the audio device before capture?

0x3333, 2023-01-20:
> how to initialize the audio device?qv4l2 UI does not initialize the audio device before capture?

Well, I don't know this software, but if it is not working, probably doesn't. In my case I use the camera in a software I build, so I initialize alsa device, in your case I don't know which is better. Try another player.