TheImagingSource/tiscamera #255: make install failed: cmake error "file INSTALL given no DESTINATION"

phamlehuy53, 2019-03-06:


phamlehuy53, 2019-03-06:
[  1%] Built target pugixml_obj
[ 10%] Built target tcam-base
[ 19%] Built target tcam
[ 21%] Built target tcamprop
[ 23%] Built target create_gobject
[ 31%] Built target tcam-algorithms
[ 33%] Built target tcamgstbase
[ 34%] Built target gsttcambin
[ 36%] Built target gsttcamsrc
[ 37%] Built target gsttcamwhitebalance
[ 39%] Built target gsttcamautofocus
[ 41%] Built target gsttcamautoexposure
[ 43%] Built target gsttcambiteater
[ 45%] Built target tcam-dfk73
[ 49%] Built target tcam-v4l2
[ 58%] Built target tcam-libusb
[ 61%] Built target tcam-ctrl
[ 91%] Built target 33update
[ 98%] Built target firmware-update
[100%] Built target dfk73udev
Install the project...
-- Install configuration: ""
-- Installing: /usr/lib/libtcam.so.0.10.0
-- Up-to-date: /usr/lib/libtcam.so.0
-- Set runtime path of "/usr/lib/libtcam.so.0.10.0" to "/usr/lib/tcam-0"
-- Up-to-date: /usr/lib/libtcam.so
-- Installing: /usr/lib/libtcamprop.so.0.1
-- Set runtime path of "/usr/lib/libtcamprop.so.0.1" to ""
-- Up-to-date: /usr/lib/libtcamprop.so
-- Installing: /usr/share/gir-1.0/Tcam-0.1.gir
-- Installing: /usr/lib/x86_64-linux-gnu/girepository-1.0/Tcam-0.1.typelib
-- Installing: /usr/include/tcamprop.h
-- Installing: /usr/lib/libtcamgstbase.so
-- Set runtime path of "/usr/lib/libtcamgstbase.so" to ""
-- Installing: /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamsrc.so
-- Set runtime path of "/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamsrc.so" to ""
-- Installing: /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamwhitebalance.so
-- Set runtime path of "/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamwhitebalance.so" to ""
-- Installing: /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamautoexposure.so
-- Set runtime path of "/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamautoexposure.so" to ""
-- Installing: /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamautofocus.so
-- Set runtime path of "/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamautofocus.so" to ""
-- Installing: /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcambiteater.so
-- Set runtime path of "/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcambiteater.so" to ""
-- Installing: /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcambin.so
-- Set runtime path of "/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcambin.so" to ""
-- Installing: /usr/lib/libtcam-algorithms.so.0.10.0
-- Up-to-date: /usr/lib/libtcam-algorithms.so.0
-- Up-to-date: /usr/lib/libtcam-algorithms.so
-- Installing: /usr/lib/tcam-0/libtcam-v4l2.so.0.10.0
-- Up-to-date: /usr/lib/tcam-0/libtcam-v4l2.so.0
-- Set runtime path of "/usr/lib/tcam-0/libtcam-v4l2.so.0.10.0" to ""
-- Up-to-date: /usr/lib/tcam-0/libtcam-v4l2.so
-- Installing: /usr/lib/libtcam-dfk73.so.0.10.0
-- Up-to-date: /usr/lib/libtcam-dfk73.so.0
-- Up-to-date: /usr/lib/libtcam-dfk73.so
-- Installing: /usr/lib/tcam-0/libtcam-libusb.so.0.10.0
-- Up-to-date: /usr/lib/tcam-0/libtcam-libusb.so.0
-- Set runtime path of "/usr/lib/tcam-0/libtcam-libusb.so.0.10.0" to ""
-- Up-to-date: /usr/lib/tcam-0/libtcam-libusb.so
-- Installing: /usr/lib/pkgconfig/tcam.pc
-- Up-to-date: /usr/share/tiscamera/images
-- Installing: /usr/share/tiscamera/images/logo.jpg
-- Installing: /usr/share/tiscamera/images/23uc256.png
-- Installing: /usr/share/uvcdynctrl/data/199e/tisEUVC.xml
-- Installing: /usr/share/uvcdynctrl/data/199e/usb3.xml
-- Installing: /usr/lib/udev/rules.d/80-theimagingsource-cameras.rules
-- Installing: /usr/bin/tcam-ctrl
-- Set runtime path of "/usr/bin/tcam-ctrl" to ""
-- Installing: /usr/bin/tcam-capture
CMake Error at tools/tcam-capture/cmake_install.cmake:45 (file):
  file INSTALL given no DESTINATION
Call Stack (most recent call first):
  tools/cmake_install.cmake:43 (include)
  cmake_install.cmake:45 (include)

make: *** [Makefile:86: install] Error 1

TIS-Stefan, 2019-03-06:
Hello 

I have no idea, what you did or not.

Questions:
Did you change something on the code?
How did you configure the project with cmake?

phamlehuy53, 2019-03-06:
> Hello
> 
> I have no idea, what you did or not.
> 
> Questions:
> Did you change something on the code?
> How did you configure the project with cmake?

i'm using ubuntu 18.10 cosmic. All required packages are ok.
when i ran cmake, it looks like:

cmake -DBUILD_ARAVIS=OFF -DBUILD_GST_1_0=ON -DBUILD_TOOLS=ON -DBUILD_V4L2=ON -DCMAKE_INSTALL_PREFIX=/usr ..
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
-- Found GObject: /usr/include/glib-2.0  
-- Found PYTHON_MODULE_PyQt5: /home/huy/anaconda3/lib/python3.5/site-packages/PyQt5  
**Traceback (most recent call last):
  File "<string>", line 1, in <module>
IndexError: list index out of range**
-- 
-- Build gstreamer-1.0 plugins:   ON
-- Support for GigE via aravis:   OFF
-- Support for USB cameras:       ON
-- Support for LibUsb cameras:    ON
-- Build additional utilities:    ON
-- 
-- Installation prefix:           /usr
-- Installing binaries to:        /usr/bin
-- Installing libraries to:       /usr/lib
-- Installing header to:          /usr/include
-- Installing udev rules to:      /usr/lib/udev/rules.d
-- Installing systemd units to:   /lib/systemd/system
-- Installing uvcdynctrl rules to:/usr/share/uvcdynctrl/data/199e
-- Installing desktop files to:   /usr/share/applications
-- Installing gstreamer-1.0 to:   /usr/lib/x86_64-linux-gnu/gstreamer-1.0
-- Installing gobject-introspection to: /usr/share/gir-1.0
-- Installing introspection typelib to: /usr/lib/x86_64-linux-gnu/girepository-1.0
-- 
-- Configuring done

phamlehuy53, 2019-03-06:
running `make` was ok
then `sudo make install` raised the issue like above.
thank you for help!

TIS-Stefan, 2019-03-07:
Hello
sudo make install can not rise an cmake error like
```
Traceback (most recent call last):
File "", line 1, in 
IndexError: list index out of range
```
Also I guess, there is an issue with Python Anaconda, but I do not know, that causes the tcam-capture issue. I do have have these problems, neither on Ubuntu 16.04 nor 18.04, not on Arm, Odroid and x86.

So lets switch to C++, in case your Python causes the issue. Please compile and try the sample from 

https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/tree/master/QT5%20Device%20Selection%20and%20Property%20Dialogs%20in%20C%2B%2B

Stefan

phamlehuy53, 2019-03-09:
Thank you so much. The anaconda python caused the issues. Then i solved it

phamlehuy53, 2019-03-09:
Then other issue raised up, that configuring ethernet on ubuntu was not easy. So can you help me with this. Thanks a lot!

TIS-Stefan, 2019-03-11:
Hello

> Then other issue raised up, that configuring ethernet on ubuntu was not easy. So can you help me with this. 
Do you have some details about what you want to do for me, so I can try to show, how to configure a network in Ubuntu? Also did you search in internet for Ubuntu network configuration?

Stefan

phamlehuy53, 2019-03-12:
hi **TIS-Stefan**. First, i `ifconfig -a`

**eno1: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether f0:1f:af:02:2a:70  txqueuelen 1000  (Ethernet)
        RX packets 21  bytes 1344 (1.3 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 375  bytes 36124 (36.1 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 20  memory 0xf7e00000-f7e20000**  

**lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 7347  bytes 628227 (628.2 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 7347  bytes 628227 (628.2 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0**

**wlp3s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.38  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::ef8:364b:fd50:7bfe  prefixlen 64  scopeid 0x20<link>
        ether 3c:a9:f4:13:64:cc  txqueuelen 1000  (Ethernet)
        RX packets 29464  bytes 29815636 (29.8 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 18737  bytes 2803030 (2.8 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0**

then `sudo ifconfig wlp3s0 down; sudo ifup eno1` to turn off wifi and use ethernet only.
i manually configured eno1 `sudo nano /etc/network/interfaces` like this:

**# interfaces(5) file used by ifup(8) and ifdown(8)
auto lo
iface lo inet loopback
auto eno1
iface eno1 inet static
    address 192.168.0.20
    netmask 255.255.255.0
    broadcast 192.168.0.255
    gateway 192.168.0.1**

but **eno1** connection status was always idle or disconnected.
That is my issue.
Thanks!

TIS-Stefan, 2019-03-12:
Hello

What is the output of 
`camera-ip-conf -l`
on the terminal?

Do you have an DHCP server?

Stefan

phamlehuy53, 2019-03-14:
hi Stefan!
i'm not sure about what was going on my computer. So can we talk more about my issue through gmail, twitter, facebook...
Thank you a lot!

TIS-Stefan, 2019-03-14:
Hi

You can start here: https://theimagingsource.deskpro.com/

TIS-Stefan, 2019-03-14:
Hello 

Anyway, please let me know, what  is the output of
camera-ip-conf -l

Thank you in advance!

Stefan

closed 2019-11-29 by TIS-Stefan