TheImagingSource/tiscamera #257: ROS Multicam publishes the left images to both left and right topics

eloibove, 2019-03-07:
I am using Ubuntu 16.04 with ROS kinetic. I have two DFK33UX252 plugged using USB3.0 and I am using the tis_rosstarter_multicam script with the following configuration:

first_cam = tiscamera.Camera(
    args.left_serial, 640, 480, 15, True, False, "camera_left_raw", "left_camera_node")
if args.right_serial:
    second_cam = tiscamera.Camera(
        args.right_serial, 640, 480, 15, True, False, "camera_right_raw", "right_camera_node")

I get the following output: 

topic_name: camera_left_raw
node_name: left_camera_node
self.topic_name: /camera/image_raw:=/camera_left_raw
self.node_name: __name:=left_camera_node
tcambin serial="49810271" name=source ! video/x-raw,format=BGRx,width=640,height=480,framerate=15/1 ! videoconvert ! video/x-raw,format=RGB ,width=640,height=480,framerate=15/1! shmsink socket-path=/tmp/ros_mem
topic_name: camera_right_raw
node_name: right_camera_node
self.topic_name: /camera/image_raw:=/camera_right_raw
self.node_name: __name:=right_camera_node
tcambin serial="49810042" name=source ! video/x-raw,format=BGRx,width=640,height=480,framerate=15/1 ! videoconvert ! video/x-raw,format=RGB ,width=640,height=480,framerate=15/1! shmsink socket-path=/tmp/ros_mem
[ INFO] [1551983386.406055550]: Using gstreamer config from env: "shmsrc socket-path=/tmp/ros_mem ! video/x-raw-rgb, width=640,height=480,framerate=15/1,bpp=24,depth=24,blue_mask=16711680, green_mask=65280, red_mask=255 ! ffmpegcolorspace"
[ INFO] [1551983386.409297873]: using default calibration URL
[ INFO] [1551983386.409427189]: camera calibration URL: file:///home/eloibove/.ros/camera_info/camera.yaml
[ INFO] [1551983386.409506153]: Unable to open camera calibration file [/home/eloibove/.ros/camera_info/camera.yaml]
[ WARN] [1551983386.409542856]: Camera calibration file /home/eloibove/.ros/camera_info/camera.yaml not found.
[ INFO] [1551983386.409557627]: Loaded camera calibration from 
[ INFO] [1551983386.419324103]: Time offset: 1551981247,153
Press Enter to end program[ INFO] [1551983391.076837304]: Publishing stream...
[ INFO] [1551983391.076973877]: Started stream.
[ INFO] [1551983391.412461487]: Using gstreamer config from env: "shmsrc socket-path=/tmp/ros_mem ! video/x-raw-rgb, width=640,height=480,framerate=15/1,bpp=24,depth=24,blue_mask=16711680, green_mask=65280, red_mask=255 ! ffmpegcolorspace"
[ INFO] [1551983391.415606048]: using default calibration URL
[ INFO] [1551983391.415670674]: camera calibration URL: file:///home/eloibove/.ros/camera_info/camera.yaml
[ INFO] [1551983391.415757741]: Unable to open camera calibration file [/home/eloibove/.ros/camera_info/camera.yaml]
[ WARN] [1551983391.415773293]: Camera calibration file /home/eloibove/.ros/camera_info/camera.yaml not found.
[ INFO] [1551983391.415803639]: Loaded camera calibration from 
[ INFO] [1551983391.425506977]: Time offset: 1551981247,153
[ INFO] [1551983391.480167571]: Publishing stream...
[ INFO] [1551983391.480269640]: Started stream.

[ INFO] [1551983415.306737075]: Stream ended.
[ INFO] [1551983415.306845680]: Stopping gstreamer pipeline...
[ INFO] [1551983415.307904268]: GStreamer stream stopped!
[ INFO] [1551983415.307977018]: Cleaning up stream and exiting...
Program ended

I attach also a screen capture
[image]

TIS-Stefan, 2019-03-08:
Thank you very much, that looks good to me.

eloibove, 2019-03-08:
Hi Stefan, sorry I didn't explain properly... The images published to the topics /left_camera_raw and /right_camera_raw come from the same camera, it seems that one of the cameras is not being used!

TIS-Stefan, 2019-03-08:
Aha. I think, that is pointing to my question regarding the shared memory file used in tiscamera.py. For both pipelines you use the same and I asked, whether that works. I do not remember getting an answer, however, I am getting old and memory...

You may enhance the tiscamera.py and exchange
```
p += ' ! videoconvert ! video/x-raw,format=RGB ,width=%d,height=%d,framerate=%d/1! shmsink socket-path=/tmp/ros_mem' % (
                width, height, framerate,)

```
against something
```
p += ' ! videoconvert ! video/x-raw,format=RGB ,width=%d,height=%d,framerate=%d/1! shmsink socket-path=/tmp/ros_mem_%s' % (
                width, height, framerate,serial)

```
Somewhat down you have
```
 # Create gscam_config variable with content
        gscam = 'shmsrc socket-path=/tmp/ros_mem ! video/x-raw-rgb, width=%d,height=%d,framerate=%d/1' % (
            width, height, framerate,)
```
which should be enhanced to

```
 # Create gscam_config variable with content
        gscam = 'shmsrc socket-path=/tmp/ros_mem_%s ! video/x-raw-rgb, width=%d,height=%d,framerate=%d/1' % (
            width, height, framerate,serial)

```
So the gscam pipeline knows, which pipeline accesses which shared memory file.
However, we run into the next problem: there is only one gscam environment variable. I am not sure, how to solve this issue, that is a question for the ROS gscam author. I am very sorry for that.

Stefan

eloibove, 2019-03-08:
Hi Stefan, I tried what you suggested and unfortunately it didn't work... I will keep working on it and will keep you updated! Thank you

TIS-Stefan, 2019-03-08:
Very well.

Do you have an idea, how to solve the issue with the one any only gscam environment varialbe, which allows to use one pipeline only?

Stefan

eloibove, 2019-03-08:
I saw that another user had a similar problem as mine and I am currently trying to adapt his code to run with ROS but I am not sure whether it is going to work... What do you think?

(https://github.com/TheImagingSource/tiscamera/issues/206)

TIS-Stefan, 2019-03-11:
Hello 

I do not think, that helps, because we ended in #206 in an external software for stereo capture, that has no relation to ROS.
I suggest, you may look into the ROS forums and ask for how to use two pipelines with gscam.

Stefan

closed 2019-07-04 by TIS-Edgar