TheImagingSource/tiscamera #259: Problem with ROS example code

NovoG93, 2019-03-11:
Hello,

When I run ./tis_rosstarter the following error occurs.

> workstation@tauworkstation:~/git/tiscamera/examples/ROS$ ./tis_rosstarter 
topic_name: None
node_name: None
self.topic_name: 
self.node_name: __name:=tis_camera_node
tcambin serial="07510213" name=source ! video/x-raw,format=BGRx,width=1280,height=960,framerate=258/1 ! tee name=t t. ! queue ! videoconvert ! video/x-raw,format=RGB ,width=1280,height=960,framerate=258/1! shmsink socket-path=/tmp/ros_mem t. ! queue ! videoconvert ! ximagesink
Press Enter to end programTraceback (most recent call last):
  File "./tis_rosstarter", line 23, in <module>
    input("Press Enter to end program")
EOFError
[ INFO] [1552295769.936847183]: Using gstreamer config from env: "shmsrc socket-path=/tmp/ros_mem ! video/x-raw-rgb, width=1280,height=960,framerate=258/1,bpp=24,depth=24,blue_mask=16711680, green_mask=65280, red_mask=255 ! ffmpegcolorspace"
[ INFO] [1552295769.940035700]: using default calibration URL
[ INFO] [1552295769.940071932]: camera calibration URL: file:///home/workstation/.ros/camera_info/camera.yaml
[ INFO] [1552295769.940114530]: Unable to open camera calibration file [/home/workstation/.ros/camera_info/camera.yaml]
[ WARN] [1552295769.940131113]: Camera calibration file /home/workstation/.ros/camera_info/camera.yaml not found.
[ INFO] [1552295769.940144022]: Loaded camera calibration from 
[ INFO] [1552295769.950435767]: Time offset: 1552295323,286
Speicherzugriffsfehler (Speicherabzug geschrieben)
[FATAL] [1552295770.242880616]: Failed to PAUSE stream, check your gstreamer configuration.
[FATAL] [1552295770.242910113]: Failed to initialize gscam stream!

Output of dmesg:

> [  210.136107] usb 1-4: new high-speed USB device number 5 using xhci_hcd
> [  211.813515] usb 1-4: New USB device found, idVendor=199e, idProduct=8308
> [  211.813520] usb 1-4: New USB device strings: Mfr=2, Product=1, SerialNumber=3
> [  211.813523] usb 1-4: Product: DMK 42AUC03
> [  211.813527] usb 1-4: Manufacturer: The Imaging Source Europe GmbH
> [  211.813529] usb 1-4: SerialNumber: 07510213
> [  211.827948] media: Linux media interface: v0.10
> [  211.830644] Linux video capture interface: v2.00
> [  211.835304] uvcvideo: Found UVC 1.00 device DMK 42AUC03 (199e:8308)
> [  213.833914] uvcvideo 1-4:1.0: Entity type for entity Extension 4 was not initialized!
> [  213.833920] uvcvideo 1-4:1.0: Entity type for entity Processing 3 was not initialized!
> [  213.833923] uvcvideo 1-4:1.0: Entity type for entity Camera 1 was not initialized!
> [  213.834193] input: DMK 42AUC03 as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/input/input14
> [  213.834456] usbcore: registered new interface driver uvcvideo
> [  213.834458] USB Video Class driver (1.1.2.realsense-1.3.4)
> [  213.846810] uvcvideo: Failed to query (GET_LEN) UVC control 5 on unit 4: -32 (exp. 2).
> [  213.847383] uvcvideo: Failed to query (GET_LEN) UVC control 11 on unit 4: -32 (exp. 2).
> [  310.017811] show_signal_msg: 12 callbacks suppressed
> [  310.017816] tcambin-source:[4018]: segfault at 7f1994125494 ip 00007f1994125494 sp 00007f19817f9a10 error 14 in libgsttcamsrc.so[7f1994366000+33000]
> [  376.268903] tcambin-source:[4208]: segfault at 7fd63a71f494 ip 00007fd63a71f494 sp 00007fd624d44a10 error 14
> [  410.634600] uvcvideo: Non-zero status (-75) in status completion handler.

**Setup:**

- Linux tauworkstation 4.15.0-45-generic #48~16.04.1-Ubuntu SMP Tue Jan 29 18:03:48 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
- ROS Kinetic
- Build the tiscamera package with ARAVIS (as described in [https://github.com/TheImagingSource/tiscamera#building-tiscamera](url) )
- Tried binary and source install of gscam

**Used files**:
(without '.txt: extension of course)
*tis_rosstarter'
[tis_rosstarter.txt](https://github.com/TheImagingSource/tiscamera/files/2951913/tis_rosstarter.txt)

*tiscamera.py*
[tiscamera.py.txt](https://github.com/TheImagingSource/tiscamera/files/2951908/tiscamera.py.txt)

TIS-Stefan, 2019-03-11:
Hello

The DMK 42AUC03 does not support 258 frames per second. Thus please exchange it against 30, because the maximum fps of this camera is 30 frames per second.

Just to make sure, the Linux firmware is installed into the camera? Please see https://github.com/TheImagingSource/tiscamera/wiki/Getting-Started-with-USB-2.0-CMOS-Cameras

NovoG93, 2019-03-12:
Thank you for your quick response.

258 was a typo, changed it to 30 and also tried 25 still same error, see below.
>workstation@tauworkstation:~/git/tiscamera/examples/ROS$ ./tis_rosstarter                                               topic_name: None                                                                                                        node_name: None                                                                                                         self.topic_name:                                                                                                        self.node_name: __name:=tis_camera_node                                                                                 tcambin serial="07510213" name=source ! video/x-raw,format=GRAY8,width=1280,height=960,framerate=30/1 ! videoconvert ! video/x-raw,format=RGB ,width=1280,height=960,framerate=30/1! shmsink socket-path=/tmp/ros_mem                           Press Enter to end programTraceback (most recent call last):                                                              File "./tis_rosstarter", line 23, in <module>                                                                             input("Press Enter to end program")                                                                                 EOFError                                                                                                                [ INFO] [1552374410.503495283]: Using gstreamer config from env: "shmsrc socket-path=/tmp/ros_mem ! video/x-raw-rgb, width=1280,height=960,framerate=30/1,bpp=24,depth=24,blue_mask=16711680, green_mask=65280, red_mask=255 ! ffmpegcolorspace"[ INFO] [1552374410.506498556]: using default calibration URL                                                           [ INFO] [1552374410.506529717]: camera calibration URL: file:///home/workstation/.ros/camera_info/camera.yaml           [ INFO] [1552374410.506568641]: Unable to open camera calibration file [/home/workstation/.ros/camera_info/camera.yaml] [ WARN] [1552374410.506583465]: Camera calibration file /home/workstation/.ros/camera_info/camera.yaml not found.       [ INFO] [1552374410.506598179]: Loaded camera calibration from                                                          [ INFO] [1552374410.516788943]: Time offset: 1552374173,282                                                             Speicherzugriffsfehler (Speicherabzug geschrieben)                                                                      [FATAL] [1552374410.785761867]: Failed to PAUSE stream, check your gstreamer configuration.                             [FATAL] [1552374410.785806680]: Failed to initialize gscam stream! 

Output of sudo ./firmware-update -id 07510213: 
> Device manufacturer: The Imaging Source Europe GmbH
Product name:        DMK 42AUC03
Serial number:       07510213
VendorID:ProductID:  199e:8308
Firmware version:    3014
UVC mode is:         on
Camera EEPROM size:  32768

When I run the .<tiscamera>/examples/python/live_video.py I see the live video.
Also the cameras worked like 4 weeks ago and now they suddenly dont.

TIS-Stefan, 2019-03-12:
Hello 

Please try in the terminal:

gst-launch-1.0 tcambin serial="07510213" !  videoconvert ! ximagesink

If this does not not work, make sure, you built the tiscamera repository correctly and made also the 
sudo make install
after make.

Stefan

NovoG93, 2019-03-13:
Hello Stefan,

gst-launch-1.0 tcambin serial="07510213" ! videoconvert ! ximagesink produces a live-view of the camera. 
[gst-launch.txt](https://github.com/TheImagingSource/tiscamera/files/2960642/gst-launch.txt)

Georg

TIS-Stefan, 2019-03-13:
Hello Georg,

> gst-launch-1.0 tcambin serial="07510213" ! videoconvert ! ximagesink produces a live-view of the camera. 

OK, the camera and software works on your computer. That is some progress.

Warum bitte tritt denn immer ein "Speicherzugriffsfehler" auf? Ich muss zugeben, ich bin da hilflos.
I stumble on GRAY8. It should have been "BGRx" instead. Can you try that please?

Can you try to figure out, where the Speicherzugriffsfehler happens? I do not have ROS installed, because I am not really a ROS expert. I only tried to create a workround around ROS gscam using the outdated GStreamer 0.10 version. 

Stefan

NovoG93, 2019-03-15:
Hello Stefan,

tried the camera on a different computer on which we last changed something with the software over a year ago. Back then the cameras were working just fine. 

I also changed "pixelformat" to "BGRx" still the same error.

I am going to try it on a fresh install of ROS Kinetic on Ubuntu 16.04 today. If this does not work I will reinstall the Linux Firmware to the cameras and then try again. Do I need to purge the Firmware first or can I just overwrite it?

Thank you for your help,
Georg

TIS-Stefan, 2019-03-15:
Georg

I just found, that ROS Kinetic ONLY supports Wily (Ubuntu 15.10), Xenial (Ubuntu 16.04) and Jessie (Debian 8) for debian packages. (source: http://wiki.ros.org/kinetic/Installation/Ubuntu)

I use 18.04 only. The only solution is trying to install Ubunuto 16.04 in a VM and hoping, that this works.
Then I can try to reproduce your issue and find a solution. 

The firmware 3014 in your camera is correct, I use the same and get a nice image using tcam-capture in my Ubuntu 18.04.

The error you receive seems to be from gscam using the pipeline "shmsrc socket-path=/tmp/ros_mem ! video/x-raw-rgb, width=1280,height=960,framerate=258/1,bpp=24,depth=24,blue_mask=16711680, green_mask=65280, red_mask=255 ! ffmpegcolorspace"

This pipeline expects an colored rgb video stream, which is created by " ! videoconvert ! video/x-raw,format=RGB ," in the source pipeline. Thus, I wonder, where the error is.

However, I have to setup an Ubuntu 16.04 in an VM first. This will take some time.

Stefan

TIS-Stefan, 2019-03-15:
Hallo Georg

I installed an Ubuntu 16.04 installed, tiscamera, ros etc. Took only 2,5 hours.....

Good news: No Speicherzugriffsfehler on my system.

The tiscamera initialization is 

cam = tiscamera.Camera("39210268", 1280, 960, 15, False, True)

This also opens an extra live video window, so we can see, whether the camera runs. If this does not popup, dis- and reconnect the camera.

If the Speicherzugriffsfehler continues, we need a stacktrace. Start gdb:
gdb --args python3 ./tis_rosstarter

when then error occurs, enter "bt". This give the stack-trace. Post it.

BTW: I got many problems with

input('Press enter to continue: ')

It seems, gscam makes some output, which is caught and so we get an unwanted "EOF" error there. For testing I exchanged it with time.sleep(30)

Stefan

NovoG93, 2019-03-20:
Hallo Stefan,

also tried it on a fresh install and it works.
Seems like something messed with our ROS installation.
Thank you for your help.
You can mark this as closed.

Georg

closed 2019-07-04 by TIS-Edgar