TheImagingSource/tiscamera #261: DFK 33UJ003

MChow1, 2019-03-13:
I want to handle the camera settings using opencv for detection application. I started with the dependencies installation using (https://github.com/TheImagingSource/tiscamera) **Building tiscamera**. During this step:

# With ARAVIS:
cmake -DBUILD_ARAVIS=ON -DBUILD_GST_1_0=ON -DBUILD_TOOLS=ON -DBUILD_V4L2=ON -DCMAKE_INSTALL_PREFIX=/usr ..
# Without ARAVIS
cmake -DBUILD_ARAVIS=OFF -DBUILD_GST_1_0=ON -DBUILD_TOOLS=ON -DBUILD_V4L2=ON -DCMAKE_INSTALL_PREFIX=/usr ..

i get this error:

>  ~/tiscamera/build$ cmake -DBUILD_ARAVIS=ON -DBUILD_GST_1_0=ON -DBUILD_TOOLS=ON -DBUILD_V4L2=ON -DCMAKE_INSTALL_PREFIX=/usr ..
> CMake Error at /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:367 (message):
>   A required package was not found
> Call Stack (most recent call first):
>   /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:532 (_pkg_check_modules_internal)
>   cmake/modules/FindGObjectIntrospection.cmake:39 (pkg_check_modules)
>   src/gobject/CMakeLists.txt:20 (find_package)
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/tiscamera/build/CMakeFiles/CMakeOutput.log".

I have no clue with this as a beginner, suggestion would be of great help.

TIS-Stefan, 2019-03-13:
Hello 

Please checkout the "development" branch. In this branch the problem is solved.

Stefan

closed 2019-03-13 by TIS-Stefan

MChow1, 2019-03-13:
How do i do that?
I wanted help with camera settings

TIS-Stefan, 2019-03-13:
Hello

I am not a git usage trainer....

However, change into your tiscamera directory. Enter
git checkout development
cd build 
make -j

And see, what happens.
Stefan