TheImagingSource/tiscamera #263: Loading firmware to the camera

yoelk, 2019-03-28:
Is it possible to reload firmware to a camera?
I have a TIS_GIGE29_WUXGA_M_Z12.

The reason is ask is that sometimes some camera parameters get "stuck" and can not be modified.
Sometimes unplugging the power from the camera helps, but sometimes not, so I was wondering what possibilities I have for resetting the camera's inner state (e.g. by loading firmware).

Thanks,
Joel

TIS-Stefan, 2019-03-28:
Hello Joel

You may let me now, which parameters get "stuck". also which software you use. 
Also I need the serial number of the camera.

There is no camera reset. Unpowering is the only reset.

Stefan

yoelk, 2019-03-29:
Hi Stefan,
The code version we use is: tiscamera_0.10.0.2329.development_1e0b465_amd64.deb
I don't have the camera's ID at the moment. I'll write it when I do.
The stuck parameters are either the Zoom or the Focus (this time the Focus).
Unpowering did make focus unstuck but, auto-focus remained always on, and we couldn't turn it off.

yoelk, 2019-03-29:
If you want to see the problem together with me, I can connect you via TeamViewer to see the problematic camera.

TIS-Stefan, 2019-03-29:
Hi Joel

I suggest to use the github code. Either master or development branch.

Stefan

yoelk, 2019-03-31:
Dear Stefan, we've been through this already. We can only use stable releases.
And regardless, this version was done by Arne especially for us, to overcome the Aravis MTU negotiation problems.

Can we get support from you regarding this problem?

TIS-Stefan, 2019-04-01:
Hello Joel

I will notice Arne on this.

Stefan

TIS-Stefan, 2019-04-01:
Hello Joel

> Dear Stefan, we've been through this already.

I did not knew that.

What does "stuck" mean: You move the sliders of zoom, but the camera does not react? Same with focus, the lens does not move?

Stefan

TIS-Arne, 2019-04-01:
Hi Joel, 
I built a package from the latest development version of tiscamera. Could you please check whether the problem persists?

[tiscamera_0.10.0.2513~development_b4a58b2_amd64.zip](https://github.com/TheImagingSource/tiscamera/files/3029001/tiscamera_0.10.0.2513.development_b4a58b2_amd64.zip)

Thanks, 

/Arne

yoelk, 2019-04-01:
Thanks Arne. Does it include the MTU change you did for us?
The one that checks the TCAM_GIGE_PACKET_SIZE environment variable.

TIS-Arne, 2019-04-01:
@yoelk Yes, it should. This change is present in all development versions since commit 1e0b465b.

yoelk, 2019-04-01:
Great! Thanks!

closed 2019-07-04 by TIS-Edgar