TheImagingSource/tiscamera #264: gst-launch-1.0 tcambin ! videoconvert ! ximagesink

StefanHope, 2019-04-04:
Hello, guys.When I try to use "gst-launch-1.0 tcambin ! videoconvert ! ximagesink" to test a usb camera in pi 3B+ in a GUI os, it just show nothing but this error. 

----------------------------------------------------
Setting Pipeline to PAUSED...
ERROR: Pipeline doesn't want to pause.
Setting Pipleline to NULL...
Freeing Pipleline...
----------------------------------------------------

Thanksss

TIS-Arne, 2019-04-04:
Hi Stefan, 

what is the model name of the camera?

StefanHope, 2019-04-04:
thank you so much. I have checked the support model except mine. Sorry...

vishal-prgmr, 2019-07-17:
Hi @TIS-Arne , I have the same error now. I built everything as per the readme ( hopefully). Installed all the dependencies as well. 
OS: Ubuntu 18.04.1

```
sdfl@sdfl:~/stcam/tiscamera/build$ gst-launch-1.0 tcamsrc ! video/x-raw, format=GRAY16_LE, width=640, height=480, framerate=30/1 ! videoconvert ! ximagesink
110324     <DEBUG> Logger:107: 
The following library versions are used:
	Tcam:	0.11.1_development/fdb3dfa_rev_2587
	Aravis:	0.6_version_1758883
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
Setting pipeline to NULL ...
Freeing pipeline ...
````

TIS-Stefan, 2019-07-17:
please let me know, which camera model you use. must be something like
DFK 33GX273 or DFK 33UX273

Why do you want to use GRAY16_LE?

vishal-prgmr, 2019-07-18:
Thank you for the reply @TIS-Stefan. I solved the problem. My problem was that the camera was on a different network.

TIS-Stefan, 2019-07-18:
Thank you for your feedback!

closed 2019-07-18 by TIS-Stefan