TheImagingSource/tiscamera #265: Hello, I have changed a camera which is DMK 23GP031. Got those error!

TIS-Arne, 2019-04-04:
Hello, I have changed a camera which is DMK 23GP031. Got those error!
------------------------------------------------------------------------------
pi@raspberrypi:~ $ gst-launch-1.0 tcambin ! videoconvert ! ximagesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
INFO:
Working with src caps: video/x-raw, framerate=(fraction)400000/57327, width=(int)2592, height=(int)1944, format=(string)GRAY16_LE
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source: Device lost
Additional debug info:
/home/pi/Desktop/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp(1045): gst_tcam_src_device_lost_callback (): /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source
Execution ended after 0:00:03.493881237
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
------------------------------------------------------------------------

_Originally posted by @StefanHope in https://github.com/TheImagingSource/tiscamera/issues/81#issuecomment-479823077_

TIS-Arne, 2019-04-04:
@StefanHope The error says "device lost" which means that the camera disconnected from the usb bus. This might be due to a faulty USB cable or - especially on the RaspberryPi - due to lack of Power. Are you using a powered USB bus with this camera? The Pi might not provide enough power by itself. 

Also keep in mind that the Pi only has USB 2.0 ports. You need to set a way lower frame rate to get the camera working with only a USB 2.0 connection.

StefanHope, 2019-04-04:
DMK 23GP031 is a GigE camera.

StefanHope, 2019-04-04:
By the way, pi 3B+ support Gigabit Network。

TIS-Arne, 2019-04-04:
My bad. I did not look close enough. In this case you should try setting a lower frame rate and only use the 8 bit GRAY format to lower the datarate. The ethernet port of the Pi is connected via USB 2.0 only and the high datarate will clogg the network connection which causes the software to think the device got lost.

StefanHope, 2019-04-04:
Thanks so much. Could you please give me some instructions for how to set in a lower frame rate?

closed 2019-11-29 by TIS-Stefan

Kunj6354, 2023-02-23:
[image]
I am using banana pi and HDF5640 camera. I changed video0/1/2 but only video1 is showing the correct value. so video port is 1.

TIS-Stefan, 2023-02-24:
I am afraid, you are in the wrong issue.