TheImagingSource/tiscamera #266: Patched DFK41AU02 wont deliver color with ROS melodic on Ubuntu 18.04

cmeghdadi, 2019-04-08:
I used  the patch from https://github.com/TheImagingSource/tiscamera/issues/225 to get the camera running on Ubuntu 18.04 with ROS melodic but when i activate color in the TIS_rosstarter the stream wont start. I tried different Pixelformats like YUY2 and UYVY (since tcam-capture shows me pictures on YUY2, bggr and gray8). I also reduced the frames to 7 when using color but still not starting. Is there any solution.

Thanks in Advance

TIS-Stefan, 2019-04-08:
Hello 

How did you patch a DFK 41AU02? There are only UDEV rules to add, https://github.com/TheImagingSource/tiscamera/wiki/UDEV-Rules

Does the standard pipeline
gst-launch-1.0 tcambin ! video/x-raw,format=RGB ,width=1280,height=960,framerate=15/1 ! videoconvert ! ximagesink

show an image?

cmeghdadi, 2019-04-10:
The "patch" stated in #225 just simplifies some lines in the tiscamera.py file. Without that the TIS_Rosstarter wasnt starting at all.

UDEV rules were applied before but could not fix the problem

I got two cameras DMK42AUC03 which works with the standard pipeline and DFK41AU02 which is not working. It just gives me that answer.
Leitung wird auf PAUSIERT gesetzt ...
Leitung ist aktiv und erfordert keinen VORLAUF …
INFO:
Working with src caps: NULL
Leitung wird auf ABSPIELEN gesetzt ...
New clock: GstSystemClock
FEHLER: Von Element /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source: Internal data stream error.
Zusätzliche Fehlerdiagnoseinformation:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.402453620
Leitung wird auf PAUSIERT gesetzt ...
Leitung wird auf BEREIT gesetzt ...
Leitung wird auf NULL gesetzt ...
Leitung wird geleert ...

cmeghdadi, 2019-04-10:
I put 7/1 instead of 15/2 so the standard pipeline is working with:
gst-launch-1.0 tcambin ! video/x-raw,format=RGB ,width=1280,height=960,framerate=15/2 ! videoconvert ! ximagesink

PS: The output with framerate 15/1 is automatically gray

TIS-Stefan, 2019-04-10:
15 fps are available in the raw data modes only, so you get bayer raw data. I suppose, that is the reason.

cmeghdadi, 2019-04-17:
When changing the framerate in line 64 of tiscamera.py from
framerate=%d/1 to
framerate=%d/2
and giving the tisrosstarter 15 frames it displays a bgr picture (the one with the bayer pattern) in rviz
just trying to convert it to normal pic now

TIS-Stefan, 2019-04-17:
Hi

Seems, you found the design error....
Stefan

closed 2019-07-04 by TIS-Edgar