TheImagingSource/tiscamera #267: TcamCamera new frame callback function stop working gradually on ARM platform

veuron2018, 2019-04-08:
Same video capture codes based on TcamCamera new frame callback are used on Windows x64 platform, Linux x64 platform and Arm platform; but could not work properly on Arm platform.

Arm platform: it has 6 cores, 2 of which are Arm A72 at 1.8GHz
Video format: 1280x960@30fps, 640x480@30fps 
Testing: only keep reading data, nothing else; also tested tcam_demo
The problem: new frame callback was called for some frames, then stopped being called any more; only 40% cpu was used. When format changed to 1280x960@10fps, callback was near to normal but still died from time to time.

Is there a buffer setting in gstreamer or elsewhere which is too small in this embedded platoform case to stop callback working?

Looking forward to your help.

TIS-Stefan, 2019-04-08:
Hello

The problem are frame drops. If the frame rate is too high, the USB controller can not pickup and save all data packets into memory, which results in incomplete images. These images are dropped.

That is, what you see. Unfortunately I can not do anything about this on ARM CPUs.

Stefan

veuron2018, 2019-04-23:
Thanks for reply.
Yes, it turns out the reason is bandwidth of USB3.0 on ARM platform.

Another question

veuron2018, 2019-04-23:
Thanks for reply.
Yes, it turns out the reason is bandwidth of USB3.0 on ARM platform.

Another question, is USB3.0 camera supported on Android platform?

TIS-Stefan, 2019-04-23:
Hi

is USB3.0 camera supported on Android platform?

Well, If USB 3.0 controllers available, then the Android system should support them, at least with higher speed than USB 3.0. The maximum speed, USB 3.0 is specified for must not necessarily be achieved. This is hardware dependent.
In our tests TI controllers were slower than Renesas, which was slower than Intel and Fresco Logic.

Stefan

closed 2019-11-29 by TIS-Stefan