TheImagingSource/tiscamera #268: Missing camera controls on Tegra TX2

thomas-perschke, 2019-04-09:
Hello,

I have a problem with a DMK 33UX290 similar to issue #241 (<case:466571>). My platform is a Nvidia Tegra TX2 module with a Connect Tech Astro carrier board with Ubuntu 16.04 LTS.

I cannot see the extended uvc controls. Trying to set it manually as described on the corresponding UVC-extension wiki page fails with errors 2(No such file or directory) and error 22 (invalid argument).

Using uvcvideo with tracing, I see the message "Can't add mapping 'Auto Functions ROI Preset', control id 0x199e2510 is invalid"  with dmesg when I plug in the camera or try to upload the usb3.xml file with uvcdynctrl.

Curiously, I see the controls on a similar Nvidia Tegra TX2 development board. All controls are added when the camera is plugged in. I added the camera to that board about 9 month ago. Additionally, I tried to use the git revision on this board on the new board but it still fails to load the extensions.

Looking at issue #241 I could not see, if and how the problem was resolved.

Best regards,
                       Thomas.

TIS-Stefan, 2019-04-10:
Hello

This is an issue of the Linux distribution of the TX2. I expect Gain Auto missing too.
The only workaround we found is using Aravis USB3Vision instead of V4L2 on the TX 2 with our tiscamera modules. 

Please unzip this patch:
[aravis_usbvision_0_10.zip](https://github.com/TheImagingSource/tiscamera/files/3062428/aravis_usbvision_0_10.zip)

Then rebuild tiscamera
```
cd tiscamera
patch -p1 aravis_usbvision_0_10.patch
mkdir build
cd build
cmake -DBUILD_ARAVIS=ON -DBUILD_TOOLS=ON ..
make
sudo make install

```
tcam-ctrl should list your camera now as Aravis device similar to 
`DFK 37UX265	ARAVIS	34819936`
== known problems:
indexing takes long -> aravis issue. gige-daemon only lists gige devices not UsbVision
high framerates do not work -> aravis issue. 

Stefan

thomas-perschke, 2019-04-10:
Hello Stefan,
yes, about 30 controls are missing. 

I followed your suggestions and can see the v4l2 and the Aravis device with tcam-ctrl. My problem is now, that I use my own camera software based on v4l2 to include cameras in my software system. If the camera only loads the controls as Aravis device, it would mean I have to rewrite the software using the Aravis API?

I am still wondering why the camera works perfectly on my Nvidia Tegra TX2 development board. It has the R28 1.0 L4T kernel, while I flashed the new board with R28 2.1 and the Connect Tech patches. Do you think it would be worthwhile to flash the new board with R28 1.0?

Best regards,
                         Thomas.

TIS-Stefan, 2019-04-10:
Hi Thomas

if you have your own v4l2 based software, you may ask NVidia, why they removed or miss the UVC Extension from their Kernel.

On the dev board you may use a different Kernel?

However, did you follow the manual at https://github.com/TheImagingSource/tiscamera/wiki/UVC-Extension-Units ?

Stefan

thomas-perschke, 2019-04-10:
Hello Stefan,

yes, as I wrote, I use R28 2.1 on the board where the  loading of the extensions fails. On the other board, where everything is fine, I use R28 1.0.

Yes, I followed the manual at https://github.com/TheImagingSource/tiscamera/wiki/UVC-Extension-Units  but it fails to add the extensions.

Another point is, that the USB3 port of the Connect Tech carrier may be influenced by the corresponding BSP. I will have a closer look.   
 
By the way, the error message "Can't add mapping 'Auto Functions ROI Preset', control id 0x199e2510 is invalid" disappeared. It seems I messed up the usb3.xml file when I tried the older git revision.

Best regards,
                      Thomas.

thomas-perschke, 2019-04-11:
Hello Stefan,

after further investigations I managed to get an uvcvideo driver, which loads the UVC extensions. It was a small change in the uvcvideo driver from the L4T kernel version R28 1.0 toR28 2.1. As reference for other people which may encounter this issue:

I had to remove the line:

`if (ctrl->info.size < mapping->size)
		return -EINVAL;`

in the function `int uvc_ctrl_add_mapping(struct uvc_video_chain *chain,const struct uvc_control_mapping *mapping)` in the file _uvc_ctrl.c_ for the R28 2.1 kernel.

It is not clear for me, what the size problem is related to, but it may help you to adapt your software to the TX2 platform. 

From my side, you can close the issue. Thanks for the help.

Best regards,
                         Thomas.

TIS-Stefan, 2019-04-11:
Hello Thomas

Thank you very much!

Stefan

TIS-Stefan, 2019-11-29:
Just an addition: With latest Jetpack all camera properties, even Gain Auto will be avaiable for USB cameras. 
Alternatively, tiscamera 0.11.0 develope branch can be build for using USB3Vision instead of V4L2. This gives:
more granulartiy in fps
regions of interrest on the sensor
all camera properties
higher performance on TX2

closed 2019-11-29 by TIS-Stefan