TheImagingSource/tiscamera #269: Cameras have different brightness under different driving conditions

xiat5, 2019-04-20:
We use “DFK 72BUC02” on Linux.   The new driver"dfk72uc02_3012.euvc" and the old driver are both executed at the lowest exposure time, the brightness of the image is not the same, and the image with the new driver is brighter.  And when the camera is set at the lowest exposure time, the brightness of the image on Windows and Linux is not the same, and the image on Linux is brighter.
Our products are used in very bright scenes. It requires the camera to work at the lowest exposure time, the lowest gain and the lowest aperture. A batch of cameras bought last year are used up. The new batch of cameras can not use the old driver. The image is too bright under the new driver. Is there a solution?
Thanks very much！

TIS-Stefan, 2019-04-23:
Hello

You are right, using the Linux firmware "dfk72uc02_3012.euvc" gives different results for brightness, than using the the older Windows firmware. 
Windows and Linux use the exposure time somewhat differently, therefore, the brightness results will be different. 
Please use the "dfk72uc02_3012.euvc" firmware in Linux only. It is not released for use with the Windows driver.

I am afraid, you have to accept and live with the differences.

Stefan

xiat5, 2019-04-24:
Hi Stefan，
If there is any way to modify the driver or firmware to reduce the brightness, please let us know. If not, we have to customize the lens, which will bring a lot of work.
Thanks all the same.

xia

TIS-Stefan, 2019-04-24:
Hello Xia

Currently the engineers are bound to other projects, the DxK 72 development is paused. Also changing the behavior of the camera would cause trouble for users, that use the camera with the 3012 firmware.

Which camera setting do you use? Can you list out all settings?

Stefan

TIS-Stefan, 2019-11-29:
Closed. In Windows and Linux different firmware versions are to be used. They use different ways to control the sensor, so the set exposure and brightness values will have different results. 
I am sorry for this.

closed 2019-11-29 by TIS-Stefan